CarlosDerSeher/snapclient #2: Cannot compile using PCM51xx

miLORD1337, 2021-11-07:
Hey Carlos,
I'm trying to compile for using my ESP32 with a PCM51xx. Unfortunately, I get the error message `../components/custom_board/generic_board/include/board.h:33:10: fatal error: esp_peripherals.h: No such file or directory`.

(Without really knowing what I'm doing) I added `../esp_peripherals/include` to the includedirs, but then `audio_common.h` is missing and that depends on `esp_adf_libs` again...

Do I have to have esp-adf installed as well despite the components you copied into your repo?

Thanks,
miLORD

CarlosDerSeher, 2021-11-08:
I have never tested this using other HW than the Lyrat v4.3. I had to edit some of the components I copied from ADF to the project to make it compile without esp_adf_libs. ADF isn't necessary to be installed. I am at work currently but when I have time I'll have a look at this and try to compile it myself using custom board to see what might be the issue. But it seems there is some entry missing in the Cmake files

CarlosDerSeher, 2021-11-12:
Ok, should be able to compile now, except MA120X0. Will have to check upstream code why there are so many errors on my fork. Can't test it though on my end. Reason it was broken was because esp_peripherals component just includes the i2c driver part. Anything else got dropped by myself. You won't be able to use any keys or sd card on a dev board with this code, currently I don't have use for this because I am only interested on the codec part and getting syncing right. Also you can check out NETCONN branch where I actively do development on, trying to minimize (peak) RAM usage as much as possible to get it to run on those ESP modules with no PSRAM. Still all this code is very much WIP and kind of dirty, same as master.

Hopefully everything works correctly to clone, especially those submodules, I've some trouble pushing these correct because I don't understand how to use them as it seems :/

miLORD1337, 2021-11-12:
Hey Carlos,
thanks a bunch for looking into it! That error went away, but then I got stuck at `ogg.h not found`. So I downloaded libogg and built it, put it into the include dir and that went away. 

But now `../components/flac/flac/src/libFLAC/windows_unicode_filenames.c:36:10: fatal error: io.h: No such file or directory` keeps holding me up. Any idea on how to proceed? Something with VS Buildtools to be installed maybe?

CarlosDerSeher, 2021-11-15:
I never enabled ogg support. From flac README:

"If you want to build without Ogg support, instead edit all .vcproj files and remove any "FLAC__HAS_OGG" definitions."

I didn't enable OGG in flac's config.h, not sure what is the Problem on your end using visual studio (if I understand correctly). I am doing development on an ubuntu machine with eclipse IDE using espressif's plugin. Maybe you have to change some entries in flac component config.h to be compatible with windows build system?

CarlosDerSeher, 2021-11-15:
You could also try to run ` idf.py build` from a shell and see if that works out. Didn't try it on windows though

miLORD1337, 2021-11-16:
Okay, I see. Thanks, I'll try disabling it on my environment as well and report back. I might try it on my linux machine as well.

miLORD1337, 2021-11-16:
It fails on Debian with the exact same errors. Is esp-idf 4.3 still the correct version?

CarlosDerSeher, 2021-11-17:
Alright, I think I found the problem. Since I have done some dirty thing to the flac repo along the road this won't work with a fresh submodule init. I deleted all ogg*.h and ogg*.c files in libFlac and it's include directory. Also I deleted windows_unicode_filenames.c Afterwards it should compile and start to work. At least on a Linux machine. I'll have to look into that at some point so this won't be necessary, but hopefully this will get you going.

CarlosDerSeher, 2021-11-18:
Ok, not sure if I did this the right way, but I excluded these files now in flac's CmakeList and compilation should work now successful. On Master I get saome error regarding certificates using IDF v4.31 but using NETCONN branch it shoudl compile. Not sure whats the problem on master, but this has something to do with IDF. I think disabling TLS removes that problem.

miLORD1337, 2021-11-20:
Without looking too deep into it, it might be this issue here: https://github.com/espressif/esp-idf/issues/5322#issuecomment-962219803

I posted a solution there as well:

> You can do it with idf.py menuconfig as follows:
Navigate to Component config --> mbedTLS --> Certificate Bundle --> Default certificate bundle options and choose use only most common certificates, then build again. Good luck!

miLORD1337, 2021-11-20:
It works! 🚀 
Great stuff, thank you very much! I'm on NETCONN now, though. I had no problem compiling master, likely due to my fix above. It didn't play anything though. I tried disabling everything regarding I2C, because I don't have that on my board. NETCONN worked out of the box for me. It keeps getting in and out of sync a bit, but I'm not picky.

One more thing though: Lines `620 `and `2858 `should use `SNAPCAST_CLIENT_NAME` to honour sdkconfig settings. A pull request for that would've been overkill I think ;) 

Thanks again!
miLORD

CarlosDerSeher, 2021-11-21:
Glad to hear you finally managed to get it to run. I have abandoned master quite some time ago. On NETCONN I am trying to reduce (peak) RAM usage as much as I can. Regarding your out of sync issue, using Lyrat and comparing it to raspberry clients I don't have any issues but assumed that in another setting this will happen. How long is your sync buffer? Using FLac, the maximum with stable playback without any errors was around 750ms maybe a bit less. I also had problems when I was using 40MHz Wifi channels on the router.

CarlosDerSeher, 2021-11-21:
Ah and I just remembered now that I did another modification to flac to reduce overall RAM usage, in file `components/flac/flac/src/libFLAC/bitreader.c` I set `FLAC__BITREADER_DEFAULT_CAPACITY` from `static const uint32_t FLAC__BITREADER_DEFAULT_CAPACITY = 65536u / FLAC__BITS_PER_WORD;` to `static const uint32_t FLAC__BITREADER_DEFAULT_CAPACITY = (65536u / 4) / FLAC__BITS_PER_WORD;` Maybe this will help with out of sync too? 

Is the problem actually out of sync playback of clients or are there resyncing events happening? Also If buffer is set too large on the server, the client might fail to allocate the audio chunk and it will get out of sync by one chunk. This is an issue I am still working on

miLORD1337, 2021-11-22:
Okay, my sync buffer is too large at 1000 ms, that's one thing. I'll dig into that a bit.
But despite the modification to `FLAC__BITREADER_DEFAULT_CAPACITY` I get `PLAYER: couldn't get memory to insert chunk` error messages and as you said sync gets worse ever after. Forcing a hard resync by muting/unmuting gets it back to perfect.

miLORD1337, 2021-11-22:
After changing the buffer to 500 ms it worked perfectly for an hour. Thanks again!

CarlosDerSeher, 2021-11-23:
That's great to read :) If you set it that low you probably could leave `FLAC__BITREADER_DEFAULT_CAPACITY` at it's original size. I just reduced that so I can push the sync buffer limit up a bit. I really would like to support 1000ms on those little WROOM modules but wasn't able to achieve this yet :/

closed 2021-11-23 by CarlosDerSeher