CarlosDerSeher/snapclient #3: Do need some help with compillation

johann1525, 2022-02-21:
Hi,

i think this is an amazing project and because i'm using Snapcast a lot i thought it would be great to be able to reduce the client side to a simple ESP. I tried to compile it with the standard hardware config. Just to see if it can compile. But unfortunately i get to this error. After googling it a bit i think it is just some easy fix for someone who can write c code. Unfortunately i'm kind of an amateur. So can you maybe bring me on the right track with this please?

Thank you very much in advance.

My final goal is to output the i2s signal from a generic ESP32 module to a ADAU1701 for equalisation and room correction. So there is some question if it can work without the i2c ADC control pins. But first the small goals have to be met. ;) 

```
Executing action: all (aliases: build)
Running ninja in directory /home/johann/projects/esp/snapclient/build
Executing "ninja all"...
[1/621] Performing build step for 'bootloader'
[1/1] cd /home/johann/projects/esp/snapclient/build/bootloader/esp-idf/esptool_py && /home/johann/.espressif/python_env/idf5.0_py3.9_env/bin/python /home/johann/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/johann/projects/esp/snapclient/build/bootloader/bootloader.bin
Bootloader binary size 0x63e0 bytes. 0xc20 bytes (11%) free.
[7/619] Building C object esp-idf/audi...dir/lyrat_v4_3/board_pins_config.c.obj
FAILED: esp-idf/audio_board/CMakeFiles/__idf_audio_board.dir/lyrat_v4_3/board_pins_config.c.obj 
/home/johann/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -Iconfig -I../components/audio_board/include -I../components/audio_board/lyrat_v4_3 -I/home/johann/esp/esp-idf/components/newlib/platform_include -I/home/johann/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/johann/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/johann/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/johann/esp/esp-idf/components/freertos/esp_additions/include -I/home/johann/esp/esp-idf/components/esp_hw_support/include -I/home/johann/esp/esp-idf/components/esp_hw_support/include/soc -I/home/johann/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/johann/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/johann/esp/esp-idf/components/heap/include -I/home/johann/esp/esp-idf/components/log/include -I/home/johann/esp/esp-idf/components/lwip/include/apps -I/home/johann/esp/esp-idf/components/lwip/include/apps/sntp -I/home/johann/esp/esp-idf/components/lwip/lwip/src/include -I/home/johann/esp/esp-idf/components/lwip/port/esp32/include -I/home/johann/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/johann/esp/esp-idf/components/soc/include -I/home/johann/esp/esp-idf/components/soc/esp32/. -I/home/johann/esp/esp-idf/components/soc/esp32/include -I/home/johann/esp/esp-idf/components/hal/esp32/include -I/home/johann/esp/esp-idf/components/hal/include -I/home/johann/esp/esp-idf/components/hal/platform_port/include -I/home/johann/esp/esp-idf/components/esp_rom/include -I/home/johann/esp/esp-idf/components/esp_rom/include/esp32 -I/home/johann/esp/esp-idf/components/esp_rom/esp32 -I/home/johann/esp/esp-idf/components/esp_common/include -I/home/johann/esp/esp-idf/components/esp_system/include -I/home/johann/esp/esp-idf/components/esp_system/port/soc -I/home/johann/esp/esp-idf/components/esp_system/port/include/private -I/home/johann/esp/esp-idf/components/xtensa/include -I/home/johann/esp/esp-idf/components/xtensa/esp32/include -I/home/johann/esp/esp-idf/components/driver/include -I/home/johann/esp/esp-idf/components/driver/esp32/include -I/home/johann/esp/esp-idf/components/driver/deprecated -I/home/johann/esp/esp-idf/components/esp_pm/include -I/home/johann/esp/esp-idf/components/esp_ringbuf/include -I/home/johann/esp/esp-idf/components/efuse/include -I/home/johann/esp/esp-idf/components/efuse/esp32/include -I/home/johann/esp/esp-idf/components/vfs/include -I/home/johann/esp/esp-idf/components/esp_wifi/include -I/home/johann/esp/esp-idf/components/esp_event/include -I/home/johann/esp/esp-idf/components/esp_netif/include -I/home/johann/esp/esp-idf/components/esp_eth/include -I/home/johann/esp/esp-idf/components/tcpip_adapter/include -I/home/johann/esp/esp-idf/components/esp_phy/include -I/home/johann/esp/esp-idf/components/esp_phy/esp32/include -I/home/johann/esp/esp-idf/components/esp_timer/include -I../components/audio_sal/include -I../components/audio_hal/include -I../components/audio_hal/driver/es8388 -I../components/audio_hal/driver/es8374 -I../components/audio_hal/driver/es8311 -I../components/audio_hal/driver/es7243 -I../components/audio_hal/driver/es7148 -I../components/audio_hal/driver/es7210 -I../components/audio_hal/driver/tas5805m -I../components/audio_hal/driver/include -I../components/esp_peripherals/driver/i2c_bus -I../components/dsp_processor/include -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -O2 -fmacro-prefix-map=/home/johann/projects/esp/snapclient=. -fmacro-prefix-map=/home/johann/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-1186-gb149ff8dda-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/audio_board/CMakeFiles/__idf_audio_board.dir/lyrat_v4_3/board_pins_config.c.obj -MF esp-idf/audio_board/CMakeFiles/__idf_audio_board.dir/lyrat_v4_3/board_pins_config.c.obj.d -o esp-idf/audio_board/CMakeFiles/__idf_audio_board.dir/lyrat_v4_3/board_pins_config.c.obj -c ../components/audio_board/lyrat_v4_3/board_pins_config.c
../components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'i2s_mclk_gpio_select':
../components/audio_board/lyrat_v4_3/board_pins_config.c:113:11: error: implicit declaration of function 'PIN_FUNC_SELECT' [-Werror=implicit-function-declaration]
           PIN_FUNC_SELECT (PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
           ^~~~~~~~~~~~~~~
../components/audio_board/lyrat_v4_3/board_pins_config.c:113:28: error: 'PERIPHS_IO_MUX_GPIO0_U' undeclared (first use in this function)
           PIN_FUNC_SELECT (PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
                            ^~~~~~~~~~~~~~~~~~~~~~
../components/audio_board/lyrat_v4_3/board_pins_config.c:113:28: note: each undeclared identifier is reported only once for each function it appears in
../components/audio_board/lyrat_v4_3/board_pins_config.c:113:52: error: 'FUNC_GPIO0_CLK_OUT1' undeclared (first use in this function)
           PIN_FUNC_SELECT (PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
                                                    ^~~~~~~~~~~~~~~~~~~
In file included from /home/johann/esp/esp-idf/components/soc/esp32/include/soc/soc.h:11,
                 from /home/johann/esp/esp-idf/components/soc/include/soc/soc_memory_types.h:12,
                 from /home/johann/esp/esp-idf/components/esp_hw_support/include/soc/compare_set.h:12,
                 from /home/johann/esp/esp-idf/components/esp_hw_support/include/soc/spinlock.h:12,
                 from /home/johann/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:42,
                 from /home/johann/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:51,
                 from /home/johann/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:63,
                 from ../components/audio_hal/include/audio_hal.h:30,
                 from ../components/audio_board/lyrat_v4_3/board.h:30,
                 from ../components/audio_board/lyrat_v4_3/board_pins_config.c:29:
../components/audio_board/lyrat_v4_3/board_pins_config.c:114:27: error: 'PIN_CTRL' undeclared (first use in this function)
           WRITE_PERI_REG (PIN_CTRL, 0xFFF0);
                           ^~~~~~~~
/home/johann/esp/esp-idf/components/esp_common/include/esp_assert.h:23:71: note: in definition of macro 'TRY_STATIC_ASSERT'
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                                                                       ^~~~~~~~~
[... cut ...]
             ASSERT_IF_DPORT_REG((addr), WRITE_PERI_REG);                                                               \
             ^~~~~~~~~~~~~~~~~~~
../components/audio_board/lyrat_v4_3/board_pins_config.c:142:11: note: in expansion of macro 'WRITE_PERI_REG'
           WRITE_PERI_REG (PIN_CTRL, 0xFF0F);
           ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[9/619] Building C object esp-idf/esp_...f_esp_hw_support.dir/clk_ctrl_os.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

CarlosDerSeher, 2022-03-03:
Which version of IDF do you have installed? Maybe a version mismatch? Also you shouldn't use master branch because it is broken. Change to NETCONN branch which is my current development branch

CarlosDerSeher, 2022-04-29:
Closing this because of no response.

closed 2022-04-29 by CarlosDerSeher