CarlosDerSeher/snapclient #7: I2S pin connection

TheHung184, 2022-07-27:
Hi, i am new to I2S and i have some UDA1334A board which i would like to try with your code. I have modify the PCM5102A board to add to config. My board is esp32-wemos-lolin32-lite
I have connect them as follow : 
        UDA1335a                      |                  ESP32 Lite
           WSEL                      --->>>            GPIO33 (mark as 33-pin13 ?)
           DIN                         --->>>            GPIO25 (mark as 25-pin14 ?)
           BCLK                       --->>>            GPIO0 (mark as 0-pin23 ?)
I havenot has any issue with compiling and detection of client but i cannot hear any sound from my DAC board. Can you share which pin connection you use ? Thank you
[image]
[image]

majoreffort, 2022-08-01:
I think you are using the wrong pin numbering scheme. The correct numbers should be the ones labeled GPIOxx in the pinout picture and also the ones printed on the PCB, so for example connect BCLK --> GPIO23 if using the default pin configuration. 
Also, if you didn't modify the code, don't connect the mute GPIO, as the logic is inverted on the PCM5102A (low is mute).

TheHung184, 2022-08-02:
> I think you are using the wrong pin numbering scheme. The correct numbers should be the ones labeled GPIOxx in the pinout picture and also the ones printed on the PCB, so for example connect BCLK [...]

I changed wiring as your suggestion and it worked. Thank you ^_^

closed 2022-08-02 by TheHung184