CarlosDerSeher/snapclient #8: Compilation error, need information about ESP-IDF version you use

markus1268, 2022-07-29:
Hi Carlos,
I try to compile the NETCONN branch, but i got some errors. Which ESP-IDF version do you use ? I use ESP-IDF version 4.4.1.

`D:/OnWorkspace/audionet/carlos/snapclient/components/custom_driver/i2s.c:1288:19: error: unknown type name 'i2s_port_t'; did you mean 'in_port_t'?
 i2s_custom_write (i2s_port_t i2s_num, const void *src, size_t size,
                   ^~~~~~~~~~
                   in_port_t
D:/OnWorkspace/audionet/carlos/snapclient/components/custom_driver/i2s.c:1348:26: error: unknown type name 'i2s_port_t'; did you mean 'in_port_t'?
 i2s_custom_write_expand (i2s_port_t i2s_num, const void *src, size_t size,
                          ^~~~~~~~~~
                          in_port_t
D:/OnWorkspace/audionet/carlos/snapclient/components/custom_driver/i2s.c:1444:18: error: unknown type name 'i2s_port_t'; did you mean 'in_port_t'?
 i2s_custom_read (i2s_port_t i2s_num, void *dest, size_t size,
                  ^~~~~~~~~~
                  in_port_t
D:/OnWorkspace/audionet/carlos/snapclient/components/custom_driver/i2s.c:655:1: warning: 'i2s_intr_handler_default' defined but not used [-Wunused-function]
 i2s_intr_handler_default (void *arg)
 ^~~~~~~~~~~~~~~~~~~~~~~~
D:/OnWorkspace/audionet/carlos/snapclient/components/custom_driver/i2s.c:113:21: warning: 'i2s_spinlock' defined but not used [-Wunused-variable]
 static portMUX_TYPE i2s_spinlock[I2S_NUM_MAX];
                     ^~~~~~~~~~~~
D:/OnWorkspace/audionet/carlos/snapclient/components/custom_driver/i2s.c:111:19: warning: 'p_i2s_obj' defined but not used [-Wunused-variable]
 static i2s_obj_t *p_i2s_obj[I2S_NUM_MAX] = { 0 };
                   ^~~~~~~~~
[1077/1295] Building CXX object esp-idf/esp-dsp/CMakeFiles/__idf_esp-dsp.dir/modules/matrix/mat/mat.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

Thanks for your help

TheHung184, 2022-08-01:
[image]
He said it is 4.3

CarlosDerSeher, 2022-08-06:
@markus1268 yes should compile using v4.3

markus1268, 2022-08-10:
Using ESP-IDF v4.3 can be build well. I only make small modification on main.c line 682 : `char *tmp = NULL;` To fix the compiler setting [-Werror=maybe-uninitialized]. Thanks all.

closed 2022-08-10 by markus1268