CarlosDerSeher/snapclient #9: Clients getting out of sync after a few days

CarlosDerSeher, 2022-09-07:
When clients are powered on they get perfectly synced with e.g. RPi clients. But I found if they stay powered on for some time (1 to 3 days) at some point they start playback out of sync and continue to play with the same amount of time out of sync.

Did anybody encounter this too?

majoreffort, 2022-09-11:
I think I had something similar in the past, but usually don't have my clients powered on that long. Until now one of my primary clients was using a WROVER board with additional PSRAM.

But lately I experienced another issue which may or may not be connected on WROOM boards: I can reproduce playback just completely stopping after about 30 - 45 minutes.

The log looks like the following:
`
I (03:11:42.829) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 2
I (03:11:42.830) c_I2S: DMA Malloc info, datalen=blocksize=2304, dma_buf_count=2
I (03:11:42.847) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (03:11:42.866) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (03:11:42.869) PLAYER: created new queue with 21
I (03:11:42.875) PLAYER: snapserver config changed, buffer 500ms, chunk 24ms, sample rate 48000, ch 2, bits 16 mute 0 latency 0
I (03:11:46.843) PLAYER: initial sync 93us
W (03:53:28.993) PLAYER: RESYNCING HARD 2: age 98074us, latency 491023us, free 171624, largest block 110592, 0, rssi: -64
W (03:53:29.000) PLAYER: RESYNCING HARD 1: age 56383us, latency 491023us, free 172596, largest block 110592, 0, rssi: -64
`
The resync messages is where it stops. Sometimes there are some other successful resyncs before, e. g. if connection is bad.
Maybe you have an idea what to look for here or maybe where to add additional log messages to figure out what is happening? Haven't looked into the player/syncing algorithm myself yet.

CarlosDerSeher, 2022-09-16:
You mean this happens always after 30 - 45 minutes? I have to say I haven't been testing much with PSRAM enabled... It seems that initial sync can never be achieved again. You could try commenting line 2764 in `lightsnapcast/player.c` 
`if (sendResyncMSg == 1)`
so we can see if it tries to resync and just isn't executing `free_pcm_chunk (chnk);` fast enough. This is one of the things on my ToDo list, I also could see such behavior sometimes but didn't have time to do coding on this project lately. My idea was to iterate over the queue (right before the message `RESYNCING HARD 1`) popping elements from the front in a while loop until a element with age < 0 is found or queue is empty. Maybe this would speed things up a little.

CarlosDerSeher, 2022-09-19:
@majoreffort I did some changes to NETCONN branch hopefully this will resolve your and my issue. I still need to do some long term testing though.

CarlosDerSeher, 2022-09-23:
Ok at least my own issue hasn't been resolved. From what I can tell it looks like server to client latency is increasing with time for some reason. I still have to compare this to badaix original snapclients to see if they behave similar but I doubt it. Shouldn't be wifi latency be more or less steady (if I take the median of 200 seconds) if environmental situation isn't really changing?!

majoreffort, 2022-09-23:
So I finally had a chance to test your changes, and as you suspected it just starts to spam RESYNCING HARD 1 once it stopped now. The only way to recover without resetting the ESP32 is to restart the snapserver to force a reconnect... 

> You mean this happens always after 30 - 45 minutes? I have to say I haven't been testing much with PSRAM enabled

Just to clarify btw: This happens on all my ESP32s _without_ PSRAM and absolutely always after 30-45 minutes. If I load the same binary file on one of my modules _with_ PSRAM it works fine (at least for a few hours). And being an identical binary, PSRAM is disabled in menuconfig. 

> Shouldn't be wifi latency be more or less steady (if I take the median of 200 seconds) if environmental situation isn't really changing?!

I can't imagine that to be the case, except if using other devices etc. on the same channel or neighbours interfering ;)
Spikes would be normal but as you said should average out for a large enough time frame.

CarlosDerSeher, 2022-09-23:
> This happens on all my ESP32s without PSRAM and absolutely always after 30-45 minutes.

Ok this is really strange... I just have a Lyrat v4.3 board for development at the moment, so I am unable to look into this. I cannot see any reason for this to happen though do you have a theory? These modules should look the same for the firmware if PSRAM is disabled in menuconfig shouldn't they? Maybe you use some pin on the Wroom modules which is related to PSRAM connection on Wrover modules?

Regarding increasing latency and my syncing issues, it seems like this was linked with too high buffer setting in snapserver and therefore memory allocation errors in esp snapclient. Reduced buffer and still testing, but playing music the whole day without any issues now.

majoreffort, 2022-09-25:
I only can think of a minor difference in hardware between the two chips which manifests itself somewhere in software, but as you said, they should be the same apart from the PSRAM. These are off the shelf dev boards, one NodeMCU and a TTGO T8 (WROVER).

Thinking about it, every time it stops, the queue is empty and calculated age is just fluctuating around a certain value > 1 chunk duration (was already the case before your changes) and not really decreasing. From my understanding this means that either chunks are coming in too slow or something about the timing/age is off and the chunks are not actually old?

Glad your issue is resolved. I use a 500 ms buffer @ 48000 Hz with FLAC (24 ms chunks) and never had issues with memory usage.

CarlosDerSeher, 2022-09-28:
Maybe you should try reducing flash memory clock in menuconfig? Or put spi driver ISRs into IRAM through menuconfig. Not sure how this is configured currently...

CarlosDerSeher, 2022-12-04:
Seems to be fixed with current NETCONN branch

closed 2022-12-04 by CarlosDerSeher