CarlosDerSeher/snapclient #10: Current NETCONN branch doesn't build

velaar, 2022-11-15:
Hello, 

Thank you for your work on this project. It's amazing! I was able to make previous commit build and run flawlessly. Current version doesn't build for some reason (idf 4.3.4). 

`
[ 99%] Built target __idf_websocket_if
[ 99%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/root/src/snapclient/snapclient/main/main.c: In function 'flac_task':
/root/src/snapclient/snapclient/main/main.c:510:34: error: 'scSet' is a pointer; did you mean to use '->'?
         dsp_setup_flow(500, scSet.sr, scSet.chkDur_ms);
                                  ^
                                  ->
/root/src/snapclient/snapclient/main/main.c:510:44: error: 'scSet' is a pointer; did you mean to use '->'?
         dsp_setup_flow(500, scSet.sr, scSet.chkDur_ms);
                                            ^
                                            ->
/root/src/snapclient/snapclient/main/main.c:511:23: error: 'pcm_chunk_message' undeclared (first use in this function); did you mean 'pcm_chunk_message_t'?
         dsp_processor(pcm_chunk_message.payload, pcm_chunk_message.size,
                       ^~~~~~~~~~~~~~~~~
                       pcm_chunk_message_t
/root/src/snapclient/snapclient/main/main.c:511:23: note: each undeclared identifier is reported only once for each function it appears in
/root/src/snapclient/snapclient/main/main.c: In function 'http_get_task':
/root/src/snapclient/snapclient/main/main.c:1409:47: error: 'pcm_chunk_message' undeclared (first use in this function); did you mean 'pcm_chunk_message_t'?
                                 dsp_processor(pcm_chunk_message.payload,
                                               ^~~~~~~~~~~~~~~~~
                                               pcm_chunk_message_t
/root/src/snapclient/snapclient/main/main.c:780:25: warning: variable 'endTime' set but not used [-Wunused-but-set-variable]
     uint64_t startTime, endTime;
                         ^~~~~~~
/root/src/snapclient/snapclient/main/main.c:780:14: warning: variable 'startTime' set but not used [-Wunused-but-set-variable]
     uint64_t startTime, endTime;
              ^~~~~~~~~
/root/src/snapclient/snapclient/main/main.c:2010:36: warning: this statement may fall through [-Wimplicit-fallthrough=]
                       internalState++;
                       ~~~~~~~~~~~~~^~
/root/src/snapclient/snapclient/main/main.c:2014:21: note: here
                     case 5: {
                     ^~~~
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:76: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:5572: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2
`

Sorry if this is something obvious.

CarlosDerSeher, 2022-11-21:
You could just disable DSP processor as I didn't test with CONFIG_USE_DSP_PROCESSOR set or do as suggested and replace `dsp_setup_flow(500, scSet.sr, scSet.chkDur_ms);`  with `dsp_setup_flow(500, scSet->sr, scSet->chkDur_ms);` which will make it right as I changed the type of scSet to a pointer.

CarlosDerSeher, 2022-11-21:
Feel free to issue a pull request if you can make it work doing these changes so I don't have to do them too :)

CarlosDerSeher, 2022-11-30:
Looks quite right. Please submit your pull request and I'll test this myself.

velaar, 2022-12-03:
@CarlosDerSeher  sorry for slow response. I was working out of country . Thank you for looking into this! Let me get back ti why I was trying to engage DSP because maybe I'm solving the wring problem with the wrong tools. 

I've boticed that the sound output quality is horrible for Lyrat v 4.3 board that I'm using. The format doesn't really matter tried PCM/FLAC with 48000:16:2. Running a desktop client on a cheap laptop with same settings produces a way better output.

CarlosDerSeher, 2022-12-03:
@velaar what is your buffer setting on the server? Maximum should be 500ms because if you go higher there will be dropped chunks which will be audible.

@luar123 thanks, but I didn't have time to test yet. Will do soon though

CarlosDerSeher, 2022-12-04:
Builds fine now but another issue addressed in #12 persists.

PR #11 (2022-12-04): Fix pointers when DSP processor is used
Fix scSet and pcmData pointers in CONFIG_USE_DSP_PROCESSOR sections.
Fixes #10
closed 2022-12-04 by CarlosDerSeher