CarlosDerSeher/snapclient #14: Player error while client is muted (NETCONN branch)

luar123, 2022-12-04:
When I mute the client I get `E (99720) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 2419048s, 850.862ms` once a second. It is still working fine, can be unmuted, and will resume. So not sure if that is a problem or not, but maybe the player and decoder tasks could be stopped while it is muted?

CarlosDerSeher, 2022-12-05:
This is intended behavior. Stopping the task could be done but I didn't have the muse yet also it isn't really necessary. Only thing I can think of is that it is better if power saving modes are activated

velaar, 2022-12-05:
Should we include a reference snapserver.conf somewhere? 
I'm not sure how well it is tuned to ESP32 but seems to work well for me:

```
chunk_ms = 24
codec = flac
sampleformat = 48000:16:2
buffer = 408
```

CarlosDerSeher, 2022-12-06:
I guess this issue is closed then? if so, please close.

closed 2022-12-06 by luar123