CarlosDerSeher/snapclient #15: LyraT 4.3 NETCON-Branch - broken Audio and continous resyncing

gh7531, 2022-12-05:
Hi,

I have an issue with the snapclient on my LyraT v4.3. I build the firmware using the NETCON-Branch and changed only my wifi and snapcast server IP using the menuconfig. 
Flashing and connecting to the snapcast server works, but the playback using librespot as input to snapcast is a horrible stuttery noise. Other clients like my phone work just fine with my librespot / snapcast setup.

The output of the LyraT throws a ton of `E (250821) PLAYER: Failed to allocate IRAM memory for pcm chunk payload`  and `E (250821) PLAYER: Failed to allocate IRAM memory for pcm chunk payload` warnings / errors.

```
E (249040) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
I (249057) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249057) PLAYER: RESYNCING HARD 2: age 25197us, latency 592103734159us, free 37932, largest block 13824, 34, rssi: -75
I (249161) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249162) PLAYER: RESYNCING HARD 2: age 25305us, latency 592103734159us, free 41456, largest block 18432, 33, rssi: -74
I (249266) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249266) PLAYER: RESYNCING HARD 2: age 25239us, latency 592103734159us, free 37928, largest block 12544, 34, rssi: -73
I (249370) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249371) PLAYER: RESYNCING HARD 2: age 25229us, latency 592103734137us, free 37932, largest block 13824, 34, rssi: -74
I (249475) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249475) PLAYER: RESYNCING HARD 2: age 25271us, latency 592103734137us, free 37944, largest block 18432, 34, rssi: -73
I (249579) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249580) PLAYER: RESYNCING HARD 2: age 25189us, latency 592103734137us, free 37944, largest block 18432, 34, rssi: -72
I (249684) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249684) PLAYER: RESYNCING HARD 2: age 25203us, latency 592103734137us, free 37924, largest block 9216, 34, rssi: -74
E (249721) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
E (249762) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
E (249784) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
I (249788) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249789) PLAYER: RESYNCING HARD 2: age 25248us, latency 592103734137us, free 37928, largest block 13824, 34, rssi: -73
E (249883) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
I (249893) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249893) PLAYER: RESYNCING HARD 2: age 25267us, latency 592103734137us, free 37936, largest block 13824, 34, rssi: -74
I (249997) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (249998) PLAYER: RESYNCING HARD 2: age 25242us, latency 592103734137us, free 37928, largest block 11520, 34, rssi: -68
E (250061) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
E (250099) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
I (250102) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (250103) PLAYER: RESYNCING HARD 2: age 25665us, latency 592103734137us, free 37932, largest block 9216, 34, rssi: -69
I (250206) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (250207) PLAYER: RESYNCING HARD 2: age 25266us, latency 592103734137us, free 37936, largest block 12544, 34, rssi: -70
E (250302) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
I (250311) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (250311) PLAYER: RESYNCING HARD 2: age 24591us, latency 592103733475us, free 37924, largest block 9216, 34, rssi: -72
I (250416) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (250416) PLAYER: RESYNCING HARD 2: age 25257us, latency 592103733475us, free 37916, largest block 18432, 34, rssi: -71
I (250520) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (250521) PLAYER: RESYNCING HARD 2: age 25235us, latency 592103733475us, free 37888, largest block 18432, 34, rssi: -72
I (250625) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (250625) PLAYER: RESYNCING HARD 2: age 25249us, latency 592103733475us, free 36156, largest block 13824, 34, rssi: -73
E (250705) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
E (250718) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
I (250729) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (250730) PLAYER: RESYNCING HARD 2: age 25244us, latency 592103733475us, free 37832, largest block 13824, 34, rssi: -76
E (250821) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
I (250834) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (250834) PLAYER: RESYNCING HARD 2: age 25268us, latency 592103733475us, free 37824, largest block 13824, 34, rssi: -74
E (250919) PLAYER: Failed to allocate IRAM memory for pcm chunk payload
```

This is another output right after reseting the LyraT and starting the playback on spotify

```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7660
ho 0 tail 12 room 4
load:0x40078000,len:15748
load:0x40080400,len:4392
0x40080400: _init at ??:?
[... 177 lines cut ...]
I (13363) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (13363) PLAYER: RESYNCING HARD 2: age 25270us, latency 592987098227us, free 33192, largest block 9216, 34, rssi: -74
I (13468) PLAYER: initial sync age: 2us, chunk duration: 26000us
W (13468) PLAYER: RESYNCING HARD 2: age 25251us, latency 592987098227us, free 28560, largest block 15360, 35, rssi: -77
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4000c2e0  PS      : 0x00060830  A0      : 0x800e8b78  A1      : 0x3ffc0040  
A2      : 0x3ffbdc98  A3      : 0x0000090c  A4      : 0x000008f4  A5      : 0x3ffbdc98  
A6      : 0x00000000  A7      : 0x0000008f  A8      : 0x00000000  A9      : 0x3ffbfff0  
A10     : 0x3ffb4040  A11     : 0x3ffbb3a4  A12     : 0x3ffbf80c  A13     : 0x00000000  
A14     : 0xfffffffd  A15     : 0x00000002  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000090c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x0000008e  

Backtrace:0x4000c2dd:0x3ffc0040 0x400e8b75:0x3ffc0050 0x400de98a:0x3ffc0090
0x400e8b75: i2s_custom_write at /home/jakob/develop/git/snapclient-CarlosDerSeher/components/custom_driver/i2s.c:1149 (discriminator 15)

0x400de98a: player_task at /home/jakob/develop/git/snapclient-CarlosDerSeher/components/lightsnapcast/player.c:1189

ELF file SHA256: 76d602c9e1dc9a00

Rebooting...
```

Any idea what is going on here?

CarlosDerSeher, 2022-12-05:
@gh7531 maximum supported buffer length is around 500ms if psram is disabled. You could enable psram from menuconfig which should work but I didn't test for a long time because my aim was always to get it to work without psram

velaar, 2022-12-05:
(previously posted under the wrong issue) 

Should we include a reference snapserver.conf somewhere? 
I'm not sure how well it is tuned to ESP32 but seems to work well for me:

```
chunk_ms = 24
codec = flac
sampleformat = 48000:16:2
buffer = 408
```

CarlosDerSeher, 2022-12-06:
@velaar I didn't see that sample rate was set to 44.1kHz. Probably we should change snapserver.conf located in `snapclient/snapserver.conf`. Maybe also the folder should be renamed and `README.MD` edited to make it more clear there is the reference config file for the server. I really want to merger NETCONN into master too but didn't find the time to do it yet.

@gh7531 Please use 48000:16:2 and flac (as suggested by @velaar) on snapserver for now as all other settings are either not implemented or hardly tested.

gh7531, 2022-12-06:
Thank you both for your help.

The sample rate of 48000 does indeed work. My issue now is that I can't get my source for the snapcast server (in my case librespot) to output at this sample rate properly. But this is not related to your project. 

Thank you for your time working on this project

velaar, 2022-12-08:
> checkout https://github.com/badaix/snapcast/blob/master/doc/configuration.md and jump to the Alsa section. I guess you'll find your solution there

Sadly no solution there. They suggest to use snd-aloop but it doesn't do any resampling. I tried to resample it with this;
```
pcm.!snapcast {
    type plug
    slave.pcm hw:1,1,0
    slave {
        pcm rate48000Hz # Direct default output to the below converter
    }
}

pcm.rate48000Hz {
     type rate
     slave {
          pcm writeFile # Direct to the plugin which will write to a file
          format S16_LE
          rate 48000
     }
}

pcm.writeFile {
    type file
    slave.pcm null
    file "/tmp/snapfifo48"
    format "raw"
}
```

But shairport-sync refuses to play into snapcast output, It might help @gh7531 though.

Ideally would love to learn how to resample a pipe into a pipe on Linux but that is definitely not related to project. Fixing 44100 on ESP32 might be nice at some point as well. Sadly, I'm far from your level of programming so can only offer testing. Presently running the latest pull request on my hardware.

CarlosDerSeher, 2022-12-08:
Maybe this can be of help?
https://github.com/badaix/snapcast/blob/master/doc/player_setup.md
Also see the alsa section

CarlosDerSeher, 2022-12-09:
> Ideally would love to learn how to resample a pipe into a pipe on Linux but that is definitely not related to project.

Probably cpiped from the line in example in combination with the alsa example can achieve what you want? Let shairport do its playback at 44.1khz and capture with cpiped at 48khz from the loopback device.

I will definitly try to fix the samplerate problem at some point but at the moment i am designing a custom PCB which will run snapclient so i can spread it around the house. Time is scarce with a two year old at home so this can take a while :)

Great to have someone else to test and find my errors.

CarlosDerSeher, 2022-12-10:
Ok, just tested with sample rate 44.1kHz and found there is a problem with chunk duration detection in `http_get_task()` around line 1387.
`size_t decodedSize =  pcmData->totalSize;  // pFlacData->bytes;
 scSet.chkDur_ms = (1000UL * decodedSize) / (uint32_t)(scSet.ch * (scSet.bits / 8)) / scSet.sr;`
decodedSize results in 4608 bytes which are 1152 frames using `44100:16:2` setting. When calculating `scSet.chkDur_ms` this results in a fractional number (26.122ms) while `scSet.chkDur_ms` is an `uint32_t` from there on all other calculations using `scSet.chkDur_ms` will be wrong and we will get memory allocation/access errors at some point.

I am currently not sure how to handle that though.

CarlosDerSeher, 2022-12-10:
found that `dsp_processor()` causes kernel panic with sample rate set to 44.1kHz and if disabled "only" syncing isn't working because of above reason.

luar123, 2022-12-10:
> found that `dsp_processor()` causes kernel panic with sample rate set to 44.1kHz and if disabled "only" syncing isn't working because of above reason.

I did not test, but stereo flow should work even with soft volume enabled. The other flows use the buffer that is created with `dsp_setup_flow(500, scSet->sr, scSet->chkDur_ms)`, so probably the buffer is too small.

Could you simply change scSet.chkDur_ms to float and use ceil() for memory allocation?

closed 2022-12-14 by CarlosDerSeher