CarlosDerSeher/snapclient #17: Feature: Suggested snapserver.conf file

velaar, 2022-12-13:
Hi, 

I would like to create this issue to track the suggested workarounds and limitations specifically on **snapserver.conf** side. 
Please find below the current setup I'm using but as per discussion in pr#16 my buffer of 400 ms is too lowcausing resyncs and audible clicks on resyncing. 

Here is my current config:
```
source = meta:///AirPlay/MPD?name=Main
source = pipe:///tmp/shairport-sync-audio?name=AirPlay
source = pipe:///tmp/snapfifo?name=MPD

chunk_ms = 26
codec = flac
sampleformat = 44100:16:2
buffer = 400
```

> The Zero just before rssi value means there are no chunks available. So maybe your wifi connection is bad sometimes and you would need a higher buffer time in your snapserver.conf. I found that 500ms could be too low for my setup very often.

Presently testing with 600ms buffer. Had one unexplained freeze but didn't capture the log so not sure.
What will be an estimated max / reasonable buffer for ESP32-WROVER-E? I might be off here but I think I read somewhere that 500ms is max that fits reliably. 

Also how does the DSP buffer affect max buffer available for snapcast?

CarlosDerSeher, 2022-12-13:
So at 48000:16:2 an 1000ms buffer length you will need 192000 Bytes of RAM just for the PCM data. If you'll activate DSP processer you will need even more please look for yourself how much that is as I can't remember. All that wifi an LWIP and freertos stuff will consume RAM too. The datasheet states you have 520KB of internal RAM available. Just the PCM buffering part will consume about 1/3 of that. I thought I could squeeze it in but found it is impossible to do without static RAM enabled.
So I reduced buffer length to around 500ms to get it a 100% stable. Hardly because now I have these occasional hiccups because of late wifi packets. I played a lot with wifi buffers and LWIP buffer sizes and different settings to optimize throughput but couldn't find a 100% satisfying solution. 
The highest buffer setting where I didn't get many memory allocation errors was around 720ms I believe, but these were never fully used because of wifi latency and some chunks being placed in the DMA buffer. 

I think we could do better, because when memory allocation errors happen there is still some free RAM but not a large enough block available. That's the reason for the "fragment" part in the player code. I wasn't able to get this stable yet though. Maybe we should generally allocate the chunks in smaller fragments of say 128 Bytes and link them together. I started to implement this but as I said I never finished it.

luar123, 2022-12-13:
@velaar If you don't use the bassboost or bi-amp dsp flows and don't need the software volume mixer, there is no benefit in using the dac. So just disable it and safe some memory.

CarlosDerSeher, 2022-12-14:
@velaar as to your question 

> Also how does the DSP buffer affect max buffer available for snapcast?

After addressing #15 I found if dsp processer is enabled the needed buffers will consume as much RAM as 3 chunks of PCM audio would have. For 48000:16:2, flac, 24ms this is 13824 bytes.

CarlosDerSeher, 2022-12-14:
Feel free to reopen but I consider this done now

CarlosDerSeher, 2022-12-15:
Ok, so I found this issue could be probably be left open and useful for discussion

CarlosDerSeher, 2022-12-15:
@velaar @luar123 
Most recent commit 430cd080f31cb0751077ad47659c6a87eded8321 got improved buffering. With my setup I can do around 900ms buffer on the small WROOM modules now. Also I updated the snapserver.conf

luar123, 2022-12-15:
@CarlosDerSeher Thanks for the update, working great so far.
Some feedback:
With 48000:16:2, DSP active and chunk duration=24ms, 700ms buffer is working fine, 800ms still sounds good but gives a lot of `PLAYER: Failed to heap_caps_malloc(4608, 3)` messages and 900ms is the same, but when booting while music is already playing I get `Core  0 panic'ed (LoadProhibited)` in dsp_processor(). So I will go with 700ms now. (small WROOM module)

And a small bug: main.c:117 should not be a commend

CarlosDerSeher, 2022-12-15:
@luar123 oh i see was trying to do some heap analysis and left that commented by accident.

I am running at 44100:16:2 because you can squeeze out a little more buffer time that way.

CarlosDerSeher, 2022-12-16:
@luar123 oh i see was trying to do some heap analysis and left that commented by accident.

I am running at 44100:16:2 because you can squeeze out a little more buffer time that way.

@velaar @luar123
Just as a side note I finally found the time to merge NETCONN back into master.