CarlosDerSeher/snapclient #19: Compiler Error

Medel, 2022-12-26:
If I try to compile I get following errors:

`...
[ 53%] Building C object esp-idf/audio_sal/CMakeFiles/__idf_audio_sal.dir/audio_sys.c.obj
In file included from /home/mario/esp/snapclient/components/custom_driver/i2s.c:28:
/home/mario/esp/snapclient/components/custom_driver/include/i2s.h:66:30: error: unknown type name 'i2s_port_t'; did you mean 'in_port_t'?
 esp_err_t i2s_custom_set_pin(i2s_port_t i2s_num, const i2s_pin_config_t *pin);
                              ^~~~~~~~~~
                              in_port_t
/home/mario/esp/snapclient/components/custom_driver/include/i2s.h:66:56: error: unknown type name 'i2s_pin_config_t'
 esp_err_t i2s_custom_set_pin(i2s_port_t i2s_num, const i2s_pin_config_t *pin);
                                                        ^~~~~~~~~~~~~~~~
/home/mario/esp/snapclient/components/custom_driver/include/i2s.h:126:37: error: unknown type name 'i2s_port_t'; did you mean 'in_port_t'?
 esp_err_t i2s_custom_driver_install(i2s_port_t i2s_num,
                                     ^~~~~~~~~~
                                     in_port_t
/home/mario/esp/snapclient/components/custom_driver/include/i2s.h:127:43: error: unknown type name 'i2s_config_t'
                                     const i2s_config_t *i2s_config,
                                           ^~~~~~~~~~~~
/home/mario/esp/snapclient/components/custom_driver/include/i2s.h:139:39: error: unknown type name 'i2s_port_t'; did you mean 'in_port_t'?
 esp_err_t i2s_custom_driver_uninstall(i2s_port_t i2s_num);
                                       ^~~~~~~~~~
                                       in_port_t

...

/home/mario/esp/snapclient/components/custom_driver/i2s.c:1248:27: error: unknown type name 'i2s_port_t'; did you mean 'in_port_t'?
 esp_err_t i2s_custom_read(i2s_port_t i2s_num, void *dest, size_t size,
                           ^~~~~~~~~~
                           in_port_t
/home/mario/esp/snapclient/components/custom_driver/i2s.c:581:23: warning: 'i2s_intr_handler_default' defined but not used [-Wunused-function]
 static void IRAM_ATTR i2s_intr_handler_default(void *arg) {
                       ^~~~~~~~~~~~~~~~~~~~~~~~
/home/mario/esp/snapclient/components/custom_driver/i2s.c:109:21: warning: 'i2s_spinlock' defined but not used [-Wunused-variable]
 static portMUX_TYPE i2s_spinlock[I2S_NUM_MAX];
                     ^~~~~~~~~~~~
/home/mario/esp/snapclient/components/custom_driver/i2s.c:107:19: warning: 'p_i2s_obj' defined but not used [-Wunused-variable]
 static i2s_obj_t *p_i2s_obj[I2S_NUM_MAX] = {0};
                   ^~~~~~~~~
make[2]: *** [esp-idf/custom_driver/CMakeFiles/__idf_custom_driver.dir/build.make:76: esp-idf/custom_driver/CMakeFiles/__idf_custom_driver.dir/i2s.c.obj] Fehler 1
make[2]: Verzeichnis „/home/mario/esp/snapclient/build“ wird verlassen
make[1]: *** [CMakeFiles/Makefile2:5489: esp-idf/custom_driver/CMakeFiles/__idf_custom_driver.dir/all] Fehler 2
make[1]: Verzeichnis „/home/mario/esp/snapclient/build“ wird verlassen
make: *** [Makefile:136: all] Fehler 2
make failed with exit code 2
`
ESP-IDF v4.4.2-155-g23d5a582cb

velaar, 2022-12-26:
Please use IDF 4.3.x. IDF 4.4 is not supported at the moment.

Medel, 2022-12-26:
After a complete new install without esp-adf it compiles. 
Install like this:

`mkdir ~/esp
cd esp
git clone https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout v4.3.1
./install.sh
. ./export.sh
cd ~/esp
git clone https://github.com/CarlosDerSeher/snapclient
cd snapclient
git submodule update --init
idf.py menuconfig

> Set Board, Wifi, Name and Pins -> Save, Quit

>  snapserver.conf change codec to opus

idf.py build
idf.py flash -p /dev/ttyACM0 monitor
`
**But the codec does not work. After the wifi connection it reboots:**

`I (1643) WIFI: Connected with IP Address:192.168.1.83
I (1644) esp_netif_handlers: sta ip: 192.168.1.83, mask: 255.255.255.0, gw: 192.168.1.2
I (1647) WIFI: connected to ap SSID: xxxxxx
I (1652) SC: Connected to AP
I (1655) NETF: Setup mdns
I (1670) SC: Lookup snapcast service on network
**E (1671) esp_ota_ops: Running firmware is factory**
I (1671) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (4760) SC: Found 192.168.1.31:1704
I (4774) SC: netconn connected
I (4776) SC: netconn sent hello message
I (4783) SC: Buffer length:  1000
I (4783) SC: Latency:        0
I (4783) SC: Mute:           0
I (4785) SC: Setting volume: 100
**I (4789) SC: OPUS not implemented yet**
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d872c: b39ee51a 1df4ce46 000000f0
0x400d872c: http_get_task at /home/mario/esp/snapclient/main/main.c:723

Core  1 register dump:
PC      : 0x400d8733  PS      : 0x00060030  A0      : 0x00000000  A1      : 0x3ffcd620  
0x400d8733: http_get_task at /home/mario/esp/snapclient/main/main.c:2452

A2      : 0x00000004  A3      : 0x00000000  A4      : 0x3ffb4484  A5      : 0x3ffb8a78  
A6      : 0x3ffb4488  A7      : 0x3f402620  A8      : 0x800d6c99  A9      : 0x3ffcd5d0  
A10     : 0x00000003  A11     : 0x3f402620  A12     : 0x3f402df0  A13     : 0x000012b5  
A14     : 0x3f402620  A15     : 0x3ffcd5e0  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace:0x400d8730:0x3ffcd620
0x400d8730: http_get_task at /home/mario/esp/snapclient/main/main.c:726

ELF file SHA256: 450b52b379876b4f

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7336
load:0x40078000,len:15112
load:0x40080400,len:4416
0x40080400: _init at ??:?

entry 0x400806a0
`

velaar, 2022-12-26:
Please see the error here:  > 4789) SC: OPUS not implemented yetIt is a configuration error on server side.

Medel, 2022-12-26:
Server side is OK.
The Android tablet play and the esp client appears and disappears when it reboots.
Also the linux client. Here is the output:

`2022-12-26 18-42-43.400 [Info] (Snapclient) Version 0.25.0
2022-12-26 18-42-43.400 [Info] (Connection) Resolving host IP for: 192.168.1.31
2022-12-26 18-42-43.400 [Info] (Connection) Connecting
2022-12-26 18-42-43.401 [Notice] (Connection) Connected to 192.168.1.31
2022-12-26 18-42-43.401 [Info] (Connection) My MAC: "6c:f0:49:78:62:2e", socket: 8
2022-12-26 18-42-43.461 [Info] (Controller) ServerSettings - buffer: 1000, latency: 0, volume: 100, muted: 0
2022-12-26 18-42-43.462 [Info] (Controller) Codec: opus, sampleformat: 48000:16:2
2022-12-26 18-42-43.462 [Info] (Player) Player name: alsa, device: default, description: Default ALSA Output (currently PulseAudio Sound Server), idx: 4, sharing mode: unspecified, parameters: <none>
2022-12-26 18-42-43.462 [Info] (Player) Mixer mode: software, parameters: <none>
2022-12-26 18-42-43.462 [Info] (Player) Sampleformat: 48000:16:2, stream: 48000:16:2
2022-12-26 18-42-43.462 [Info] (Alsa) Using default buffer_time: 80 ms, default fragments: 4
2022-12-26 18-42-43.516 [Info] (Alsa) PCM name: default, sample rate: 48000 Hz, channels: 2, buffer time: 80000 us, periods: 4, period time: 20000 us, period frames: 960
2022-12-26 18-42-43.520 [Info] (Stream) No chunks available
2022-12-26 18-42-43.585 [Info] (Controller) diff to server [ms]: 8.50515e+09
`

velaar, 2022-12-26:
At the moment OPUS is not supported. Please use flac container at 44,100 sample rate.

Medel, 2022-12-26:
I changed in the snapserver.conf sampleformat and codec to
sampleformat = 44100:16:2
codec = flac
also the snapcast server
the linux client shows these settings the Tablet plays too
but there is always something wrong, see attached file
Is there a setting in menuconfig to change?
The ESP has an spi-flash with 16384k
[Error.txt](https://github.com/CarlosDerSeher/snapclient/files/10304609/Error.txt)

velaar, 2022-12-26:
Which board are you using? Haven’t had this issue on LyRaT v4.3 and have multiple running. Any chance you have a buffer size over 800ms?

Medel, 2022-12-26:
ESP32 Wrover lILYGO T7_V1.5 with PCM5102A, runs on two other boards with a older version.

Medel, 2022-12-26:
Is this right (1ULL)?

in pcm5102a.c
esp_err_t pcm5102a_init(audio_hal_codec_config_t *codec_cfg) {
...
io_conf.pin_bit_mask = (**1ULL** << CONFIG_PCM5102A_MUTE_PIN);

CarlosDerSeher, 2022-12-26:
Please ensure you set a buffer size below 800ms on the server. Yours is 1000ms. Try with 500ms and work your way up to see what's possible with your setup. If setting to 500ms doesn't remove the stack overflow error find the line where flac_decoder_task is created and add 256bytes of stack.

luar123, 2022-12-26:
I would try to run `idf.py erase_flash` before flashing the firmware.
(Using pcm5102a, but with a simple Wroom here)

CarlosDerSeher, 2022-12-26:
Also if you have a wrover module you could try to enable spi ram in menuconfig. Then you could set buffers bigger than 800ms

Medel, 2022-12-26:
After erasing flash and new flashing it does not connect to my wifi. I will take look why in the next days.

luar123, 2022-12-27:
Right, I had the same problem in the beginning. I could solve it by trying different settings, like WiFi provisioning and wrong ssid or password. At some point it worked again. In this case I was always flashing without erase.

Medel, 2022-12-27:
Wifi connects again.
I set the buffer to 400, sampleformat to 44100:16:2, change the menu setting to: Support for external, SPI-connected RAM.
But always errors, see atached file
[Error2.txt](https://github.com/CarlosDerSeher/snapclient/files/10310023/Error2.txt)

CarlosDerSeher, 2022-12-28:
> Right, I had the same problem in the beginning. I could solve it by trying different settings, like WiFi provisioning and wrong ssid or password. At some point it worked again. In this case I was always flashing without erase.

So if this is an actual problem, please open another issue regarding that. I never had a problem with wifi connection but didin't erase ssid and key for a long time..

CarlosDerSeher, 2022-12-28:
> Wifi connects again.
> I set the buffer to 400, sampleformat to 44100:16:2, change the menu setting to: Support for external, SPI-connected RAM.
> But always errors, see attached file
> [Error2.txt](https://github.com/CarlosDerSeher/snapclient/files/10310023/Error2.txt)
> 

Ok after checking the log and the relevant part of the code it seems there is an issue with the mutex responsible for syncing flac decoder and http task. I know this part should be handled more elegant is is kind of messy. Not sure why this error doesn't happen for me. I'll have to look into that which could take a while.
A quick and dirty solution to make the error go away for now could be to disable "check that owner task gives mutex" (or similar) from menuconfig. I think you find this setting within freertos configuration.

Medel, 2022-12-28:
I disable in menuconfig: Check that mutex semaphore is given by owner task
but then there is a different error.
[Error3.txt](https://github.com/CarlosDerSeher/snapclient/files/10315244/Error3.txt)

CarlosDerSeher, 2022-12-28:
@Medel ok, please try d9c19e744ecdbb551de37f2cc04a1c07e550d789 where I removed these mutex completely and flac data is handled a little different in a dedicated task. Hopefully this will resolve your issue. It's working for me without any problems currently, though it's not thoroughly tested yet.

Medel, 2022-12-28:
It works!
Here's the log
[Log.txt](https://github.com/CarlosDerSeher/snapclient/files/10315594/Log.txt)

CarlosDerSeher, 2022-12-28:
great :) so I'll be closing this issue

closed 2022-12-28 by CarlosDerSeher

Medel, 2022-12-28:
For my first try i used my old menuconfig but there are changes in the new so I test it with the new one and there is a compile error:
`At top level:
/home/mario/esp/snapclient/main/main.c:303:19: warning: 'flacOutData' defined but not used [-Wunused-variable]
 static flacData_t flacOutData;
                   ^~~~~~~~~~~
/home/mario/esp/snapclient/main/main.c:145:13: warning: 'time_message_serialized' defined but not used [-Wunused-variable]
 static char time_message_serialized[TIME_MESSAGE_SIZE];
             ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/mario/esp/snapclient/main/main.c:12:
/home/mario/esp/snapclient/main/main.c: In function 'http_get_task':
/home/mario/esp/esp-idf/components/log/include/esp_log.h:338:47: error: 'tmp' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         else                                { esp_log_write(ESP_LOG_INFO,       tag, LOG_FORMAT(I, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                               ^~~~~~~~~~~~~
/home/mario/esp/snapclient/main/main.c:889:11: note: 'tmp' was declared here
     char *tmp;
           ^~~
cc1: some warnings being treated as errors
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:76: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:5473: esp-idf/main/CMakeFiles/__idf_main.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
make failed with exit code 2
`

Medel, 2022-12-28:
Sorry my mistake, renaming the sdconfig. A new one was created which produce this error.

CarlosDerSeher, 2022-12-28:
@Medel so no problem on your end?

Medel, 2022-12-28:
Yes this problem is solved.