CarlosDerSeher/snapclient #20: Error message if muted (PCM5102A)

Medel, 2022-12-28:
Tho log shows an Error if muted (PCM5102A)
`
W (2913328) PLAYER: RESYNCING HARD 2: age -285393us, latency 8727629802110us, free 4313326, largest block 2097152, 0, rssi: -69
I (2914317) PLAYER: initial sync age: 6us, chunk duration: 26122us
I (3366617) SC: Buffer length:  1000
I (3366617) SC: Latency:        0
I (3366617) SC: Mute:           1
I (3366628) SC: Setting volume: 100
W (3367539) PLAYER: RESYNCING HARD 2: age -12us, latency 8727629804095us, free 4313478, largest block 2097152, 0, rssi: 
E (3369540) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 8727629s, 804.95ms
E (3371540) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 8727629s, 804.95ms
E (3373540) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 8727629s, 804.95ms
E (3375540) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 8727629s, 804.95ms
E (3377540) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 8727629s, 804.95ms
E (3379540) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 8727629s, 804.95ms
I (3379858) SC: Buffer length:  1000
I (3379859) SC: Latency:        0
I (3379859) SC: Mute:           0
I (3379859) SC: Setting volume: 100
I (3380862) PLAYER: initial sync age: 2us, chunk duration: 26122us
`

luar123, 2022-12-28:
Nothing to worry about, see #14

Medel, 2022-12-28:
OK

closed 2022-12-28 by Medel