CarlosDerSeher/snapclient #22: Occasional "clicks" during playback

CarlosDerSeher, 2023-01-06:
Not sure if this is related to snapclient firmware itself or related to my hardware/audio setup. The log won't show any hints but I can hear clicks during playback from time to time and was wondering if somebody else is having such issues too? I really can't pin it down yet but I have the feeling this happens more often during low volume playback.

velaar, 2023-01-11:
I can confirm the clicks. While the latest updates with SPI RAM works great overall the clicks are happening from time-to-time at what seems like irregular interval. Happened with buffer = 2000 and buffer = 600 so unlikely to be buffer-dependent. 

Not much I can help with here but will monitor further.

CarlosDerSeher, 2023-01-12:
You are using Lyrat v4.3 too don't you? Could this be a hardware issue (I doubt it though)? Maybe someone else with different hardware setup can confirm this?
Are these clicks happening to somebody else too?
@majoreffort

73f14nd, 2023-01-13:
I can confirm the clicks on a custom board MAX98357A.

I tried to stream a sinus wave, you can hear the first click after ~6sec followed by a lot of more clicks later.
The click pattern seems to be identical after I start to fill the /snapfifo on the server.

Using pcm instead of flac as transport codec it gets worse!

CarlosDerSeher, 2023-01-13:
@73f14nd so you are saying using a sine wave as a signal these clicks reveal itself even more often? Which interval we are talking about here? Do you have the possibility to record a click and post the data/pictures here? Maybe we can pin the problem down that way. Resolving that is really important to me as this is kind of annoying on daily use.

velaar, 2023-01-13:
@73f14nd  Also, can you also test if timing of the clicks is affected by buffer side? I seem to have weird quality issues associated with it but don’t have enough data to make a coherent description other than at some buffer sizes the sound becomes “flat”? As if the bass cuts off. But again, I’ll try to collect better data before reporting.

CarlosDerSeher, 2023-01-14:
Ok I think I got it captured. Don't have more time at the moment to further look into this. I still have to run a test with dsp_processor disabled. I can't imagine where this is coming from. At first I thought about some lost data from http_get_task() but seeing this is somehow confusing. Why does the signal get attenuated?

[image]

Here is the audacity project with the raw data:
[click.zip](https://github.com/CarlosDerSeher/snapclient/files/10417197/click.zip)

73f14nd, 2023-01-14:
I think could now identify a problematic piece of code in `components/lightsnapcast/player.c`

The time synchronization control loop sets the direction [-1|0|1] of time adjustment.
Every time when dir changes I can her a click.

If I disable the call to `rtc_clk_apll_enable(..);` there are no clicks
(But than the control loop does not work as designed and after some time a hard resync happens.)

Perhaps the reconfiguration of i2s timings while playing is questionable.

CarlosDerSeher, 2023-01-14:
But why doesn't his happen all the time then? dir changes a lot while playback...

73f14nd, 2023-01-14:
Switched now my esp-idf from 4.3.4 to 4.3.1 as written in the README.

With version 4.3.4 I could her a click on every `dir` change, 
Version 4.3.1 sounds much better in the sine wave test, only one sporadic click per minute,

CarlosDerSeher, 2023-01-14:
> Version 4.3.1 sounds much better in the sine wave test, only one sporadic click per minute, 

How about some more data? Please connect a aux cable to your computer's microphone input and record those clicks and also append the esp32 log which shows dir changes.

CarlosDerSeher, 2023-01-15:
@73f14nd I can't really reproduce the behavior you are describing. Are you sure your buffer is set to a low enough value from server side? Assuming you don't have PSRAM on your board this would also result in similar audible artifacts if memory allocation errors are happening. Because then the actual chunk is dropped and chunk full of zeros is inserted.

I've done some changes and hopefully the issue will be resolved with commit 016a131f3385124ed73dedb849f7c7a01c3f3abc. Still need more testing. Also I've started to add a http server where at some point all the dsp processor related stuff will be configurable while the system is running. For now only `dspfEQBassTreble` works and only gain of bass can be adjusted with the slider, but you see where this should go.

miLORD1337, 2023-01-20:
Do you still need additional data from another piece of hardware? Doesn't seem so. But if so, I could test with WROVER and PCM5102...

CarlosDerSeher, 2023-01-20:
@miLORD1337 no thanks. I finally finished my custom PCB design which uses ADAU1961 as a DAC giving me left and right channel balanced line outputs with an additional headphone out. Additionally it got an LAN8720A for Ethernet. Works quite well, so I now have another platform to test too. As I already captured the click myself I think I am OK at this end.
I still can't reproduce the behavior described by 73f14nd. I get the occasional clicks which look like on the picture I appended above. Still no idea where these are coming from and how this is even possible... Maybe you have a theory? Are you experiencing these clicks too and are you sure they are not resyncing events?

luar123, 2023-01-22:
@CarlosDerSeher How often do the clicks happen for you? I have not noticed any, but was not using esp-snapclient a lot in the past weeks. Also I am still using 48kHz, maybe that is a difference?

CarlosDerSeher, 2023-01-22:
@luar123 sporadically. Maybe once an hour. Or maybe once in 2h, can't really tell as I didn't use it for very long durations either lately

majoreffort, 2023-01-23:
> You are using Lyrat v4.3 too don't you? Could this be a hardware issue (I doubt it though)? Maybe someone else with different hardware setup can confirm this? Are these clicks happening to somebody else too? @majoreffort @luar123

What I could just quickly confirm is that there is constant but ever so slightly noticeable clicking/noise when using **PCM**, as @73f14nd also stated. I suspect this is maybe unrelated to the issue you identified in the meantime, because with **FLAC** _I didn't notice anything_. 

Using PCM5102A, MAX98357A and another hardware with ES8388 (same codec as Lyrat 4.3). My setups with MAX98357A are clicking constantly, but this is a HW/connection issue in my prototype setups, nothing to do with the software.

Let me know if you need additional input on this issue, then I could do some more testing as well.

CarlosDerSeher, 2023-01-23:
@luar123 `Or run a linux client and record the output.`
Just now started that test using my notebook :)

@majoreffort thanks for your input this brings my hopes up the problem could be unrelated to esp snapclient. Hopefully the sinewave test using my notebook will reveal bluealsa as the troublemaker.

Regarding the clicking when using PCM, there the problem could be a wrong reordering of samples we got from NETCONN, before they are put in player queue. I don't really use PCM so it's hardly tested. Will try when there is time.

CarlosDerSeher, 2023-01-26:
Ok, using my custom built board I don't hear any clicks. Only thing happening sometimes are resync 2 events happening, even if connected through Ethernet. Same for the Linux client. 
I didn't expect that but I guess this is another issue.

closed 2023-01-26 by CarlosDerSeher