CarlosDerSeher/snapclient #24: Build fails with ethernet disabled

luar123, 2023-01-22:
Build fails with ethernet disabled:
```
.../esp/snapclient/components/eth_interface/eth_interface.c: In function 'eth_init':
.../esp/snapclient/components/eth_interface/eth_interface.c:100:25: error: 'CONFIG_SNAPCLIENT_ETH_PHY_ADDR' undeclared (first use in this function); did you mean 'CONFIG_SNAPCLIENT_NAME'?
   phy_config.phy_addr = CONFIG_SNAPCLIENT_ETH_PHY_ADDR;
```
eth_interface.c should probably not be included at all.

CarlosDerSeher, 2023-01-22:
Yes I realized that already. Can this be excluded through cmake? I repaired that locally already but didn't have time to push yet

CarlosDerSeher, 2023-01-22:
Ok, found the answer myself.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#adding-conditional-configuration

commit 7fca2eeb (2023-01-23): fix #24
closed 2023-01-23 by CarlosDerSeher