CarlosDerSeher/snapclient #33: Opus decoding failure

escalator2015, 2023-05-03:
Hi, in  develop branch the opus_decoder_task seems to have a mistake (at least for 16 bit samples) when data is decoded and transfered to audio output buffer, there is no need to convert data as the decoder already  gives 16 bits samples ready to play.
So a single line of code does the job:    
  
      memcpy((char *)pcmData->fragment->payload, (char *)audio, bytes);`

I am not compiling dsp processor code and the full path is 16 bits from server. It works flawessly in opus mode this way.
You would remove or comment the following as is not correct for 16 bits:

```
if (pcmData->fragment->payload) {
              volatile uint32_t *sample;
              uint32_t tmpData;
              uint32_t cnt = 0;

              for (int i = 0; i < bytes; i += 4) {
                sample =
                    (volatile uint32_t *)(&(pcmData->fragment->payload[i]));
                tmpData = (((uint32_t)audio[cnt] << 16) & 0xFFFF0000) |
                          (((uint32_t)audio[i + 1] << 0) & 0x0000FFFF);
                *sample = (volatile uint32_t)tmpData;

                cnt += 2;
              }
            }
```

CarlosDerSeher, 2023-05-03:
You need this conversion, as pcm chunks could be allocated in IRAM or DRAM. If memory is located in IRAM you will get a kernel panic if you do 8 bit writes.

escalator2015, 2023-05-03:
I understand, you are using internal ram and I am using wrover module with external spi ram. So my code only works for spi ram allocation.

closed 2023-05-03 by escalator2015