CarlosDerSeher/snapclient #35: Latest NETCONN branch, frequent resyncs without apparent reason

velaar, 2023-09-02:
Log is below (some repeats are omitted). 
Most worrisome are latency numbers, RSSI is always good. Doesn't seem to be a network issue at the server is on the same VLAN so no router is involved. 

Running at 44100:16:2 FLAC

```
I (191820) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (191820) PLAYER: got pcm chunk
I (191820) PLAYER: now: 190867400us
I (191824) PLAYER: diff: 467829949705us
W (191829) PLAYER: RESYNCING HARD 2: age -623135us, latency 467829949705us, free 4332690, largest block 4128768, 0, rssi: -41
W (191844) PLAYER: got pcm chunk
I (191844) PLAYER: now: 190891066us
I (191848) PLAYER: diff: 467829949705us
W (191852) c_I2S: wrote 2304 to desc[0]
W (191857) c_I2S: wrote 2304 to desc[1]
W (191862) c_I2S: wrote 2304 to desc[2]
W (191865) c_I2S: wrote 2304 to desc[3]
I (191947) PLAYER: (full) latency median: 467829949705us
I (192943) PLAYER: (full) latency median: 467829949708us
I (193000) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (193000) PLAYER: got pcm chunk
I (193000) PLAYER: now: 192047617us
I (193004) PLAYER: diff: 467829949708us
W (193183) PLAYER: got pcm chunk
I (193183) PLAYER: now: 192229701us
I (193183) PLAYER: diff: 467829949708us
W (193185) PLAYER: RESYNCING HARD 2: age -655081us, latency 467829949708us, free 4332690, largest block 4128768, 0, rssi: -41
I (193960) PLAYER: (full) latency median: 467829949736us
W (194911) PLAYER: got pcm chunk
I (194911) PLAYER: now: 193958529us
I (194911) PLAYER: diff: 467829949736us
W (194914) PLAYER: RESYNCING HARD 1: age 681939us, latency 467829949736us, free 4329514, largest block 4128768, 0, rssi: -42
I (194962) PLAYER: (full) latency median: 467829949782us
W (194975) PLAYER: got pcm chunk
I (194976) PLAYER: now: 194022728us
I (194976) PLAYER: diff: 467829949782us
W (194977) c_I2S: wrote 2304 to desc[0]
W (194981) c_I2S: wrote 2304 to desc[1]
W (194992) c_I2S: wrote 2304 to desc[2]
W (194992) c_I2S: wrote 2304 to desc[3]`
``

CarlosDerSeher, 2023-09-02:
Please use master or develop branch as i merged netconn to master some time ago.

velaar, 2023-09-02:
Thank you for a quick response. Rebuilt on current develop branch with SPI ram disabled getting this picture regardless of buffer 450...900. Board is still Lyrat4.3. 

With sdkconfig from develop branch and minimal changes (wifi, snapserver, disabled ethernet).

``` 
I (2605) SC: netconn connected
I (2609) SC: netconn sent hello message
I (2616) SC: Buffer length:  450
I (2616) SC: Latency:        0
I (2620) SC: Mute:           0
I (2624) SC: Setting volume: 100
I (2631) SC: fLaC sampleformat: 44100:16:2
I (2712) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 4
I (2713) c_I2S: DMA Malloc info, datalen=blocksize=2304, dma_buf_count=4
I (2725) c_I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (2730) LYRAT_V4_3: I2S0, MCLK output by GPIO0
I (2744) c_I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
W (2749) PLAYER: no pcm chunk queue created?
I (2754) PLAYER: created new queue with 16
I (2759) PLAYER: snapserver config changed, buffer 450ms, chunk 1152 frames, sample rate 44100, ch 2, bits 16 mute 0 latency 0
I (4626) SC: latency buffer full
I (5073) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (7630) PLAYER: RESYNCING HARD 2: age -294839us, latency 520678917639us, free 187680, largest block 110592, 2, rssi: -44
I (8005) PLAYER: initial sync age: 2us, chunk duration: 26122us

W (17695) PLAYER: RESYNCING HARD 2: age 4us, latency 520678917639us, free 202912, largest block 110592, 0, rssi: -43

W (18165) PLAYER: RESYNCING HARD 1: age 390917us, latency 520678917639us, free 202924, largest block 110592, rssi: -44
I (18635) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (43215) PLAYER: RESYNCING HARD 2: age 3us, latency 520678917681us, free 201340, largest block 110592, 0, rssi: -43
E (45216) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 520678s, 917.681ms
E (47217) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 520678s, 917.681ms
E (49218) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 520678s, 917.681ms
E (51219) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 520678s, 917.681ms
...
```

Intended configuration with SPIRAM enabled causes this set of logs:

```
I (3515) SC: netconn sent hello message
I (3524) SC: Buffer length:  450
I (3524) SC: Latency:        0
I (3525) SC: Mute:           0
I (3529) SC: Setting volume: 100
I (3536) SC: fLaC sampleformat: 44100:16:2
I (3604) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 4
I (3605) c_I2S: DMA Malloc info, datalen=blocksize=2304, dma_buf_count=4
I (3617) c_I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (3622) LYRAT_V4_3: I2S0, MCLK output by GPIO0
I (3636) c_I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
W (3641) PLAYER: no pcm chunk queue created?
I (3646) PLAYER: created new queue with 16
I (3651) PLAYER: snapserver config changed, buffer 450ms, chunk 1152 frames, sample rate 44100, ch 2, bits 16 mute 0 latency 0
I (5530) SC: latency buffer full
I (5952) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (24340) PLAYER: RESYNCING HARD 2: age -2us, latency 521039671206us, free 4334758, largest block 4128768, 0, rssi: -43
I (24419) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (24431) PLAYER: RESYNCING HARD 2: age -376385us, latency 521039671206us, free 4334758, largest block 4128768, 0, rssi: -43
I (24889) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (35282) PLAYER: RESYNCING HARD 2: age -9us, latency 521039671206us, free 4334758, largest block 4128768, 0, rssi: -43
E (37284) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 521039s, 671.206ms
E (43287) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 521039s, 671.206ms

```

Network as a possible source of latency is unlikely - did multiple ping tests
```
--- 172.20.22.8 ping statistics ---
198 packets transmitted, 198 received, 0% packet loss, time 197254ms
rtt min/avg/max/mdev = 1.978/74.339/194.658/33.132 ms
```

CarlosDerSeher, 2023-09-02:
Hmmm strange. I use three custom built board and didn't notice any audible hickups lately. Didn't look at the logs for quite some time now. When I have time I'll flash one of the lyrats and see if this also happens to me. Maybe I broke something with the latest commits... That you could test, maybe it will help to step back 2 or 3 commits. Or try using a different board but a hardware problem is unlikely, it's almost always a software problem from my experience :)

CarlosDerSeher, 2023-09-03:
I upgraded cod recently to IDF v4.3.5 maybe this could be a problem on your end?

CarlosDerSeher, 2023-09-03:
seems I didn't push these changes yet though.

Also I have it running with my lyrat board for a while now without any issues. Does this happen frequently to you? Maybe a server issue? I am running snap server v0.27 on openwrt.

velaar, 2023-09-03:
Sorry, this is a long one. Don't know if there is anything here that helps. Just trying to provide more info. 

> I upgraded code recently to IDF v4.3.5 maybe this could be a problem on your end?
I am running with IDF 4.3.5  but the only code upgrade was chip -> pkg.  So there might be something that I missed in the 4.3.4 -> 4.3.5 

```
diff --git a/components/custom_driver/i2s.c b/components/custom_driver/i2s.c
index e5d1c28..ac98873 100644
--- a/components/custom_driver/i2s.c
+++ b/components/custom_driver/i2s.c
@@ -186,7 +186,7 @@ static float i2s_apll_get_fi2s(int bits_per_sample, int sdm0, int sdm1,
 #if CONFIG_IDF_TARGET_ESP32
   /* ESP32 rev0 silicon issue for APLL range/accuracy, please see ESP32 ECO
    * document for more information on this */
-  if (esp_efuse_get_chip_ver() == 0) {
+  if (esp_efuse_get_pkg_ver() == 0) {
     sdm0 = 0;
     sdm1 = 0;
   }
```

```
--- a/components/dsp_processor/include/dsp_processor.h
+++ b/components/dsp_processor/include/dsp_processor.h
@@ -1,5 +1,6 @@
 #ifndef _DSP_PROCESSOR_H_
 #define _DSP_PROCESSOR_H_
+#include "esp_err.h"
 
 typedef enum dspFlows {
   dspfStereo,
```

> Also I have it running with my lyrat board for a while now without any issues. Does this happen frequently to you? Maybe a server issue? I am running snapserver v0.27 on openwrt.
Debian x86 with plenty of resources ( load average: 3.01, 3.56, 3.66 out of 16 cores). What goes into ```latency 520678917639us,``` number? I've seen the code but not sure I understand. It is more about the internal latency or network? Is it affected by timezone? It can't be 144h latency anyway. 

One time I was able to get this, but it is wit buffer = 450. 

```
I (34664) PLAYER: initial sync age: 2us, chunk duration: 26122us
E (37216) PLAYER: Failed to malloc memory for pcm chunk payload
W (37217) PLAYER: couldn't get memory to insert chunk, inserting an chunk containing just 0
W (37221) PLAYER: 20627, 5632, 79860, 59392
E (37635) PLAYER: Failed to malloc memory for pcm chunk payload
W (37635) PLAYER: couldn't get memory to insert chunk, inserting an chunk containing just 0
W (37640) PLAYER: 20647, 7936, 79880, 59392
E (38055) PLAYER: Failed to malloc memory for pcm chunk payload
W (38055) PLAYER: couldn't get memory to insert chunk, inserting an chunk containing just 0
W (38060) PLAYER: 20651, 4608, 79884, 59392
W (38631) PLAYER: RESYNCING HARD 2: age -225147us, latency 591343215908us, free 127919, largest block 59392, 4, rssi: -45
I (38938) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (39613) PLAYER: RESYNCING HARD 2: age -629us, latency 591343215908us, free 146523, largest block 73728, 0, rssi: -45
```
Changing buffer to 400 (minimum viable) went to repeating:
```E (47640) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 591397s, 827.862ms```

**SDKconfig diff for reference - can you share your Lyrat4.3 known good sdkconfig and snapserver conf?** 

Linux / Windows implementations of snapclient don't seem to have an issue so server side is less likely. 

Just in case you see something unusual that menuconfig might have changed. 

```
--- a/components/dsp_processor/include/dsp_processor.h
+++ b/components/dsp_processor/include/dsp_processor.h
@@ -1,5 +1,6 @@
 #ifndef _DSP_PROCESSOR_H_
 #define _DSP_PROCESSOR_H_
+#include "esp_err.h"
 
[... 845 lines cut ...]
+# CONFIG_MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH is not set
+# CONFIG_MBEDTLS_X509_TRUSTED_CERT_CALLBACK is not set
+# CONFIG_MBEDTLS_SSL_CONTEXT_SERIALIZATION is not set
+CONFIG_MBEDTLS_SSL_KEEP_PEER_CERTIFICATE=y
+# end of mbedTLS v2.28.x related
+
 #
 # Certificate Bundle
 #
@@ -1116,6 +1127,8 @@ CONFIG_SPI_FLASH_SUPPORT_ISSI_CHIP=y
 CONFIG_SPI_FLASH_SUPPORT_MXIC_CHIP=y
 CONFIG_SPI_FLASH_SUPPORT_GD_CHIP=y
 CONFIG_SPI_FLASH_SUPPORT_WINBOND_CHIP=y
+# CONFIG_SPI_FLASH_SUPPORT_BOYA_CHIP is not set
+# CONFIG_SPI_FLASH_SUPPORT_TH_CHIP is not set
 # end of Auto-detect flash chips
 
 CONFIG_SPI_FLASH_ENABLE_ENCRYPTED_READ_WRITE=y
@@ -1226,6 +1239,7 @@ CONFIG_WPA_MBEDTLS_CRYPTO=y
 # CONFIG_WPA_TESTING_OPTIONS is not set
 # CONFIG_WPA_WPS_STRICT is not set
 # CONFIG_WPA_11KV_SUPPORT is not set
+# CONFIG_WPA_DPP_SUPPORT is not set
 # end of Supplicant
 ```

CarlosDerSeher, 2023-09-03:
The changes making it compatible with v4.3.5 are exactly the same as I have them locally, so that's ok.

The naming of latency number in these resync logs is actually a bit misleading as it describes the difference of the microseconds since boot of server and boot of client.

Also you can ignore those warnings about the semaphore not available these logs could actually be removed from the code.

These logs with malloc errors are strange because allocation errors should only be happening when buffer is set to high...

CarlosDerSeher, 2023-09-03:
[sdkconfig_lyrat_v4.3.txt](https://github.com/CarlosDerSeher/snapclient/files/12506509/sdkconfig_lyrat_v4.3.txt)
[snapserver.conf.txt](https://github.com/CarlosDerSeher/snapclient/files/12506513/snapserver.conf.txt)

velaar, 2023-09-05:
So redoing everything from scratch improved the situation significantly for both no SPIRAP and SPIRAM scenarios. 

```
I (5018330) SC: try connecting to static configuration 172.20.22.1:1704
I (5018342) SC: netconn connected
I (5018345) SC: netconn sent hello message
I (5018351) SC: Buffer length:  950
I (5018351) SC: Latency:        0
I (5018354) SC: Mute:           0
I (5018358) SC: Setting volume: 99
I (5018366) SC: fLaC sampleformat: 44100:16:2
I (5018888) PLAYER: created new queue with 35
I (5018888) PLAYER: snapserver config changed, buffer 950ms, chunk 1152 frames, sample rate 44100, ch 2, bits 16 mute 0 latency 0
I (5020361) SC: latency buffer full
W (5020364) PLAYER: RESYNCING HARD 2: age -898165us, latency 789194628023us, free 4337078, largest block 4063232, 0, rssi: -48
I (5021342) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (5442146) PLAYER: RESYNCING HARD 2: age -194870us, latency 789194628617us, free 4216170, largest block 3932160, 26, rssi: -47
I (5442421) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (5467497) PLAYER: RESYNCING HARD 2: age -141335us, latency 789194628667us, free 4211502, largest block 3932160, 27, rssi: -49
I (5467719) PLAYER: initial sync age: 2us, chunk duration: 26122us
W (5471452) PLAYER: RESYNCING HARD 2: age -10095us, latency 789194628670us, free 4183566, largest block 3932160, 33, rssi: -49
I (5471542) PLAYER: initial sync age: 2us, chunk duration: 26122us
```
But occasional hard resyncs and associated sound artifacts still happen despite a very good RSSI and messages available on server. 

Played with buffer size but it doesn't seem to have an effect. 

Would appreciate any further guidance as I'm at a loss as to how to diagnose this issue.

CarlosDerSeher, 2023-09-06:
Maybe you have too many TCP retransmissions for some reason. Not sure how to enable those logs though, you'll have to check IDF documentation. Did you try to reboot your server? Change WiFi channel if it is crowded at your location? I chose a wifi channel which isn't occupied by any of my neighbors, not sure if this is possible for you...

velaar, 2023-09-06:
Thank you for all your help! I'll put a traffic capture/analysis VM into that network. to confirm if retransmission is indeed happening. Quick question on lightsnapcast though, shouldn't a sufficiently large buffer (say 1000-15000) cover for any retransmission issues? It just looks as if this behaviour is not affected by the buffer length. 

Say network is poor and say there is a complete dropout for 100ms. Should 1000ms buffer just cover for it?

CarlosDerSeher, 2023-09-06:
Yes you are right if the buffer is large enough TCP congestion shouldn't be an issue.

CarlosDerSeher, 2023-09-06:
Mybe enable the log on this line to see how age is behaving over time

https://github.com/CarlosDerSeher/snapclient/blob/8728191db796e89f56763c356522808dd9d45826/components/lightsnapcast/player.c#L1424C24-L1425C56

velaar, 2023-09-08:
``` --
I (550154) PLAYER: 0, -5us, q 47
I (550154) PLAYER: 0, -5us, -5us 1049774s, 559.407ms
I (550154) PLAYER: 0, -5us, -74us, 1us, q:47
I (550206) PLAYER: 0, -7us, q 45
I (550206) PLAYER: 0, -7us, -7us 1049774s, 559.407ms
I (550206) PLAYER: 0, -7us, -68us, 1us, q:45
I (550258) PLAYER: 0, -1us, q 43
I (550258) PLAYER: 0, -1us, -1us 1049774s, 559.407ms
I (550258) PLAYER: 0, -1us, -54us, 1us, q:43
I (550310) PLAYER: 0, 68us, q 41
I (550310) PLAYER: 0, 68us, 68us 1049774s, 559.407ms
I (550310) PLAYER: 0, 68us, -45us, 4us, q:41
I (550363) PLAYER: 0, -1us, q 39
I (550363) PLAYER: 0, -1us, -1us 1049774s, 559.407ms
I (550363) PLAYER: 0, -1us, -28us, 4us, q:39
W (550415) PLAYER: RESYNCING HARD 2: age -220697us, latency 1049774559407us, free 4165114, largest block 3801088, 37, rssi: -47
I (550717) PLAYER: initial sync age: 2us, chunk duration: 26122us
I (550717) PLAYER: 0, 2776us, q 43
I (550717) PLAYER: 0, 2776us, 2776us 1049774s, 559.407ms
I (550722) PLAYER: 0, 2776us, 0us, 0us, q:43
I (550766) PLAYER: 0, -452us, q 43
--
I (553012) PLAYER: 0, -507550us, q 44
I (553012) PLAYER: 0, -507550us, -507550us 1049774s, 559.409ms
I (553013) PLAYER: 0, -507550us, -507442us, -507531us, q:44
I (553065) PLAYER: 0, -507535us, q 43
I (553065) PLAYER: 0, -507535us, -507535us 1049774s, 559.409ms
I (553065) PLAYER: 0, -507535us, -507463us, -507535us, q:43
I (553117) PLAYER: 0, -507471us, q 42
I (553117) PLAYER: 0, -507471us, -507471us 1049774s, 559.409ms
I (553117) PLAYER: 0, -507471us, -507485us, -507535us, q:42
I (553169) PLAYER: 0, -507540us, q 43
I (553169) PLAYER: 0, -507540us, -507540us 1049774s, 559.409ms
I (553169) PLAYER: 0, -507540us, -507513us, -507535us, q:43
I (553221) PLAYER: 0, -507545us, q 43
I (553221) PLAYER: 0, -507545us, -507545us 1049774s, 559.409ms
I (553222) PLAYER: 0, -507545us, -507515us, -507535us, q:43
W (553274) PLAYER: RESYNCING HARD 2: age -507528us, latency 1049774559409us, free 4132570, largest block 3866624, 44, rssi: -46
I (553862) PLAYER: initial sync age: 2us, chunk duration: 26122us
I (553863) PLAYER: 0, 3197us, q 63
I (553863) PLAYER: 0, 3197us, 3197us 1049774s, 559.409ms
I (553868) PLAYER: 0, 3197us, 0us, 0us, q:63
I (553911) PLAYER: 0, -419us, q 62
--
I (588863) PLAYER: 0, -19us, q 57
I (588864) PLAYER: 0, -19us, -19us 1049774s, 559.535ms
I (588864) PLAYER: 0, -19us, 6us, 0us, q:57
I (588916) PLAYER: 0, 43us, q 56
I (588916) PLAYER: 0, 43us, 43us 1049774s, 559.535ms
I (588916) PLAYER: 0, 43us, 6us, 1us, q:56
I (588968) PLAYER: 0, -16us, q 57
I (588968) PLAYER: 0, -16us, -16us 1049774s, 559.529ms
I (588968) PLAYER: 0, -16us, 6us, 0us, q:57
I (589020) PLAYER: 0, 5us, q 58
I (589020) PLAYER: 0, 5us, 5us 1049774s, 559.529ms
I (589020) PLAYER: 0, 5us, 5us, 0us, q:58
I (589072) PLAYER: 0, -18us, q 57
I (589073) PLAYER: 0, -18us, -18us 1049774s, 559.529ms
I (589073) PLAYER: 0, -18us, 5us, -10us, q:57
W (589099) PLAYER: RESYNCING HARD 2: age -151127us, latency 1049774559529us, free 4069510, largest block 3801088, 57, rssi: -46
I (589330) PLAYER: initial sync age: 2us, chunk duration: 26122us
I (589354) PLAYER: 0, -422us, q 63
I (589354) PLAYER: 0, -422us, -422us 1049774s, 559.529ms
I (589354) PLAYER: 0, -422us, 0us, 0us, q:63
I (589406) PLAYER: 0, -440us, q 63```

So right before a resync  uxQueueMessagesWaiting(pcmChkQHdl) does go down. But not so 0. Actually far from it. 

And here it looks like there is no apparent reason at all: 

```
--
I (750121) PLAYER: 0, -26578us, -456us, -26578us, q:64
I (750173) PLAYER: 0, -26600us, q 63
I (750173) PLAYER: 0, -26600us, -26600us 1049774s, 559.925ms
I (750173) PLAYER: 0, -26600us, -457us, -26579us, q:63
I (750225) PLAYER: 0, -26577us, q 63
I (750225) PLAYER: 0, -26577us, -26577us 1049774s, 559.925ms
I (750226) PLAYER: 0, -26577us, -458us, -26578us, q:63
I (750278) PLAYER: 0, -26452us, q 63
I (750278) PLAYER: 0, -26452us, -26452us 1049774s, 559.925ms
I (750278) PLAYER: 0, -26452us, -461us, -26578us, q:63
W (750304) PLAYER: RESYNCING HARD 2: age -26573us, latency 1049774559925us, free 4039522, largest block 3801088, 64, rssi: -46
I (750411) PLAYER: initial sync age: 2us, chunk duration: 26122us
I (750434) PLAYER: 0, -466us, q 64
I (750434) PLAYER: 0, -466us, -466us 1049774s, 559.925ms
I (750434) PLAYER: 0, -466us, 0us, 0us, q:64
W (750459) PLAYER: send: pcmChunkQueue full, messages waiting 65
I (750487) PLAYER: 0, -467us, q 63
I (750487) PLAYER: 0, -467us, -467us 1049774s, 559.925ms
I (750487) PLAYER: 0, -467us, 0us, 0us, q:63
I (750539) PLAYER: 0, -468us, q 64
I (750539) PLAYER: 0, -468us, -468us 1049774s, 559.925ms
```

CarlosDerSeher, 2023-09-09:
ok, so there should be a new log every `CHNK_CTRL_CNT * chunk duration` µs, but when the resync happens in your first log the entry is indeed after `~53µs` but age shows `-220697us` these numbers just don't add up, so it seems there is a wrong calculation of age somewhere. I can't really interpret the second log it seems like player picks a chunk from the queue too early but `W (750459) PLAYER: send: pcmChunkQueue full, messages waiting 65` shouldn't happen at any time. This seems to be a hard one to find. I guess we have to look at all the numbers involved with age calculation, maybe some integer over- or underflow somewhere in the chain, I doubt it though knowing this calculations are using `int64_t`.
 
So these log messages you enabled before are somewhat redundant, could you please remove redundant entries?

There is a log message just below the line `age = serverNow - chunkStart - buf_us + clientDacLatency_us;` please enable this. 

Also enable the logs in `player_latency_insert()` which will return the median latency.

in `main.c` just below `player_latency_insert(tmpDiffToServer);` is a log message, please enable this one too.

That's a lot of data for the uart, let's hope this doesn't mess up timing of player too much.

CarlosDerSeher, 2023-09-09:
Ok, maybe I found the problem already now by having another look. The if condition resulting in `RESYNCING HARD 2:` is using `avg` which is the same as `age`. Shouldn't it be using `shortMedian` instead?

CarlosDerSeher, 2023-09-09:
Another interesting thing to know, at which commit did it stop working for you?

CarlosDerSeher, 2023-09-11:
I logged for about 5h on lyrat without any issues here.

velaar, 2023-09-11:
Here is a new set of logs. Still don't see anything obvious to my eye..  Will try to adjust the if condition that triggers resync,

```
I (61157) PLAYER: -1, -4237us, q 10
E (61183) PLAYER: age: -56479, serverNow 1314401639572, chunkStart 1314400996051, buf_us 700000
E (61209) PLAYER: age: -56476, serverNow 1314401665698, chunkStart 1314401022174, buf_us 700000
I (61210) PLAYER: -1, -4232us, q 8
E (61236) PLAYER: age: -56472, serverNow 1314401691824, chunkStart 1314401048296, buf_us 700000
E (61262) PLAYER: age: -56471, serverNow 1314401717948, chunkStart 1314401074419, buf_us 700000
I (61262) PLAYER: -1, -4227us, q 6
E (61288) PLAYER: age: -56470, serverNow 1314401744071, chunkStart 1314401100541, buf_us 700000
E (61314) PLAYER: age: -56466, serverNow 1314401770198, chunkStart 1314401126664, buf_us 700000
I (61314) PLAYER: -1, -4222us, q 4
E (61340) PLAYER: age: -56462, serverNow 1314401796324, chunkStart 1314401152786, buf_us 700000
E (61366) PLAYER: age: -56459, serverNow 1314401822449, chunkStart 1314401178908, buf_us 700000
I (61366) PLAYER: -1, -4215us, q 2
E (61392) PLAYER: age: -56457, serverNow 1314401848574, chunkStart 1314401205031, buf_us 700000
E (61418) PLAYER: age: -56453, serverNow 1314401874700, chunkStart 1314401231153, buf_us 700000
W (61419) PLAYER: RESYNCING HARD 2: age -4209us, latency 1314341007778us, free 202144, largest block 110592, 0, rssi: -44
E (61474) PLAYER: age: -26993, serverNow 1314401930283, chunkStart 1314401257276, buf_us 700000
I (61504) PLAYER: initial sync age: 2us, chunk duration: 26122us
E (61504) PLAYER: age: -49514, serverNow 1314401960007, chunkStart 1314401309521, buf_us 700000
I (61509) PLAYER: 0, 2730us, q 1
E (61527) PLAYER: age: -52487, serverNow 1314401983156, chunkStart 1314401335643, buf_us 700000
W (61527) PLAYER: RESYNCING HARD 2: age -243us, latency 1314341007778us, free 193564, largest block 110592, 0, rssi: -42
E (61538) PLAYER: age: -67311, serverNow 1314401994455, chunkStart 1314401361766, buf_us 700000
I (61616) PLAYER: initial sync age: 2us, chunk duration: 26122us
E (61616) PLAYER: age: -41807, serverNow 1314402072203, chunkStart 1314401414010, buf_us 700000
I (61624) PLAYER: (full) latency median: 1314341006070us
I (61627) SC: Current latency:1314340971526:
E (61639) PLAYER: age: -46610, serverNow 1314402093523, chunkStart 1314401440133, buf_us 700000
I (61641) PLAYER: 0, 5634us, q 22
E (61665) PLAYER: age: -46609, serverNow 1314402119646, chunkStart 1314401466255, buf_us 700000
E (61691) PLAYER: age: -46609, serverNow 1314402145769, chunkStart 1314401492378, buf_us 700000
I (61691) PLAYER: 0, 5635us, q 24
E (61717) PLAYER: age: -46625, serverNow 1314402171875, chunkStart 1314401518500, buf_us 700000
E (61743) PLAYER: age: -46624, serverNow 1314402197999, chunkStart 1314401544623, buf_us 700000
I (61744) PLAYER: 0, 5620us, q 23
E (61770) PLAYER: age: -46616, serverNow 1314402224129, chunkStart 1314401570745, buf_us 700000
E (61796) PLAYER: age: -46624, serverNow 1314402250244, chunkStart 1314401596868, buf_us 700000
```

another case

```
E (69293) PLAYER: age: -67955, serverNow 1314409752178, chunkStart 1314409120133, buf_us 700000
E (69319) PLAYER: age: -67946, serverNow 1314409778309, chunkStart 1314409146255, buf_us 700000
I (69319) PLAYER: 0, -15702us, q 23
E (69345) PLAYER: age: -67950, serverNow 1314409804427, chunkStart 1314409172377, buf_us 700000
E (69371) PLAYER: age: -67952, serverNow 1314409830548, chunkStart 1314409198500, buf_us 700000
I (69372) PLAYER: 0, -15708us, q 24
E (69398) PLAYER: age: -67958, serverNow 1314409856664, chunkStart 1314409224622, buf_us 700000
E (69424) PLAYER: age: -67956, serverNow 1314409882789, chunkStart 1314409250745, buf_us 700000
I (69424) PLAYER: 0, -15712us, q 23
E (69450) PLAYER: age: -67948, serverNow 1314409908919, chunkStart 1314409276867, buf_us 700000
I (69470) PLAYER: (full) latency median: 1314341010724us
I (69470) SC: Current latency:1314341034822:
E (69476) PLAYER: age: -67824, serverNow 1314409935166, chunkStart 1314409302990, buf_us 700000
I (69479) PLAYER: 0, -15580us, q 23
E (69502) PLAYER: age: -67828, serverNow 1314409961284, chunkStart 1314409329112, buf_us 700000
W (69502) PLAYER: RESYNCING HARD 2: age -15584us, latency 1314341010724us, free 95240, largest block 51200, 23, rssi: -41
E (69513) PLAYER: age: -82897, serverNow 1314409972338, chunkStart 1314409355235, buf_us 700000
I (69606) PLAYER: initial sync age: 2us, chunk duration: 26122us
E (69607) PLAYER: age: -41528, serverNow 1314410065952, chunkStart 1314409407480, buf_us 700000
E (69630) PLAYER: age: -44561, serverNow 1314410089041, chunkStart 1314409433602, buf_us 700000
I (69630) PLAYER: 0, 7683us, q 24
E (69656) PLAYER: age: -44561, serverNow 1314410115163, chunkStart 1314409459724, buf_us 700000
E (69682) PLAYER: age: -44563, serverNow 1314410141284, chunkStart 1314409485847, buf_us 700000
I (69682) PLAYER: 0, 7681us, q 22
E (69708) PLAYER: age: -44565, serverNow 1314410167404, chunkStart 1314409511969, buf_us 700000
[... 14 lines cut ...]
I (72033) PLAYER: 0, -14113us, q 23
E (72059) PLAYER: age: -66363, serverNow 1314412522749, chunkStart 1314411889112, buf_us 700000
E (72085) PLAYER: age: -66364, serverNow 1314412548871, chunkStart 1314411915235, buf_us 700000
I (72085) PLAYER: 0, -14120us, q 23
E (72111) PLAYER: age: -66367, serverNow 1314412574990, chunkStart 1314411941357, buf_us 700000
E (72138) PLAYER: age: -66345, serverNow 1314412601134, chunkStart 1314411967479, buf_us 700000
[... cut ...]
E (72342) PLAYER: age: -44536, serverNow 1314412805801, chunkStart 1314412150337, buf_us 700000
I (72342) PLAYER: 0, 7708us, q 23
E (72368) PLAYER: age: -44531, serverNow 1314412831928, chunkStart 1314412176459, buf_us 700000
E (72394) PLAYER: age: -44540, serverNow 1314412858041, chunkStart 1314412202581, buf_us 700000
I (72395) PLAYER: 0, 7704us, q 22
E (72421) PLAYER: age: -44529, serverNow 1314412884175, chunkStart 1314412228704, buf_us 700000
E (72447) PLAYER: age: -44536, serverNow 1314412910290, chunkStart 1314412254826, buf_us 700000
I (72447) PLAYER: 0, 7708us, q 23
```

velaar, 2023-09-12:
Hooray!! @CarlosDerSeher 

So, I probably did something that completely makes no sense but somehow works. And I keep 2 ESP32s (with and without modifications) side by side. The one without modifications does resync multiple times an hour. The one with modification also resyncs but a LOT less frequent and it's virtually inaudible. (I think it's actually related to start/stop of playback altogether so it's a silent resync). 

Important to note that over 3-4 hours the ESP32s did NOT desync (or at least it is not audible) 

Not the question is, does the modification below break the resync via age/median altogether as the only resyncs I see are q:0 related? 

```

@@ -1417,9 +1416,11 @@ static void player_task(void *pvParameters) {
 
         // resync hard if we are getting very late / early.
         // rest gets tuned in through apll speed control
+// velaar test 
+
         if ((msgWaiting == 0) || (MEDIANFILTER_isFull(&shortMedianFilter) &&
-                                  (abs(avg) > hardResyncThreshold)))
-        //        if (msgWaiting == 0)
+                                  (abs(shortMedian) > hardResyncThreshold)))
+        //if (msgWaiting == 0)
         {
           if (chnk != NULL) {
             free_pcm_chunk(chnk);

```

velaar, 2023-09-13:
> Do I understand correct, it works for you if you use `shortMedian `instead of `avg`? And your player to the server stops stuffing chunks to the snapfifo and thats why you see `q = 0` ?

So with the  `shortMedian `instead of `avg` I get significantly fewer resyncs. But I don't know if I just broke the || part of the if-clause. (I don't know if abs(shortMedian) > hardResyncThreshold will ever hit. 

```
|| (MEDIANFILTER_isFull(&shortMedianFilter) &&
                                  (abs(shortMedian) > hardResyncThreshold)))
```

q = 0 still happens from time to time. But I believe that it happens for unrelated reasons (wifi / network / server or microwave turned on by me/heighbours)

CarlosDerSeher, 2023-09-13:
So I will include this change in my next push alongside with the upgrade to IDF v4.3.5

commit 2ed3614b (2023-09-15): upgrade to IDF v4.3.5

close #35
closed 2023-09-15 by CarlosDerSeher