CarlosDerSeher/snapclient #36: DAC MAX98357A

jmlxxvi, 2023-09-06:
Hi, I have the SparkFun I2S Audio Breakout (MAX98357A) DAC at hand.
Is that DAC compatible with this client?
Would I need to change some code to make it work?
I am a developer but just trying my first steps on embedded.
A lot of fun (and doubts at the same time) so far.
Thanks!

CarlosDerSeher, 2023-09-06:
The codec isn't supported but feel free to add it. Look at components/custom_board folder to see how it is done. You'll have to write a driver for the codec, I recommend using i2s for communication. Start by using a copy of an existing driver in custom_board and edit the source files. Also you'll have to add the menuconfig entries in the kconfig file

jmlxxvi, 2023-09-07:
Hi Carlos, thanks for your prompt and clear response.
I think that I got it but now I am dealing with ETH issues.
By the way, if you know the configuration that works with [ESP32-POE-ISO](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware) board, that would help.

CarlosDerSeher, 2023-09-07:
There is a demo linked on the page you linked above. Maybe start from there? I didn't see which PHY is used imidiatly but you could use the demo to find a working config.

CarlosDerSeher, 2023-10-08:
@jmlxxvi is this still work in progress or can the issue be closed?

closed 2023-10-22 by CarlosDerSeher