CarlosDerSeher/snapclient #37: Building in docker

Morcatko, 2023-09-11:
Hello,
First let me thank you for this great piece of software.

I was not able to build it on windows (I guess the reason is because I was not able to install IDF :D ) so I made it work in docker.
So far I can run menuconfig, build it, and merge all bins into single binary. Flashing does not work but there are other/non idf tools that can do it well.

My question is would you be interested in PR with "build in docker" documentation?
If so do you want to have it in main readme or make an additional one?

Thanks.
Ondra

CarlosDerSeher, 2023-09-11:
Sure, documentation is a thing which still needs a lot of work. Maybe put it in a folder called doc.

Morcatko, 2023-09-12:
Here it is #38

commit bad6d67f (2023-09-12): Merge pull request #38 from Morcatko/DockerBuild

Docker build documentation

close #37
closed 2023-09-12 by CarlosDerSeher