CarlosDerSeher/snapclient #39: Suggestion: Custom DAC - no I2C

tortfeaser, 2023-09-21:
Hi there,

Really just wanted to say thanks for the work on this. Now I've got a really cost effective way to do sync'd up all of house audio.

But a suggestion? I struggled to get the code to build because I wasn't entering values for some of the DAC hardware ports, as my DAC didn't have the relevant pins. I'm using a cheap PCM5102 module that doesn't have I2C, so until I put some nonsense GPIO numbers in, I got CMake errors. Can the menuconfig tool be changed to omit I2C for custom DACs?

FWIW, I get continual errors: `E (696269) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload` but they don't seem to affect performance.

CarlosDerSeher, 2023-09-21:
About i2c pins, I am not sure about this as i don't use such DACs but you could try setting these to -1 which means no change normally. 

The error you are seeing is related to snapserver buffer length being too high. I am guessing you have PSRAM disabled and kept the default setting in on the server, e.g. 1000ms? Try lowering this value to something around 700ms.

tortfeaser, 2023-09-21:
Thanks Carlos. I'll try the -1 pin id. Will do re the buffer length. No PSRAM on this processor, ESP-WROOM-32. For the production items I'll use an S3.

Last thing. Should the volume control on the snapweb control thing for snapcast work? It doesn't for me. I can see via the serial monitor that the ESP is receiving the volume setting instruction, but there's no change.

luar123, 2023-09-21:
Your DAC has no volume control, so you need to enable DSP processing and softvolume. (Not sure about the latest menu config structure and names, I am using an old version.)

CarlosDerSeher, 2023-10-08:
@tortfeaser is this issue resolved and can be closed?

tortfeaser, 2023-10-09:
Yes, and yes. THanks.

closed 2023-10-09 by CarlosDerSeher