CarlosDerSeher/snapclient #47: Suggestion: Auto-reconnect on disconnect

jonzobot, 2023-11-05:
I love this project. Works great for me running a LyraT 4.3.

I've been tweaking my server a lot lately and when I restart the server or the Snapserver service, I have to manually power cycle the LyraT's. 

I wonder if, on disconnection, the clients could try reconnecting periodically? (in my particular use case, I use a static IP, not mDNS if that matters).

CarlosDerSeher, 2023-11-05:
Try using mdns of possible for you. Should work with that, at least for me.

Regarding static IP, will need to check the code to see why it doesn't reconnect automatically.  It definitely should though.

jonzobot, 2023-11-05:
Typical. Once I get in a position to debug the thing, it's working fine now. Not sure what the issue was before, but if it comes up again I'll troubleshoot first before posting.

Thanks for replying, this can be closed.

closed 2023-11-05 by jonzobot