CarlosDerSeher/snapclient #54: FLAC/OPUS codec not working

BenediktMueller91, 2024-01-04:
I had a go again with this repo after some while and this time around building and flashing worked mostly without issues using a ESP32-Wrover-B + MAX98357 DAC.

Only issue i still have now is that only PCM codec works. When using OPUS or FLAC the playback does not start.

Am i missing something?

Playback Log using codec PCM:
```
[...]
I (2539) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (5622) SC: Found 192.168.1.16:1704
I (5629) SC: netconn connected
I (5631) SC: netconn sent hello message
I (5636) SC: Buffer length:  500
I (5636) SC: Latency:        0
I (5637) SC: Mute:           0
I (5637) SC: Setting volume: 44
I (5641) SC: pcm sampleformat: 48000:16:2
I (5727) PLAYER: player_setup_i2s: dma_buf_len is 720, dma_buf_count is 8
I (5728) c_I2S: DMA Malloc info, datalen=blocksize=2880, dma_buf_count=8
I (5733) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (5745) GENERIC_BOARD: I2S0, MCLK output by GPIO0
I (5753) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
W (5765) PLAYER: no pcm chunk queue created?
I (5770) PLAYER: created new queue with 7
I (5774) PLAYER: snapserver config changed, buffer 500ms, chunk 2880 frames, sample rate 48000, ch 2, bits 16 mute 0 latency 0
I (5838) SC: latency buffer full
I (6310) PLAYER: initial sync age: 2us, chunk duration: 60000us
[...]
```

Playback Log using Codec FLAC:
```
[...]
I (4048) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (7131) SC: Found 192.168.1.16:1704
I (7138) SC: netconn connected
I (7141) SC: netconn sent hello message
I (7146) SC: Buffer length:  500
I (7146) SC: Latency:        0
I (7146) SC: Mute:           0
I (7147) SC: Setting volume: 50
I (7152) SC: fLaC sampleformat: 48000:16:2
```

Playback Log using codec OPUS:
```
[...]
I (3538) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (6621) SC: Found 192.168.1.16:1704
I (6628) SC: netconn connected
I (6630) SC: netconn sent hello message
I (6635) SC: Buffer length:  500
I (6635) SC: Latency:        0
I (6635) SC: Mute:           0
I (6636) SC: Setting volume: 50
I (6641) SC: Opus sample format: 48000:16:2

I (6646) SC: Initialized opus Decoder: 0
E (6650) SC: samples_per_frame: 960, pOpusData->bytes 1501, bytes 3840
E (6657) SC: Decode error : -2 

E (6711) SC: samples_per_frame: 960, pOpusData->bytes 1446, bytes 3840
E (6712) SC: Decode error : -2 

E (6770) SC: samples_per_frame: 960, pOpusData->bytes 1407, bytes 3840
E (6770) SC: Decode error : -2 

E (6828) SC: samples_per_frame: 960, pOpusData->bytes 1330, bytes 3840
E (6828) SC: Decode error : -2 

I (6842) SC: latency buffer full
E (6891) SC: samples_per_frame: 960, pOpusData->bytes 1566, bytes 3840
E (6891) SC: Decode error : -2
[...]
```

CarlosDerSeher, 2024-01-05:
@BenediktMueller91 please post your snapserver.conf, from your PCM log it looks like you have a high setting for chunk duration?
> I (6310) PLAYER: initial sync age: 2us, chunk duration: 60000us

Except that nothing suspicious here.

I just tested the master branch on my lyrat v4.3 board and don't see your issues with flac.

```
I (5676) SC: netconn connected
I (5678) SC: netconn sent hello message
I (5692) SC: Buffer length:  758
I (5692) SC: Latency:        0
I (5692) SC: Mute:           0
I (5692) SC: Setting volume: 100
I (5699) SC: fLaC sampleformat: 44100:16:2
I (5893) SC: latency buffer full
W (41832) PLAYER: pcm chunk queue not created
W (41833) PLAYER: pcm chunk queue not created
W (41834) PLAYER: pcm chunk queue not created
W (41837) PLAYER: pcm chunk queue not created
W (41842) PLAYER: pcm chunk queue not created
W (41847) PLAYER: pcm chunk queue not created
W (41852) PLAYER: pcm chunk queue not created
W (41857) PLAYER: pcm chunk queue not created
W (41863) PLAYER: pcm chunk queue not created
W (41872) PLAYER: pcm chunk queue not created
I (41875) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 4
I (41879) c_I2S: DMA Malloc info, datalen=blocksize=2304, dma_buf_count=4
I (41888) c_I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (41900) LYRAT_V4_3: I2S0, MCLK output by GPIO0
I (41907) c_I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
W (41920) PLAYER: no pcm chunk queue created?
I (41925) PLAYER: created new queue with 28
I (41930) PLAYER: snapserver config changed, buffer 758ms, chunk 1152 frames, sample rate 44100, ch 2, bits 16 mute 0 latency 0
I (42612) PLAYER: initial sync age: 2us, chunk duration: 26122us
```

I am running on `snapserver v0.27.0`

CarlosDerSeher, 2024-01-05:
OK, i just tested with 60ms chunk duration and then I see the same behavior. Must be related to chunk duration measurement in the esp32 client. I'll have to look into that. Please use 20ms for chunk duration in your snapserver.conf for now and it should work.

BenediktMueller91, 2024-01-05:
Thanks! 
Setting the chunk size to 20 did the trick. The player starts now at least.
There is still some crackling left but i will have a look if i can find the source of that.

CarlosDerSeher, 2024-01-05:
Unleast you have enabled parameter 800ms buffer time is too much. I use around 750ms but at a sample rate of 44100Hz, so for 48000Hz it will be lower without psram. But the module you are using should support psram, so enable it using menuconfig and 1000ms should also be ok

BenediktMueller91, 2024-01-05:
I based my configuration on the `sdkconfig_PSRAM_lyrat_v4.3` and only changed to `CONFIG_DAC_MAX98357=y` and added my wifi configuration.  So yes, PSRAM should already be in use.

I have to correct myself. The crackling is also present when using PCM, just as with FLAC. The audio starts out alright and then quickly starts to crackle - worsening over some time until there is heavy crackling/distortion and then stays that way.

I think this is not the right place to discuss this issue as it is not tied to the codec issue i had. Should i start a new issue or discussion on this?

CarlosDerSeher, 2024-01-05:
Yes please

CarlosDerSeher, 2024-05-25:
I am a bit lost here. Maybe it is a misunderstanding of how the protocol works and hopefully you can enlightening me. I logged some decoded chunks with their timestamps and sizes but somehow the numbers don't add up. Here is an example of what I mean

Chunk 0 --> timestamp 0ms --> size 1152 frames

| chunk   | timestamp [ms] | size [frames] |
| ------------- | ------------- | --------|
|  1  |  0  | 2304  |
|  2  | 20  |   |
|  2  | 60  |   |
|  2  | 20  |   |

badaix, 2024-05-25:
Which codec are you using?

CarlosDerSeher, 2024-05-26:
Flac

badaix, 2024-05-27:
It should be the time when it has been recorded. When using compression (like FLAC), the captured PCM data (e.g. 40ms) will be passed to the encoder. The encoder has his own latency, i.e. it needs a specific amount of data until it returns the compressed chunk. What's sent over the wire are always decodable chunks, flac needs 1152 frames, ~26.1224ms of audio, so 40ms will be passed to FLAC, 26ms will be encoded, a new timestamp for this chunk is computed, 14ms are still buffered in the encoder, the next 40ms are passed to the encoder => 52ms will be encoded, 2ms are buffered in the encoder, the next 40ms are passed to the encoder => 26ms of 42ms will be encoded, timestamped and sent, 16ms are buffered, ...
This is why alternating sizes are received on the client side. But the chunks should still be timestamped with the record start time of the chunk. 
This is just a brain dumb of what I did ~8 years ago. I will check in more detail next week, when I'm back from vacation.

2q2code, 2024-12-21:
Just an observation - your snapserver is serving streams by default at 48kHz, and the examples from the dev are running at 44.1kHz. I'm debugging a similar issue atm.

2q2code, 2024-12-21:
Switching to 44100 fixes this problem, but I'm using WROVER modules, so I should be able to handle a lot more. Is there a way to specify the buffer allotted on the device?

CarlosDerSeher, 2024-12-21:
Buffer is specified server side. It is dependent on sample rate. But this issue is definitely not related to sample rate

2q2code, 2024-12-21:
Well, switching to the 44100 sample rate allows me to use the firmware without error. Buffer size set to 20ms (default, though in the config file the noted requirement for flac is ~26ms). When I am up to debugging this at the firmware level, if it is still unresolved I will give it a look. I have a lot of experience with ESP32, but only using the Arduino framework. Should I open a new issue for my particular error messages or stick with this one?

CarlosDerSeher, 2024-12-22:
Sounds like your issue is different. This problem will come up if you set >40ms chunk size serverside

CarlosDerSeher, 2025-01-08:
Finally found a solution to this problem. It is fixed for flac but not OPUS. Feel free to test and comment

2q2code, 2025-01-09:
I think my issue was different to this but I will see if it changes that too. I also have a new WEIRD issue that I will raise separately. The good part is that I can duplicate it consistently and quickly. I have forked the repo so if I am feeling well enough I may even be able to help towards fixes and improvements.

CarlosDerSeher, 2025-01-13:
OK, I think this is a good step forward. Still needs some testing but seems to be stable on 2 out of 3 of my boards. Not sure why this one board doesn't play, it receives chunks at the beginning but then starves, I suspect the hardware is bad... Maybe some of you guys (@luar, @DerPcknicker) could test stability and also (measure) sync quality on your ends.

DerPicknicker, 2025-01-14:
Hi @CarlosDerSeher .. I can test it. But it will take some time. I guess in 2 weeks I can comeback and tell you how the sync works. I can only test it for TAS5805M and Lyrat Mini.

2q2code, 2025-01-14:
Hi all,

I can do no testing at the moment. I am literally, at this moment, as I
have been for weeks now, trying to arrange warranty-covered repair on
my main workstation. It's a mess over here. I will, however, test when
I can.

luar123, 2025-01-14:
Quick test sounds mostly good, didn't check sync or long term.
My setup: ESP32 without PSRAM, pcm5102a
48000:16:2, 700ms buffer, 20/52/56ms chunks, pcm/opus/flac

Everything works fine, just for flac/56ms after a short time playing it crashes with:
```
E (225283) SC: write_callback, failed to allocate PCM chunk payload
E (225284) SC: http_get_task: FLAC__stream_decoder_process_single failed
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d9550: 000000f9 0000f01d 88006136
--- 0x400d9550: http_get_task at /.../esp/snapclient/main/main.c:600

Core  1 register dump:
PC      : 0x400d9554  PS      : 0x00060e30  A0      : 0x00000000  A1      : 0x3ffd7080  
--- 0x400d9554: http_get_task at /.../esp/snapclient/main/main.c:2395

A2      : 0x3ffb436c  A3      : 0x00000507  A4      : 0x00001049  A5      : 0x00000001  
A6      : 0x00000507  A7      : 0x00000000  A8      : 0x800d8605  A9      : 0x3ffd7020  
A10     : 0x40181b98  A11     : 0x3f404238  A12     : 0x3f404950  A13     : 0x00037004  
--- 0x40181b98: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:30

A14     : 0x3f404238  A15     : 0x3f405318  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  
--- 0x400014fd: strlen in ROM
0x4000150d: strlen in ROM

Backtrace: 0x400d9551:0x3ffd7080
--- 0x400d9551: http_get_task at /...esp/snapclient/main/main.c:600

ELF file SHA256: 546670a8cccec528

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
```

Not sure when I will find time to test sync or run more extensive tests.

CarlosDerSeher, 2025-01-14:
I've tested with lyrat v4.3 and my custom board and sync seems pretty stable longterm (< 400µs) at least using flac and pcm. Opus sometimes seems to drift around a bit and also drift apart 2ms sometimes if chunk sizes are below 20ms but the problem is less intense if chunk size is increased. Not sure why this is. I've done a test for about 15min, so long term is relative here.

Regarding the error log, it is related to this part of the code:

```
while (decoderChunk.bytes > 0) {
  if (FLAC__stream_decoder_process_single(
          flacDecoder) == 0) {
    ESP_LOGE(
        TAG,
        "%s: FLAC__stream_decoder_process_single "
        "failed",
        __func__);

    return;
  }
}
```
FLAC__stream_decoder_process_single fails and the http_get_task returns. This is a low memory issue and could probably resolved by changing the code to wait until player task frees some memory. Like so:
```
while (decoderChunk.bytes > 0) {
  if (FLAC__stream_decoder_process_single(
          flacDecoder) == 0) {
    ESP_LOGE(
        TAG,
        "%s: FLAC__stream_decoder_process_single "
        "failed",
        __func__);

    // TODO: should insert some abort condition?
    vTaskDelay(pdMS_TO_TICKS(10));
  }
}
```

I'll push this change right now, but as ESP32 is a low RAM device we should probably add a notice to the README not to push chunk duration too high as such big memory blocks will sometimes just not be available

luar123, 2025-01-18:
Right, reducing the buffer solves the issue, too, because more free RAM is available.
However, the fix seems not work for me, although it probably would be a good idea, too.

I think the issue is in line 360 in write_callback()

CarlosDerSeher, 2025-01-18:
> I think the issue is in line 360 in write_callback()

Yeah I saw that later too. 

> at least a buffer of 1500ms

This really depends on where you live. In an urban area crowded with networks you are probably right

luar123, 2025-01-18:
Good to know, but for me 700ms usually works fine. We don't have many networks here and my snapclient esp is very close to the access point. Most of my snapclients are wired anyway.

CarlosDerSeher, 2025-01-19:
I've merged master now into this branch. As I've started work on the 20ms bug quite a while ago I've had to resolve a lot of conflicts manually. It seems to work, at least for the config I've used (Lyrat v4.3, custom Board with Ethernet).

Also I've done an upgrade to IDF v5.1.5. If you all find it working OK for your setups I'll do the merge back to master.

luar123, 2025-01-19:
Awesome! I was using IDF 5.1.5 already before without issues.

In general it is working (had to disable ethernet in menuconfig...was never enabled before). However I am not sure about stability. I am back to my default settings: `I (14880) PLAYER: snapserver config changed, buffer 700ms, chunk 1152 frames, sample rate 48000, ch 2, bits 16 mute 0 latency 0` (flac):

* Initially I got frequent hard resyncs, but now it runs fine for a long time....not sure what changed, but can't reproduce.
* When changing the volume a lot, sometimes I get this error (never saw before):
```
E (44211) libSNAPCAST: Error before: lq�񤀄�O���V#[�E����9��
                                                          ~s�¨>eٸ��⩈Q���B��V�F
E (44211) SC: Failed to read server settings: 1
E (44214) libSNAPCAST: Error before: lq�_+�^݀�w��f���u�D��ݰ�m�1)!~�A����DS�DR�Dr���
E (44223) SC: Failed to read server settings: 1
������]6�;`�6޻�$�Z�ST: Error before: �UJ����`Hhc�ש�F��a~�D��
E (44237) SC: Failed to read server settings: 1
E (44243) libSNAPCAST: Error before: ДRD ب�;U�\M�p3����!�=�	�?%�. �_�7X
                                                                           �A8^x�
E (44252) SC: Failed to read server settings: 1
TB�":M)y␦ libSNAPCAST: Error before: oD)␦��q6��R
S��ObMð  Y�~�.����q��]
--- Warning: Failed to decode multiple lines in a row. Try checking the baud rate and XTAL frequency setting in menuconfig
E (44266) SC: Failed to read server settings: 1
E (44271) libSNAPCAST: Error before: �4�ԇ�HL��@��jk�oQ�$YK�B�[��d����
E (44279) SC: Failed to read server settings: 1
E (44284) libSNAPCAST: Error before: �����Հ� (JN�(��Q�
E (44291) SC: Failed to read server settings: 1
```
Could you remove the comments in line 1698 and 1736 in player.c (or set log level to verbose) when merging to master? And maybe add #ifndef USE_SAMPLE_INSERTION around line 798 to resolve #94.

CarlosDerSeher, 2025-01-20:
Ok I've just pushed some fixes and updates right now.

> When changing the volume a lot, sometimes I get this error (never saw before):

This I can't reproduce, I've been sliding the volume from the android app constantly up and down for like a minute or so but no errors. Did you have those logs from player enabled while this happened?

luar123, 2025-01-22:
Logs were not enabled and with the latest update it is still the same. Even used the sdkconfig template to make sure I don't have wrong settings. I will check the server and network side next, could also be an issue there.

Anyway, I don't think this is in any way critical.

CarlosDerSeher, 2025-01-23:
I've found that when you have two stream sources defined in server and start playback to stream 1, client is assigned to stream 2. Now you change client to use stream 1, thn you'll have to change volume once until playback starts. Do see this too?

luar123, 2025-01-26:
> I've found that when you have two stream sources defined in server and start playback to stream 1, client is assigned to stream 2. Now you change client to use stream 1, thn you'll have to change volume once until playback starts. Do see this too?

I see a similar issue when booting esp32 in that state. Changing volume won't help though. I need to switch client to stream 2 and then stream 1 again to make it play. 

Regarding the server settings issue:

I found nothing in snapserver logs and didn't manage to sniff the network packages.
However, changing volume while nothing is playing can also lead to the error (had to move the slider a lot):
```
I (43807) dspProc: Set volume to 0.520000
I (43812) SC: got string: ufferMs":500,"latency":0,"muted":false,"volume":37}
E (43820) libSNAPCAST: Error before: ufferMs":500,"latency":0,"muted":false,"volume":37}
E (43829) SC: Failed to read server settings: 1
I (43834) SC: got string: {"bufferMs":500,"latency":0,"muted":false,"volume":43}
```
Here it seems only a few characters are missing.

If it is running, it is worse:
```
I (388334) dspProc: Set volume to 0.650000
I (388350) SC: got string: 
E (388350) libSNAPCAST: Error before: 
E (388350) SC: Failed to read server settings: 1
I (388354) SC: got string: ē*V����D�����#/�UI� �U�Y�B(�u�!4��{�I,V�����
E (388362) libSNAPCAST: Error before: ē*V����D�����#/�UI� �U�Y�B(�u�!4��{�I,V�����
E (388371) SC: Failed to read server settings: 1
I (388377) SC: got string: ��&*����o��d�]�qX @�B�I�r��W1yޯ��
                                                           e��Q��	��(
                                                                           �
E (388385) libSNAPCAST: Error before: ��&*����o��d�]�qX @�B�I�r��W1yޯ��
                                                                      e��Q��	��(
                                                                                   �
E (388394) SC: Failed to read server settings: 1
I (388404) SC: got string: {"bufferMs":500,"latency":0,"muted":false,"volume":56}
```
I don't plan to investigate this further.

Btw, master branch has both issues as well.

CarlosDerSeher, 2025-01-26:
Alright. Thanks for your time and reporting this. I'll see ifI can find the root cause of this

CarlosDerSeher, 2025-01-27:
@luar123 
> However, changing volume while nothing is playing can also lead to the error (had to move the slider a lot):

newest commit on the "fix 20ms" branch should fix this issue.

luar123, 2025-01-28:
> newest commit on the "fix 20ms" branch should fix this issue.

Perfect, thanks! Issue is gone 👍

CarlosDerSeher, 2025-01-28:
So as far as I can tell this is ready to be merged. Bit let's wait for @DerPicknicker 's comments.

DerPicknicker, 2025-01-28:
@CarlosDerSeher which branch with which IDF version should I test?

luar123, 2025-01-28:
> [@CarlosDerSeher](https://github.com/CarlosDerSeher) which branch with which IDF version should I test?

`fix_20ms_chunk_size_bug` branch with IDF v5.1.5

DerPicknicker, 2025-01-30:
@CarlosDerSeher @luar123 ..
Sorry for the delay .. I will test it next week - I need to setup my dev env again.

2q2code, 2025-02-14:
Current environment (using docker) is 5.1.1 - on my 12yo laptop until my workstation parts are returned to me (warranty replacement) - will that work or do I have to redo my docker env?

CarlosDerSeher, 2025-02-14:
you have to use idf v5.1.5

DerPicknicker, 2025-02-14:
@CarlosDerSeher .. For me it's working fine. But I can't guarantee that I tested all race conditions because I use a vanilla snapcast server without any custom configuration.

CarlosDerSeher, 2025-02-14:
Thanks. In my opinion it is ready to be merged.

2q2code, 2025-02-15:
Side question @CarlosDerSeher (or @DerPicknicker if applicable) - do you use OTA to flash this project (or any project)? Before this project, I worked with the Arduino framework version of the ESP32 libraries, and used OTA all the time. I had a bit of a scout around last night and it seems that the ESPIDF toolkit OTA methods are not as clear-cut/easily used. I ask for any of your experience on this because I embed devices quite a lot, so remotely flashing them is something I am quite accustomed to and use often (also with esphome). Ideally, I'd like to set up something like esphome for all my custom projects too.

CarlosDerSeher, 2025-02-15:
There is OTA already but it is a bit inconvenient to use. You can check the log to see the command you need to use from a terminal. I think it was using curl... Hopefully it still works as I didn't use this for quite some time...

CarlosDerSeher, 2025-02-15:
I've updated the readme [here](https://github.com/CarlosDerSeher/snapclient/tree/network_if_component) with some short instructions

2q2code, 2025-02-15:
Wow. That is far more complicated. I wonder if platformio with esp-idf is any better? Thank you so much for the guide notes - hopefully it will mean that I don't have to remove the ESP32 that is fully enclosed in a wooden enclosure beside my bed. It's only a relatively minor inconvenience but when you are disabled, you have to play smart - especially with motivations and energy.

CarlosDerSeher, 2025-02-15:
I've updated all my clients today and it worked well. Just make sure you are not playing any music while doing it or you will hear static/noise

CarlosDerSeher, 2025-02-15:
Also there is #92 which will hopefully add esphome compatibility which should increase usability?!

luar123, 2025-02-16:
I guess no-one tested with APLL tuning, here is a fix #114

CarlosDerSeher, 2025-02-16:
No I didn't as it seems to be the best solution using apll tuning. I was thinking of dropping this completely. What do you think?

luar123, 2025-02-16:
For me both approaches work fine, so no need to keep support for both.

DerPicknicker, 2025-02-16:
@CarlosDerSeher .. The TAS5805M which I use a lot don't like the APLL stuff (results in loosing high frequencies). I would drop it.

luar123, 2025-02-16:
Also newer esp versions don't support it

CarlosDerSeher, 2025-02-16:
So this was my thinking, sample stuffing is universal and works for all cases

CarlosDerSeher, 2025-02-16:
fingers crossed this goes well :) thanks for all your time!

commit c2285846 (2025-01-09): fix #54 for PCM
commit 111501e6 (2025-01-13): fix #54 for OPUS. Rework sync algorithm.

- remove some unused code and variables
- remove APLL dependency when using sample stuffing
- bug fixes
commit 34478557 (2025-02-16): fix #54 for flac

remove some comments
remove some logs
closed 2025-02-16 by CarlosDerSeher