CarlosDerSeher/snapclient #55: Crackling/Distortion after a few seconds of playback

BenediktMueller91, 2024-01-05:
With current `master` on my setup there is crackling/distortion building up after a few seconds. The audio then stays in a listenable but distorted state after that. This is best discernable in speech.

* Demo: [https://github.com/CarlosDerSeher/snapclient/assets/16675245/58227724-9cee-49d3-a4df-3088a2a84ae0 (mp4 audio)](https://github.com/CarlosDerSeher/snapclient/assets/16675245/58227724-9cee-49d3-a4df-3088a2a84ae0)
* Original: https://www.jamendo.com/track/1918929/feiticeira-louca

Current Setup:
* Server: `snapserver v0.27.0` on archlinux
* Client: ESP32-WROVER-B + Adafruit MAX98357A (with PSRAM, DSP, software-volume)
* sdkconfig: Based on `sdkconfig_PSRAM_lyrat_v4.3` plus DSP and MAX98357 ([diff](https://github.com/CarlosDerSeher/snapclient/files/13846452/diff.txt))

[image]

CarlosDerSeher, 2024-01-05:
Maybe related to DSP/Software volume? Could you test without?

Also could you try using standard sdkconfig and do settings for psram, board etc. yourself. Maybe sdkconfig_PSRAM_lyrat_v4.3 is faulty for some reason.

If that doesn't help maybe go back in time, you could start by trying commit 06161ae385524cf52a8ad29472a81db39e296563

BenediktMueller91, 2024-01-06:
Unfortunately the logs are pretty empty and do not indicate any issues. 
I recompiled with `CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG=y` (and changed the `nvs` partition size to fix the overlap issue when doing so).
```
I (4759) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (7839) SC: Found 192.168.1.16:1704
I (7843) SC: netconn connected
I (7845) SC: netconn sent hello message
I (7854) SC: Buffer length:  750
I (7854) SC: Latency:        10
I (7854) SC: Mute:           0
I (7856) SC: Setting volume: 14
I (7861) SC: fLaC sampleformat: 48000:16:2
I (7894) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 4
I (7895) c_I2S: DMA Malloc info, datalen=blocksize=2304, dma_buf_count=4
I (7900) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (7912) GENERIC_BOARD: I2S0, MCLK output by GPIO0
I (7920) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
W (7932) PLAYER: no pcm chunk queue created?
I (7937) PLAYER: created new queue with 30
I (7941) PLAYER: snapserver config changed, buffer 750ms, chunk 1152 frames, sample rate 48000, ch 2, bits 16 mute 0 latency 10
I (8055) SC: latency buffer full
I (8802) PLAYER: initial sync age: 2us, chunk duration: 24000us
```

Also the issue still persists when disabling the DSP functionality - except now there is also some distortion from the poor speaker struggling with the volume ;)

I then cherry-picked the commit `Added support for MAX98357 (#46)` onto https://github.com/CarlosDerSeher/snapclient/commit/06161ae385524cf52a8ad29472a81db39e296563 and recompiled. The issue persists.

The crackling is gone for a few seconds everytime i switch snapcast sources or pause the playback on the source. Are there some debug logs regarding the i2s/playback buffer stats that i could activate? See if there are some buffers that run dry.

CarlosDerSeher, 2024-01-06:
You could enable logs around here https://github.com/CarlosDerSeher/snapclient/blob/master/components%2Flightsnapcast%2Fplayer.c#L1510 but I don't think it will help here. Still, please try.

Because of the loud volume could you just lower the volume on your input source to the server to ensure it isn't a dsp issue. I guess you are unable to hear the crackling with soft volume disabled because of the speaker distortion?

Also did you ensure it isn't a server issue? Do other architecture clients have this issue too (probably not)?

Another thing you could try is to [disable](https://github.com/CarlosDerSeher/snapclient/blob/master/components%2Flightsnapcast%2Fplayer.c#L1469) apll tuning temporarily. The DAC doesn't have a MCLK line does it? I am not sure how this affects operation. I don't think it should be a problem as the others signal timings also change with MCLK timing but who knows.

I don't have this particular DAC so maybe someone else  can report if he has similar issues, you could check if the DAC is correctly initialized by the driver. Maybe something is wrong there?

BenediktMueller91, 2024-01-06:
Other clients on my network play fine. I have a few around the house (Raspberry Pi 4B, Raspberry Pi 5, Snapweb).

Even without DSP and softvolume i can still hear the crackling over the slight distortion of the overdriven speaker. Reducing volume on the source side was a good idea. Just had to enable volume on the mopidy instance again.

I enabled the log in L1510 as mentioned. I get the feeling that the distortion is tied to the APLL direction switching. When it gets to the point where it oscillates the distortion becomes the strongest.
```
# L1510 Log Enabled
E (4259) esp_ota_ops: Running firmware is factory
I (4260) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (7340) SC: Found 192.168.1.16:1704
I (7344) SC: netconn connected
I (7346) SC: netconn sent hello message
I (7350) SC: Buffer length:  750
I (7350) SC: Latency:        10
I (7350) SC: Mute:           0
I (7352) SC: Setting volume: 14
I (7357) SC: fLaC sampleformat: 48000:16:2
I (7394) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 2
I (7395) c_I2S: DMA Malloc info, datalen=blocksize=2304, dma_buf_count=2
I (7400) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (7412) GENERIC_BOARD: I2S0, MCLK output by GPIO0
I (7420) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
W (7432) PLAYER: no pcm chunk queue created?
I (7437) PLAYER: created new queue with 31
I (7441) PLAYER: snapserver config changed, buffer 750ms, chunk 1152 frames, sample rate 48000, ch 2, bits 16 mute 0 latency 10
I (9351) SC: latency buffer full
I (10080) PLAYER: initial sync age: 2us, chunk duration: 24000us
I (10102) PLAYER: 0, -231us, -231us 6150986s, 254.116ms
I (10150) PLAYER: 0, -252us, -252us 6150986s, 254.116ms
I (10198) PLAYER: 0, -214us, -214us 6150986s, 254.116ms
I (10246) PLAYER: 0, -236us, -236us 6150986s, 254.116ms
I (10294) PLAYER: 0, -128us, -128us 6150986s, 254.116ms
I (10342) PLAYER: 0, -240us, -240us 6150986s, 254.116ms
I (10390) PLAYER: 0, -227us, -227us 6150986s, 254.121ms
I (10438) PLAYER: 0, -235us, -235us 6150986s, 254.121ms
I (10486) PLAYER: 0, -234us, -234us 6150986s, 254.121ms
I (10534) PLAYER: 0, -191us, -191us 6150986s, 254.121ms
I (10582) PLAYER: 0, -244us, -244us 6150986s, 254.121ms
I (10630) PLAYER: 0, -241us, -241us 6150986s, 254.121ms
I (10678) PLAYER: 0, -253us, -253us 6150986s, 254.121ms
I (10726) PLAYER: 0, -250us, -250us 6150986s, 254.121ms
I (10774) PLAYER: 0, -239us, -239us 6150986s, 254.121ms
I (10822) PLAYER: 0, -231us, -231us 6150986s, 254.121ms
I (10870) PLAYER: 0, -254us, -254us 6150986s, 254.121ms
I (10918) PLAYER: 0, -213us, -213us 6150986s, 254.121ms
I (10966) PLAYER: 0, -242us, -242us 6150986s, 254.121ms
I (11014) PLAYER: 0, -253us, -253us 6150986s, 254.121ms
I (11062) PLAYER: 0, -251us, -251us 6150986s, 254.121ms
I (11110) PLAYER: 0, -228us, -228us 6150986s, 254.121ms
I (11158) PLAYER: 0, -231us, -231us 6150986s, 254.121ms
I (11206) PLAYER: 0, -248us, -248us 6150986s, 254.121ms
I (11254) PLAYER: 0, -206us, -206us 6150986s, 254.121ms
I (11302) PLAYER: 0, -247us, -247us 6150986s, 254.121ms
I (11350) PLAYER: 0, -242us, -242us 6150986s, 254.121ms
I (11398) PLAYER: 0, -247us, -247us 6150986s, 254.121ms
I (11446) PLAYER: 0, -244us, -244us 6150986s, 254.121ms
I (11494) PLAYER: 0, -232us, -232us 6150986s, 254.121ms
I (11542) PLAYER: 0, -240us, -240us 6150986s, 254.121ms
I (11590) PLAYER: 0, -226us, -226us 6150986s, 254.121ms
I (11638) PLAYER: 0, -210us, -210us 6150986s, 254.121ms
I (11686) PLAYER: 0, -244us, -244us 6150986s, 254.121ms
I (11734) PLAYER: 0, -252us, -252us 6150986s, 254.121ms
I (11782) PLAYER: 0, -245us, -245us 6150986s, 254.121ms
I (11830) PLAYER: 0, -229us, -229us 6150986s, 254.121ms
I (11878) PLAYER: 0, -238us, -238us 6150986s, 254.121ms
I (11926) PLAYER: 0, -233us, -233us 6150986s, 254.121ms
I (11974) PLAYER: 0, -206us, -206us 6150986s, 254.121ms
I (12022) PLAYER: 0, -236us, -236us 6150986s, 254.121ms
I (12070) PLAYER: 0, -244us, -244us 6150986s, 254.121ms
I (12118) PLAYER: 0, -250us, -250us 6150986s, 254.121ms
[... 428 lines cut ...]
I (21695) PLAYER: 0, -297us, -297us 6152144s, 717.19ms
I (21743) PLAYER: 0, -323us, -323us 6152144s, 717.19ms
I (21791) PLAYER: 0, -321us, -321us 6152144s, 717.19ms
I (21839) PLAYER: 0, -337us, -337us 6152144s, 717.19ms
I (21887) PLAYER: 0, -314us, -314us 6152144s, 717.19ms
I (21935) PLAYER: 0, -320us, -320us 6152144s, 717.19ms
I (21983) PLAYER: 0, -330us, -330us 6152144s, 717.19ms
I (22031) PLAYER: 0, -287us, -287us 6152144s, 717.19ms
I (22079) PLAYER: 0, -329us, -329us 6152144s, 717.19ms
I (22127) PLAYER: 0, -330us, -330us 6152144s, 717.19ms
I (22175) PLAYER: 0, -324us, -324us 6152144s, 717.19ms
I (22223) PLAYER: 0, -308us, -308us 6152144s, 717.19ms
I (22271) PLAYER: 0, -329us, -329us 6152144s, 717.19ms
I (22319) PLAYER: 0, -313us, -313us 6152144s, 717.19ms
I (22367) PLAYER: 0, -318us, -318us 6152144s, 717.19ms
I (22415) PLAYER: 0, -317us, -317us 6152144s, 717.19ms
I (22463) PLAYER: 0, -329us, -329us 6152144s, 717.19ms
I (22511) PLAYER: 0, -327us, -327us 6152144s, 717.19ms
I (22559) PLAYER: 0, -327us, -327us 6152144s, 717.19ms
I (22607) PLAYER: 0, -312us, -312us 6152144s, 717.19ms
I (22655) PLAYER: 0, -321us, -321us 6152144s, 717.19ms
I (22703) PLAYER: 0, -333us, -333us 6152144s, 717.19ms
I (22751) PLAYER: 0, -329us, -329us 6152144s, 717.19ms
I (22799) PLAYER: 0, -333us, -333us 6152144s, 717.19ms
```

luar123, 2024-01-06:
> Another thing you could try is to [disable](https://github.com/CarlosDerSeher/snapclient/blob/master/components%2Flightsnapcast%2Fplayer.c#L1480) apll tuning temporarily. The DAC doesn't have a MCLK [...]

I am using pcm5102a dac without MCLK (it is optional for this dac) and don't have any distortion/crackling issues.

CarlosDerSeher, 2024-01-06:
Try playing a sine wave and maybe record it somehow. Using a oscilloscope maybe. There isn't a line out on this dac I guess so you could record it using a pc soundcard?
Maybe it helps to see how the distortion looks like...

CarlosDerSeher, 2024-01-06:
Ahhh and I see another possible problem  on your setup. The wifi antenna is very close to your audio board. This could also be to source of your troubles. Try to get as much space between the antenna and audio!

BenediktMueller91, 2024-01-06:
> Ahhh and I see another possible problem on your setup. The wifi antenna is very close to your audio board. This could also be to source of your troubles. Try to get as much space between the antenna and audio!

I put the DAC on a second breadboard. Unfortunately this did not improve anything.

> I don't have this particular DAC so maybe someone else can report if he has similar issues, you could check if the DAC is correctly initialized by the driver. Maybe something is wrong there?

The DAC does not need any runtime configuration. It should just play any I2S data sent to it.

> Try playing a sine wave and maybe record it somehow. Using a oscilloscope maybe. There isn't a line out on this dac I guess so you could record it using a pc soundcard?
Maybe it helps to see how the distortion looks like...

Unfortunately this DAC does not have a line level output. I added a voltage divider to ground on both speaker leads (with the speaker still connected) and fed that into the line in of a usb soundcard. One file is from a direct loop on the soundcard playing the same file and the other one from the esp32 playing the file. It is a 440Hz sine wave.
[Audio-Recordings.zip](https://github.com/CarlosDerSeher/snapclient/files/13850318/Audio-Recordings.zip)

CarlosDerSeher, 2024-01-06:
Honestly I am not sure where this is coming from. To me it kind of looks like noise from the power supply. Could yo try to use a different supply for the DAC. At the moment you are using to power supply from the ESP module don't you?

[image]

BenediktMueller91, 2024-01-06:
Here is the same file played on a raspberrypi connected to the DAC: [esp32-snapcast-workbench_via_raspberry.zip](https://github.com/CarlosDerSeher/snapclient/files/13850581/esp32-snapcast-workbench_via_raspberry.zip)

Maybe my way of recording is not perfect. Still the sound does seem a little better coming from the raspberry pi - but also farm from really good. I will try to wire up a separate power supply for the DAC to avoid getting interference from the usb lines or ESP32 cpu noise.

CarlosDerSeher, 2024-01-06:
There is a lot of noise when using the DAC. Maybe some of it originates from your measurement setup but RPi and ESP32 look very similar to me, so I am assuming some kind of hardware problem here. RPi noise floor is a bit lower but still, the spectrum is very similar.

ESP32 playback:
[image]

RPi playback:
[image]

loop playback:
[image]

BenediktMueller91, 2024-01-06:
So then i will put this aside as problem of the DAC for now. I was hoping to use this simple DAC for testing and switch to something more useful once i am happy with the basic playback and behaviour of the software. Something like the HiFiBerry AMP2 (TAS5713) or IQaudio DigiAMP+ (TAS5756M).

I am already using the IQaudio DigiAmp+ on one of my raspberry pi snapcast clients and it sounds great.

So the only 'real' problem with the crackling was probably the APLL correction. Could this also be something the DAC does not like? If so, I could try switching to the IQaudio amplifier and adding it as a custom board (I may have one lying around somewhere).

DerPicknicker, 2024-01-06:
@BenediktMueller91 
I use an TAS5805M board. And it sounds great with an esp32 I can't hear any difference in comparison to an raspberry pi. I am working on the support for this board: https://www.tindie.com/products/sonocotta/louder-esp32/
(Almost finished 95% it needs some adjustments but the board and the amp works fine) 
So from my perspective an raspberry is really not needed for an snapclient.

CarlosDerSeher, 2024-01-06:
> Could this also be something the DAC does not like?

I read something about jitter correction in the datasheet but I doubt this is the problem. I don't think that apll correction is the problem. Did you try to change the power supply of the DAC?

Also @normanalie could be of help here maybe, he did add the codec and maybe has some input here.

BenediktMueller91, 2024-01-06:
> USB is really noisy and I can confirm with other dacs that the sound quality is not the best if the power supply is noisy.

> I read something about jitter correction in the datasheet but I doubt this is the problem. _I don't think that apll correction is the problem. Did you try to change the power supply of the DAC?_

I added a separate lab power supply for the DAC and now the quality is overall quite good. ESP32 and DAC share only ground and the i2s lines now. It looks like even when feeding the ESP32 with a really good power supply via the USB plug the power rail became quite noisy. 

[image]

normanalie, 2024-01-07:
I've never encountered this problem with this DAC. 
First, I think that you don't have a resistor on the SD_MODE pin. With the ESP32, to have the sum L+R you need a resistor of 632ohms. With the E12 series, you can put a 560ohm resistor between SD_MODE and 3.3V.
Do you have a way of measuring the BCLK signal? In the event of jitter or noise, the DAC behaves randomly.

BenediktMueller91, 2024-01-07:
> I've never encountered this problem with this DAC. First, I think that you don't have a resistor on the SD_MODE pin. With the ESP32, to have the sum L+R you need a resistor of 632ohms. With the E12 series, you can put a 560ohm resistor between SD_MODE and 3.3V.

The Adafruit Max98357A Board already includes the right resistor for the L+R operation mode when supplying 5V (see https://learn.adafruit.com/assets/32642).

> Do you have a way of measuring the BCLK signal? In the event of jitter or noise, the DAC behaves randomly.

Unfortunately i don't have any kind of logic analyzer at the moment. I have ordered a cheap amazon usb logic analyzer (https://www.amazon.de/dp/B01MUFRHQ2) and will have a look at the i2s lines once it arrives and i can find some time.

normanalie, 2024-01-07:
On the schematic, R1 is 1mOhm, thats too much for 3.3V logic. Based on the data sheet, SD_MODE pull-up resistor should be 632ohms on 3.3V I/O. I don't think this will solve your problem but it will eliminate a potential problem and your electronic design will follow the recommendations.

[https://www.analog.com/media/en/technical-documentation/data-sheets/max98357a-max98357b.pdf](https://www.analog.com/media/en/technical-documentation/data-sheets/max98357a-max98357b.pdf)

CarlosDerSeher, 2024-01-07:
Are you sure it us 1m? To me it looks more like 1M.

BenediktMueller91, 2024-01-07:
> Did you ever try a different speaker? Maybe your speaker's impedance isn't supported by the DAC?

I switched to a Visaton SC 5.9 (10W, 8 Ohms). The same issues persist. But it still sounds a lot better with the bigger speaker. The spectrograms i posted where already recored from this speaker.

> On the schematic, R1 is 1mOhm, thats too much for 3.3V logic. Based on the data sheet, SD_MODE pull-up resistor should be 632ohms on 3.3V I/O. I don't think this will solve your problem but it will eliminate a potential problem and your electronic design will follow the recommendations.

In Table 5 `V[SD_MODE]` is given as `B1 trip point > V[SD_MODE] > B0 trip point` with `B0` and `B1` defined on page 7 (ELECTRICAL CHARACTERISTICS) as typ. 0.16v and typ. 0.77v respectively. I measured 0.47v with no additional components connected to the SD_MODE pin of the breakout board. So this should put the DAC securely into the "(Left/2 + right/2)" mode.

The pictured board is currently the only ESP32 available to me so I also ordered another ESP32-WROVER based board to make sure it is not something about this specific board. I will report back once the logic analyzer is running and i tested the new board.

BenediktMueller91, 2024-01-10:
tl;dr: When apll correction is active, i still have crackling but in a way that sounds more like short interruptions/pops. On the i2s lines the data looks fine in both cases.

Ok, so the logic analyzer did not make it any better 😕. And with the ESP32-WROVER CAM Board the issues are the same. Now only the DAC and the software are left as culprits imho.

In Sigrok/PulseView i recorded 50 MSamples at 6Mhz and added an i2s decoder. With a simple python script i transformed the exported i2s text data files into binary data files and imported them into audacity. In both cases when selecting 2-channel 16 bit signed pcm on the raw importer in audacity, a good looking/sounding sine wave was the result.

As far as i can tell the BCLK as well as LRCLK and DATA look fine and have no relevant timing inconsistencies. Maybe one of you can spot something in the recorded data?

[Sigrok-Recording-01.zip](https://github.com/CarlosDerSeher/snapclient/files/13895221/Sigrok-Recording-01.zip)

And thats the script i used to convert the data exported from PulseView:
```
#!/usr/bin/env python3

import sys

filename_in = sys.argv[1]
filename_out = filename_in + ".raw"

infile = open(filename_in, "r")
outfile = open(filename_out, "wb")

print(f"Reading: {filename_in}")
print(f"Writing: {filename_out}")
for line in infile:
    outfile.write(bytes.fromhex(line[-5:-1]))
print("Done!")
```

CarlosDerSeher, 2024-01-11:
@BenediktMueller91 BCK width jitters a little for both apll/no_apll but I guess thats because of sample rate of logic analyzer not beeing high enough.

Also I get an error using yourt conversion script:

```
python3 convert.py esp32-i2s-playback_sine_stereo_apll.sr
Reading: esp32-i2s-playback_sine_stereo_apll.sr
Writing: esp32-i2s-playback_sine_stereo_apll.sr.raw
Traceback (most recent call last):
  File "convert.py", line 13, in <module>
    for line in infile:
  File "/usr/lib/python3.8/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xbe in position 15: invalid start byte
```

BenediktMueller91, 2024-01-11:
> @BenediktMueller91 BCK width jitters a little for both apll/no_apll but I guess thats because of sample rate of logic analyzer not beeing high enough.
>
> Also I get an error using yourt conversion [...]

The script does not work on the *.sr files but the decoded/exported i2s data. You get the file when right-clicking on the i2s track and export that row. The file should look like this:
```
[...]
1177-1240 I²S: : Left channel: 0000102d
1240-1302 I²S: : Right channel: 0000102d
[...]
```

[image]

CarlosDerSeher, 2024-01-11:
I figured that out later, but still the same error
```
python3 convert.py esp32-i2s-playback_sine_stereo_apll.txt
Reading: esp32-i2s-playback_sine_stereo_apll.txt
Writing: esp32-i2s-playback_sine_stereo_apll.txt.raw
Traceback (most recent call last):
  File "convert.py", line 14, in <module>
    for line in infile:
  File "/usr/lib/python3.8/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb2 in position 8: invalid start byte
```

esp32-i2s-playback_sine_stereo_apll.txt:
```
52-114 I²S: : Left channel: 000018d8
114-177 I²S: : Right channel: 000018d8
177-239 I²S: : Left channel: 0000182d
239-302 I²S: : Right channel: 0000182d
302-364 I²S: : Left channel: 0000176d
364-427 I²S: : Right channel: 0000176d
427-489 I²S: : Left channel: 00001699
489-552 I²S: : Right channel: 00001699
552-614 I²S: : Left channel: 000015b3
614-677 I²S: : Right channel: 000015b3
677-739 I²S: : Left channel: 000014b9
739-802 I²S: : Right channel: 000014b9
802-864 I²S: : Left channel: 000013ae
864-927 I²S: : Right channel: 000013ae
927-990 I²S: : Left channel: 00001293
990-1052 I²S: : Right channel: 00001293
1052-1115 I²S: : Left channel: 00001167
1115-1177 I²S: : Right channel: 00001167
```

BenediktMueller91, 2024-01-11:
Oh, probably the encoding is something different on your system by default. Probably it will be ISO-8859-1 instead of UTF-8. Try wether it works when explicitly specifying the encoding like this:
```diff
 filename_out = filename_in + ".raw"
 
-infile = open(filename_in, "r")
+infile = open(filename_in, "r", encoding="ISO-8859-1")
 outfile = open(filename_out, "wb")
```

CarlosDerSeher, 2024-01-11:
Ok now I am getting somewhere. audio sample rate is 48kHz and it is big endian format isn't it?

CarlosDerSeher, 2024-01-11:
> In Sigrok/PulseView i recorded 50 MSamples at 6Mhz and added an i2s decoder. With a simple python script i transformed the exported i2s text data files into binary data files and imported them into [...]

So you recorded the I2S signals coming from ESP32 and we get a clean audio if we listen to it on a PC. At least it sounds / looks clean to me on my PC. This leaves your DAC as the problem I guess? Question is what is different for your hardware as it seems to work for others with this DAC type...

CarlosDerSeher, 2024-01-11:
Can you see the APLL direction switch with the logic analyzer? maybe there is a glitch on the line when this is happening? Not sure how we could visualize that as we cant really trigger on the direction switch. maybe toggle a gpio when it is happening and use a scope to trigger on that signal?

CarlosDerSeher, 2024-01-11:
And you talked about the raspberry behaving similar with this DAC connected? If that is the case then your DAC is the problem

CarlosDerSeher, 2024-01-11:
And here is yet another possible problem I found: 
[image]

In line https://github.com/CarlosDerSeher/snapclient/blob/d3e20fe0eb4d8d562d97a02f43cbccb79e9c8505/main/main.c#L2777C13-L2777C13 I activate the I2S interface before I initialize the DAC. I did that for ADAU1961 DAC but it seems yours needs a different startup sequence. Maybe just comment the I2S initialization in main for a quick test?

DerPicknicker, 2024-01-11:
@BenediktMueller91 ... Try also soldering the dac to the esp32 loose connection can cause also issues. I don't know the bandwidth but higher clock frequencies like 5mhz need short cables. Before looking that deep into software be sure that your hardware is connected properly.

BenediktMueller91, 2024-01-11:
It has to somehow be connected to software. I tried to change as little as possibel between the tests (same snapcast server, same stream, same/similar volume):
* Output sounds fine when connecting the DAC to my Raspi4
* Output sounds ok/good when connecting the DAC to my ESP32 with `adjust_apll(dir)` disabled
* Output has varying amounts of pops/crackling when connecting the DAC to my ESP32 with `adjust_apll(dir)` enabled

Note: I added a short log with the `dir` var. The artifacts in the output correspond to `dir` changing (rapidly). Unfortunately i could not get `gpio_set_level` to work. Also i discovered that i can't connect the logic analyzer and the DAC to BCK at the same time. The DAC goes silent as soon as i connect any channel of the analyzer to the BCK clock pin. LRCK and DATA i can record while the DAC is still working.
```
I (60903) PLAYER: APLL Dir: 0
I (60923) PLAYER: APLL Dir: 1
I (60943) PLAYER: APLL Dir: 1
I (60963) PLAYER: APLL Dir: 1
I (60983) PLAYER: APLL Dir: 0
I (61003) PLAYER: APLL Dir: 1
I (61023) PLAYER: APLL Dir: 1
I (61043) PLAYER: APLL Dir: 1
I (61063) PLAYER: APLL Dir: 1
I (61083) PLAYER: APLL Dir: 1
I (61103) PLAYER: APLL Dir: 1
I (61123) PLAYER: APLL Dir: 0
I (61143) PLAYER: APLL Dir: 1
I (61163) PLAYER: APLL Dir: 1
I (61183) PLAYER: APLL Dir: 1
I (61203) PLAYER: APLL Dir: 0
I (61223) PLAYER: APLL Dir: 0
I (61243) PLAYER: APLL Dir: 1
I (61263) PLAYER: APLL Dir: 1
I (61283) PLAYER: APLL Dir: 0
I (61303) PLAYER: APLL Dir: 0
I (61323) PLAYER: APLL Dir: 0
I (61343) PLAYER: APLL Dir: 0
I (61363) PLAYER: APLL Dir: 1
I (61383) PLAYER: APLL Dir: 0
I (61403) PLAYER: APLL Dir: 1
I (61423) PLAYER: APLL Dir: 1
I (61443) PLAYER: APLL Dir: 1
I (61463) PLAYER: APLL Dir: 1
I (61483) PLAYER: APLL Dir: 0
I (61503) PLAYER: APLL Dir: 1
I (61523) PLAYER: APLL Dir: 0
I (61543) PLAYER: APLL Dir: 1
I (61563) PLAYER: APLL Dir: 1
I (61583) PLAYER: APLL Dir: 1
I (61603) PLAYER: APLL Dir: 1
I (61623) PLAYER: APLL Dir: 0
I (61643) PLAYER: APLL Dir: 0
I (61663) PLAYER: APLL Dir: 0
I (61683) PLAYER: APLL Dir: 1
I (61703) PLAYER: APLL Dir: 0
```

🙁 Did not work:
```
  gpio_reset_pin(GPIO_NUM_4);
  gpio_set_direction(GPIO_NUM_4, GPIO_MODE_OUTPUT);
  gpio_set_pull_mode(GPIO_NUM_4, GPIO_PULLDOWN_ONLY);
  gpio_set_level(GPIO_NUM_4, 0);
```
```diff
             dir = -1;
+            gpio_set_level(GPIO_NUM_4, 0);
           } else if ((shortMedian > shortOffset) && (miniMedian > miniOffset) &&
                      (avg > miniOffset)) {  // we are late
             dir = 1;
+            gpio_set_level(GPIO_NUM_4, 1);
           }
```

Will continue investigating - Thanks for your help so far! 😄

CarlosDerSeher, 2024-01-12:
There is a gpio_configure function you'll need to call first for the hpio to work.

CarlosDerSeher, 2024-02-02:
@BenediktMueller91 
I've pushed a new branch https://github.com/CarlosDerSeher/snapclient/commit/1e3bda9ad01025e566f8aadd6a705e0a76ba73b2

Where I've implemented a different syncing approach. It's still experimental, maybe you are willing to test.

BenediktMueller91, 2024-02-04:
@CarlosDerSeher 
I flashed my ESP32 with a build based on branch `sync_with_sample_stuffing`. Also i found a old bookshelf speaker in the basement. That removed the artifacts from the sqeaky speaker.
It now sounds really good. When playing a pure 440Hz sinus i can faintly hear the sample manipulations but nothing that i could hear in anything but such pure waveforms.
Initially samples were added/removed rapidly so I changed `shortOffset` and `miniOffset` a little to relax the timing. It now slowly drifts around and sometimes a sample is added or removed.
```diff
-      const int64_t shortOffset = 2;              // µs, softsync
-      const int64_t miniOffset = 1;               // µs, softsync
+      const int64_t shortOffset = 128;              // µs, softsync
+      const int64_t miniOffset = 64;               // µs, softsync
```

Thanks again for all your work and help :)

CarlosDerSeher, 2024-02-04:
Awesome :)

commit f56a421a (2024-02-04): fix #55
closed 2024-02-04 by CarlosDerSeher