CarlosDerSeher/snapclient #56: Compile for esp32-s3

ReasonableHippo, 2024-01-05:
Hi,
thanks for the fantastic work! I've been experimenting on getting this to compile for the esp32-s3 but i'm getting linker warnings
`/project/build/../components/custom_driver/i2s.c:1094: undefined reference to `rtc_clk_apll_enable'`

Some quick google search reveals that this wasn't implemented in 4.3. 
(https://github.com/espressif/esp-idf/issues/7114)
Also, someone mentions that 
`Exactly, I2S on ESP32-C3 and ESP32-S3 uses GDMA instead of a dedicated DMA, and APLL is useless on these two chips.`

I'm not super sure what the apll does and if its necessary for the algorithm to run or not. Could you give me some pointers on what would be the best way to move forward?
Thanks :)

CarlosDerSeher, 2024-01-05:
APLL tuning is strictly necessary for keeping clients synced.

I can't comment much on the other esp32 types as I've never used them, but the link you provided suggests using v4.4 could be the solution? Maybe try using this version and see if you can port the project? I plan an upgrade to v5 some time but not sure when. Maybe an upgrade to v4.4 is not that much of an hassle. Try it using a seperate branch and if you can do it I'll be happy to accept your pull request  :)

CarlosDerSeher, 2024-01-06:
If you want to put in some more effort there is an [experimental feature](https://github.com/CarlosDerSeher/snapclient/blob/master/components%2Flightsnapcast%2Fplayer.c#L32) which trys to insert/remove samples to keep up sync but I never could make it work right.

luar123, 2024-01-06:
I tried to compile for esp32-c3 once and got the same error. Did not try to resolve it as it was only out of curiosity and just decided to wait until idf 5 is supported.

CarlosDerSeher, 2024-01-06:
But wouldn't it still not work even if i2s is supported? Is there an APLLon those chips?

closed 2024-01-22 by CarlosDerSeher