CarlosDerSeher/snapclient #59: Lyrat Mini Board V1.2 not working

DerPicknicker, 2024-01-17:
Hi @CarlosDerSeher , 

I tested now your develop branch. Sadly the Board doesn't play audio. I attached the Log for you. Hope its an easy fix : 

I (1042) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1563) wifi:<ba-add>idx:0 (ifx:0, 44:4e:6d:20:fa:5b), tid:6, ssn:2, winSize:64
I (1744) wifi:<ba-add>idx:1 (ifx:0, 44:4e:6d:20:fa:5b), tid:0, ssn:0, winSize:64
I (2551) WIFI: Connected with IP Address:192.168.178.132
I (2551) esp_netif_handlers: sta ip: 192.168.178.132, mask: 255.255.255.0, gw: 192.168.178.1
I (2555) WIFI: connected to ap SSID: FRITZ!Box 7590
I (2561) SC: Connected to AP
I (2564) HTTP: Initializing SPIFFS
I (2611) HTTP: Partition size: total: 534881, used: 50200
I (2611) HTTP: Mount /html to storage success
I (2611) HTTP: d_name=/html/favicon.ico d_ino=0 d_type=1
I (2617) HTTP: d_name=/html/index.html d_ino=0 d_type=1
I (2623) HTTP: d_name=/html/ESP-LOGO.txt d_ino=0 d_type=1
I (2644) NETF: Setup mdns
I (2644) HTTP: Start http task=192.168.178.132
I (2644) HTTP: Starting server on 192.168.178.132:8000
I (2647) HTTP: Starting HTTP Server on port: '8000'
I (2655) SC: Lookup snapcast service on network
E (2657) esp_ota_ops: Running firmware is factory
I (2663) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (5745) SC: Found 192.168.178.125:1704
I (5750) SC: netconn connected
I (5751) SC: netconn sent hello message
I (5758) SC: Buffer length:  1000
I (5758) SC: Latency:        0
I (5758) SC: Mute:           0
I (5759) SC: Setting volume: 100
I (5762) DRV8311: Enter into es8311_mute(), mute = 0

I (5771) SC: fLaC sampleformat: 48000:16:2
I (5836) DRV8311: Enter into es8311_mute(), mute = 1

I (5842) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 4
I (5843) c_I2S: DMA Malloc info, datalen=blocksize=2304, dma_buf_count=4
I (5849) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (5862) LYRAT_MINI_V1_1: I2S0, MCLK output by GPIO0
I (5869) c_I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
W (5881) PLAYER: no pcm chunk queue created?
I (5886) PLAYER: created new queue with 40
I (5891) PLAYER: snapserver config changed, buffer 1000ms, chunk 1152 frames, sample rate 48000, ch 2, bits 16 mute 0 latency 0
I (5983) SC: latency buffer full
I (6977) DRV8311: Enter into es8311_mute(), mute = 0

I (6978) PLAYER: initial sync age: 2us, chunk duration: 24000us
E (7429) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (7547) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (7668) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8309) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8329) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8370) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8392) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8429) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8448) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8468) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8489) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8550) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8570) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8590) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8610) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8668) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8713) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8729) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8752) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8788) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8814) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
E (8829) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload
[... 120 lines cut ...]
I (29230) DRV8311: Enter into es8311_mute(), mute = 1

E (31236) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 2010284s, 790.792ms
I (31236) DRV8311: Enter into es8311_mute(), mute = 1

E (33242) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 2010284s, 790.792ms
I (33242) DRV8311: Enter into es8311_mute(), mute = 1

E (35248) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 2010284s, 790.792ms
I (35248) DRV8311: Enter into es8311_mute(), mute = 1

E (37254) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 2010284s, 790.792ms
I (37254) DRV8311: Enter into es8311_mute(), mute = 1

E (39260) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 2010284s, 790.792ms
I (39260) DRV8311: Enter into es8311_mute(), mute = 1

E (41266) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 2010284s, 790.792ms
I (41266) DRV8311: Enter into es8311_mute(), mute = 1

E (43272) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 2010284s, 790.792ms
I (43272) DRV8311: Enter into es8311_mute(), mute = 1

E (45278) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 2010284s, 790.792ms
I (45278) DRV8311: Enter into es8311_mute(), mute = 1

CarlosDerSeher, 2024-01-17:
> E (24227) PLAYER: Failed to heap_caps_malloc(4608, 3) memory for pcm chunk payload

These errors normally happen if you don't have enough RAM. I see you set 1000ms as buffer. Did you enable SPI RAM?

DerPicknicker, 2024-01-17:
Oh man. I thought setting the  board is enough... I will do it. And report

DerPicknicker, 2024-01-17:
Works fine. But sometimes I get this static noise ... when rebooting. And if now song is played for some time there is some crackle noise. I tested now 2 esp32 and it's also here not audible.

CarlosDerSeher, 2024-01-17:
> But sometimes I get this static noise ... when rebooting.
Yes I know about that, happened to me too sometimes. Some bug I couldn't pinpoint. It happens on my Lyrat v4.3 maybe we need to gracefully shutdown or something? Mostly happened after OTA when reboot was happening.

> And if no song is played for some time there is some crackle noise. 
So this is always there if no audio is played?

DerPicknicker, 2024-01-17:
>we need to gracefully shutdown or something? Mostly happened after OTA when reboot was happening.
Sometimes after reset / reboot .. I would suggest to shutdown all pins after start wait for 5-10ms and re-set them. 

> So this is always there if no audio is played?

Yes

CarlosDerSeher, 2024-01-17:
I think the reset doesn't stop the dac correctly but it keeps running and producing noise. Not sure how to solve that.

Regarding the crackling, is the dac muted while audio isn't played? Maybe try that if  it isn't

DerPicknicker, 2024-01-17:
I can try. But I think while music is playing I can hear that noise. So I would recommend to shutdown the dac if for 30sec+ no audio comes in.

luar123, 2024-01-17:
> So I would recommend to shutdown the dac if for 30sec+ no audio comes in.

That is on my list for the power saving PR, just had no time to work on it.

DerPicknicker, 2024-01-17:
Issue can be closed - it's working. I will test what a mute will change and update here. But in general the problem is resolved. Let's do the next step in multiroom Audio ;-)

DerPicknicker, 2024-01-18:
Hello @CarlosDerSeher , 

I have now more information about the structure of ESPHOME. 

The `loop` thread calls each component's 'loop()` method once per cycle which is something like 15 or 30ms, but there is no guarantee on this. It is possible to run the loop faster -  see https://github.com/esphome/esphome/blob/1324d9e39a4f7d850ba9fc013f3d0e412e4fa5b9/esphome/core/application.h#L175 

So we can adjust the speed of the Main-Loop which calls the snapclients loop. The suggestion was to use Core0 for tight timing. And Core1 for WiFi stuff. Which is handled by ESPHOME. 

Here is another example: 

This task takes items from a queue (which is thread-safe) and transmits them. The queue is filled from the main `loop()` thread and the higher priority task pulls from it. https://github.com/clydebarrow/esphome/blob/3e556136a59f32c68760389c45b9fdf046bfeddc/esphome/components/vnc/vnc_display.h#L373

What do you think - is there a chance that we can create a component for snapclient? Or do you don't like the concept of it and don't want to spend much time on it.

DerPicknicker, 2024-01-18:
The problem was the disabled psram in the build settings. This is why I created this issue. So from my perspective we can merge into master. The board works fine. I will do some longer testing. If you want to wait until I test it a bit more it's okay for me.

Sorry GitHub app said that I was in the correct thread. Annoying.

CarlosDerSeher, 2024-01-18:
So the crackling is still there but this is another problem?

DerPicknicker, 2024-01-18:
Yep crackling is still there. But I have to figure out if the device was muted or not. I guess it was **not** muted. I will create a new issue for it soon. I have to re-install my music server and do some
Other tasks before I can test this again. General the board works.

DerPicknicker, 2024-01-23:
Hi @CarlosDerSeher , 

just for your Information: Crackling seems to be issued by power supply. I can't reproduce static noise. It happens only sometimes. For me it's fine. You can merge it into master.

ReasonableHippo, 2024-06-29:
Hi @DerPicknicker ,
do you mind sharing your configuration for the Lyrat Mini v1.2 board?

Thanks