CarlosDerSeher/snapclient #60: IDF5 Building fails

DerPicknicker, 2024-01-23:
Hi, 

I cloned your IDF5 Branch and tried building it with the Docker-Image of IDF. But for me it doesn't work. 

Here is the Build-Log: 
 ```
error: unknown type name 'uint32_t'
   47 | audio_queue_handle_t audio_queue_create(uint32_t queue_len, uint32_t item_size);
      |                                         ^~~~~~~~
/project/components/audio_sal/include/audio_queue.h:1:1: note: 'uint32_t' is defined in header '<stdint.h>'; did you forget to '#include <stdint.h>'?
  +++ |+#include <stdint.h>
    1 | /*
/project/components/audio_sal/include/audio_queue.h:47:61: error: unknown type name 'uint32_t'
   47 | audio_queue_handle_t audio_queue_create(uint32_t queue_len, uint32_t item_size);
      |                                                             ^~~~~~~~
/project/components/audio_sal/include/audio_queue.h:47:61: note: 'uint32_t' is defined in header '<stdint.h>'; did you forget to '#include <stdint.h>'?
/project/components/audio_sal/include/audio_queue.h:72:22: error: unknown type name 'uint32_t'
   72 |                      uint32_t block_time_ms);
      |                      ^~~~~~~~
/project/components/audio_sal/include/audio_queue.h:72:22: note: 'uint32_t' is defined in header '<stdint.h>'; did you forget to '#include <stdint.h>'?
/project/components/audio_sal/include/audio_queue.h:88:31: error: unknown type name 'uint32_t'
   88 |                               uint32_t block_time_ms);
      |                               ^~~~~~~~
/project/components/audio_sal/include/audio_queue.h:88:31: note: 'uint32_t' is defined in header '<stdint.h>'; did you forget to '#include <stdint.h>'?
/project/components/audio_sal/include/audio_queue.h:105:22: error: unknown type name 'uint32_t'
  105 |                      uint32_t block_time_ms);
      |                      ^~~~~~~~
/project/components/audio_sal/include/audio_queue.h:105:22: note: 'uint32_t' is defined in header '<stdint.h>'; did you forget to '#include <stdint.h>'?
```

I used this Docker-Command to build: `docker run --rm -it -v .:/project -w /project -v /dev:/dev --privileged espressif/idf:v5.1.1`

Are you using a different Version in your local Installation?

CarlosDerSeher, 2024-01-23:
Ok I'll need to check when I at home. I'll update when I know more

closed 2024-01-23 by CarlosDerSeher

CarlosDerSeher, 2024-01-23:
@DerPicknicker seems the git commit hook sorted the includes differently and the wifi_interface.h got broke. Should build now

DerPicknicker, 2024-01-23:
@CarlosDerSeher .. I will try it soon.

DerPicknicker, 2024-01-24:
Works, I create a PR with updated docs and flashing instructions for macOS.

supermar1010, 2024-01-28:
Is this still building for you? I'm getting a bunch of weird errors, starting with a missing cmake version in the container. I just `ln -s`ed it, that allowed it to start building at least. Then I ran into some missing version of the esp32 configs (or something, "esp-2021r2-patch3-8.4.0" was missing), I `ln -s`ed it as well, then I could actually build it. But it does not run

I tried using the idf-5.1.1 container as well as the release-v5.1 one. Is the build working for you?

```
E (1154) i2s_common: i2s_new_channel(812): no available channel found
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x400f7bf8
0x400f7bf8: player_setup_i2s at /project/components/lightsnapcast/player.c:213

file: "./components/lightsnapcast/player.c" line 213
func: player_setup_i2s
expression: i2s_new_channel(&tx_chan_cfg, &tx_chan, NULL)

abort() was called at PC 0x4008c7cb on core 0
0x4008c7cb: _esp_error_check_failed at /opt/esp/idf/components/esp_system/esp_err.c:50

Backtrace: 0x40081ce1:0x3ffbb910 0x4008c7d5:0x3ffbb930 0x40090a59:0x3ffbb950 0x4008c7cb:0x3ffbb9c0 0x400f7bf8:0x3ffbb9f0 0x400f7f43:0x3ffbbaa0 0x400da9e3:0x3ffbbaf0 0x40196c9a:0x3ffbbba0
0x40081ce1: panic_abort at /opt/esp/idf/components/esp_system/panic.c:472
0x4008c7d5: esp_system_abort at /opt/esp/idf/components/esp_system/port/esp_system_chip.c:84
0x40090a59: abort at /opt/esp/idf/components/newlib/abort.c:38
0x4008c7cb: _esp_error_check_failed at /opt/esp/idf/components/esp_system/esp_err.c:50
0x400f7bf8: player_setup_i2s at /project/components/lightsnapcast/player.c:213
0x400f7f43: init_player at /project/components/lightsnapcast/player.c:348
0x400da9e3: app_main at /project/main/main.c:2847
0x40196c9a: main_task at /opt/esp/idf/components/freertos/app_startup.c:208 (discriminator 13)
```

DerPicknicker, 2024-01-28:
Did you init the update the submodules?

CarlosDerSeher, 2024-01-28:
Seems your log isn't complete? Please post the first build error

supermar1010, 2024-02-04:
This the complete log from the esp32. Looking at it again, it looks like it got something to do with my i2s setup?

My ports are:
master i2s bck: 26
master i2s lrck: 25
master i2s data out: 32
The rest of the ports is set to -1

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7428
load:0x40078000,len:16652
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:4284
entry 0x40080668
I (29) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (29) boot: compile time Feb  4 2024 15:46:37
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v1.0
I (37) qio_mode: Enabling QIO for flash chip GD
I (42) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (87) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 factory          factory app      00 00 00010000 00120000
I (102) boot:  4 ota_0            OTA app          00 10 00130000 00120000
I (110) boot:  5 ota_1            OTA app          00 11 00250000 00120000
I (117) boot:  6 storage          Unknown data     01 82 00370000 00090000
I (125) boot: End of partition table
I (129) boot: Defaulting to factory image
I (134) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=30688h (198280) map
I (197) esp_image: segment 1: paddr=000406b0 vaddr=3ffb0000 size=03700h ( 14080) load
I (202) esp_image: segment 2: paddr=00043db8 vaddr=40080000 size=0c260h ( 49760) load
I (219) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=c5b48h (809800) map
I (442) esp_image: segment 4: paddr=00115b70 vaddr=4008c260 size=0798ch ( 31116) load
I (462) boot: Loaded app from partition at offset 0x10000
I (462) boot: Disabling RNG early entropy source...
I (474) cpu_start: Multicore app
I (474) quad_psram: This chip is ESP32-D0WD
I (474) esp_psram: Found 4MB PSRAM device
I (476) esp_psram: Speed: 80MHz
I (480) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
I (487) cpu_start: Pro cpu up.
I (491) cpu_start: Starting app cpu, entry point is 0x40081400
0x40081400: call_start_cpu1 at /opt/esp/idf/components/esp_system/port/cpu_start.c:154

I (483) cpu_start: App cpu up.
I (957) esp_psram: SPI SRAM memory test OK
I (965) cpu_start: Pro cpu start user code
I (965) cpu_start: cpu freq: 240000000 Hz
I (965) cpu_start: Application information:
I (968) cpu_start: Project name:     snapclient
I (973) cpu_start: App version:      1
I (978) cpu_start: Compile time:     Feb  4 2024 15:46:32
I (984) cpu_start: ELF file SHA256:  5c459e74feb5a6dc...
I (990) cpu_start: ESP-IDF:          v5.1.1
I (995) cpu_start: Min chip rev:     v0.0
I (999) cpu_start: Max chip rev:     v3.99 
I (1004) cpu_start: Chip rev:         v1.0
I (1009) heap_init: Initializing. RAM available for dynamic allocation:
I (1016) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1022) heap_init: At 3FFB9E78 len 00026188 (152 KiB): DRAM
I (1029) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1035) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1042) heap_init: At 40093BEC len 0000C414 (49 KiB): IRAM
I (1048) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (1056) spi_flash: detected chip: gd
I (1060) spi_flash: flash io: qio
I (1064) app_start: Starting scheduler on CPU0
I (1069) app_start: Starting scheduler on CPU1
I (1069) main_task: Started on CPU0
I (1079) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1087) main_task: Calling app_main()
I (1112) SC: Start codec chip
I (1113) HAL: INIT
E (1113) PCM5102A: Mute gpio config failed for pin -1
I (1113) HAL: codec_hal done
I (1117) AUDIO_BOARD: board-handle done
I (1122) SC: Audio board_init done
I (1126) AUDIO_HAL: Codec mode is 3, Ctrl:1
I (1130) SC: init player
I (1134) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 4
W (1141) i2s_common: i2s controller 0 has been occupied by i2s_driver
E (1148) i2s_common: i2s_new_channel(812): no available channel found
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x400f7928
0x400f7928: player_setup_i2s at /project/components/lightsnapcast/player.c:213

file: "./components/lightsnapcast/player.c" line 213
func: player_setup_i2s
expression: i2s_new_channel(&tx_chan_cfg, &tx_chan, NULL)

abort() was called at PC 0x4008c24f on core 0
0x4008c24f: _esp_error_check_failed at /opt/esp/idf/components/esp_system/esp_err.c:50

Backtrace: 0x40081ce5:0x3ffbb500 0x4008c259:0x3ffbb520 0x400904c9:0x3ffbb540 0x4008c24f:0x3ffbb5b0 0x400f7928:0x3ffbb5e0 0x400f7c73:0x3ffbb690 0x400da763:0x3ffbb6e0 0x4019176a:0x3ffbb790
0x40081ce5: panic_abort at /opt/esp/idf/components/esp_system/panic.c:452

0x4008c259: esp_system_abort at /opt/esp/idf/components/esp_system/port/esp_system_chip.c:84

0x400904c9: abort at /opt/esp/idf/components/newlib/abort.c:38

0x4008c24f: _esp_error_check_failed at /opt/esp/idf/components/esp_system/esp_err.c:50

0x400f7928: player_setup_i2s at /project/components/lightsnapcast/player.c:213

0x400f7c73: init_player at /project/components/lightsnapcast/player.c:348

0x400da763: app_main at /project/main/main.c:2847

0x4019176a: main_task at /opt/esp/idf/components/freertos/app_startup.c:208 (discriminator 13)

ELF file SHA256: 5c459e74feb5a6dc

Rebooting...
```

supermar1010, 2024-02-04:
I just tried building and flashing the lyrat v4.3 that works (kind of at least).
So either there was something wrong with my menuconfig or with the custom board target in general

DerPicknicker, 2024-02-04:
What type of board are you using? I would recommend start using the Lyrat config and update your changes in menuconfig.

supermar1010, 2024-02-04:
I'm using an esp32-wrover module. I compiled for lyrat v4.3 now, and that kind of works. But as described in my other issue, with breaks in playback. 

I actually did not change anything about the lyrat config (I just changed the hardware connections to match the lyrat board ones)

DerPicknicker, 2024-02-04:
@supermar1010 ...
As I am not @CarlosDerSeher .. but I think I had the same issue. Try this: https://github.com/CarlosDerSeher/snapclient/pull/63#issuecomment-1917783529

supermar1010, 2024-02-04:
@DerPicknicker yes that fixes the build but unfortunately not my sound issue :/ Thank you though!!

DerPicknicker, 2024-02-05:
@CarlosDerSeher .. Should I do the same fix for my Custom driver?

CarlosDerSeher, 2024-02-05:
I've done this fix to idf5 branch as well, so when we merge it should be ok. Your dac doesn't need mclk for initialization. It is only needed for adau1961 currently

DerPicknicker, 2024-02-05:
Good to know. I try my best to finish it this week.

commit 7150a483 (2024-01-23): fix #60
commit ef3583c5 (2024-01-23): fix #60
commit cfd34a8e (2024-01-23): fix #60
commit 9ce3ffe7 (2024-02-04): fix #60