CarlosDerSeher/snapclient #64: Lyra Mini Board Mute Bug

DerPicknicker, 2024-01-30:
Hi @CarlosDerSeher , 

you added in the development branch the support for the Lyrat Mini Board. I found a Bug which should be fixed before merging into master. 

If you stop the playback and want to play another song the DAC is not sending the Audio correctly.

Here I paused the Song: 
```
W (287114) PLAYER: RESYNCING HARD 2: age -1us, latency 614125314277us, free 4343794, largest block 4128768, 0, rssi: -55
I (287115) DRV8311: Enter into es8311_mute(), mute = 1
E (289123) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 614125s, 314.277ms
I (289123) DRV8311: Enter into es8311_mute(), mute = 1
```

Here I played another Song: 

```
I (301414) SC: fLaC sampleformat: 48000:16:2
I (303276) DRV8311: Enter into es8311_mute(), mute = 0
I (303277) PLAYER: initial sync age: 2us, chunk duration: 24000us
```

But **only if you change the volume up or down** the music starts playing.
```
I (307729) SC: Buffer length:  1500
I (307729) SC: Latency:        0
I (307729) SC: Mute:           0
I (307730) SC: Setting volume: 44
I (309381) SC: Buffer length:  1500
I (309381) SC: Latency:        0
I (309381) SC: Mute:           0
I (309382) SC: Setting volume: 50
I (315382) SC: Buffer length:  1500
I (315382) SC: Latency:        0
I (315382) SC: Mute:           0
I (315383) SC: Setting volume: 42
```

Only the Lyrat Mini board has the problem. My other Board doesn't have this problem at all. So might be only released to the custom board config of the Lyrat Mini Board 1.1 / 1.2.

Thank you for your great support. I really like the snap cast thing.

CarlosDerSeher, 2024-01-30:
@DerPicknicker seems to me current volume isn't saved anywhere inside the mute function? https://github.com/CarlosDerSeher/snapclient/blob/713cda5816605bac1b50ff998af6f81f64bbe77a/components/audio_hal/driver/es8311/es8311.c#L332 I can't find a good datasheet with register description but I guess the mute function should probably look something like this:

```
static uint8_t currentVolume;

/*
 * set es8311 dac mute or not
 * if mute = 0, dac un-mute
 * if mute = 1, dac mute
 */
static void
es8311_mute (int mute)
{
  uint8_t regv;
  ESP_LOGI (TAG, "Enter into es8311_mute(), mute = %d\n", mute);
  regv = es8311_read_reg (ES8311_DAC_REG31) & 0x9f;
  if (mute)
    {
      es8311_codec_get_voice_volume(&currentVolume);

      es8311_write_reg (ES8311_SYSTEM_REG12, 0x02);
      es8311_write_reg (ES8311_DAC_REG31, regv | 0x60);
      es8311_write_reg (ES8311_DAC_REG32, 0x00);
      es8311_write_reg (ES8311_DAC_REG37, 0x08);
    }
  else
    {
      es8311_write_reg (ES8311_DAC_REG31, regv);
      es8311_write_reg (ES8311_SYSTEM_REG12, 0x00);

      es8311_codec_set_voice_volume(currentVolume);
    }
}
```

Could you please try this?

DerPicknicker, 2024-01-30:
Hi @CarlosDerSeher , 

I store the current volume in my custom board as a variable inside the driver. I update it every time I change the volume. I will try your changes. and update here.

DerPicknicker, 2024-01-30:
So.. Some Minutes later I tested your code above: 
Here was the old song playing: 
`I (17062) PLAYER: initial sync age: 2us, chunk duration: 24000us`
**Now I started a new Song:**
```
I (32353) SC: fLaC sampleformat: 48000:16:2
I (32361) SC: fLaC sampleformat: 48000:16:2
W (34961) PLAYER: RESYNCING HARD 2: age -740416us, latency 621172902900us, free 4194582, largest block 3997696, 28, rssi: -46
I (34963) DRV8311: Enter into es8311_mute(), mute = 1

I (35776) DRV8311: Enter into es8311_mute(), mute = 0

I (35778) PLAYER: initial sync age: 2us, chunk duration: 24000us
```
The Song plays for 2sec makes half a second a brake (silence) and then goes forward. So the Lyrat Board has a complete different behaviour than my custom board LOL.

CarlosDerSeher, 2024-01-30:
Where does this take us now?

DerPicknicker, 2024-01-30:
That the Lyrat board is different from the other boards. I can't hear that silence on my custom board. I will check that.

DerPicknicker, 2024-03-11:
Hi @CarlosDerSeher .. What information do you need from my side to fix this issue?

CarlosDerSeher, 2024-03-11:
I don't really understand what's the issue. Also I can't test

DerPicknicker, 2024-03-11:
My custom board (TAS5805M) starts playing immediately and doesn't mute after 2sec... 

I will test it today again and try to describe it a bit more.

DerPicknicker, 2024-03-11:
Hi, 

the issue can be marked as resolved. The Problem is the volume control. Below 60% the audio output is not useable. It's only noise. But no music. The Mute thing was a problem related to my snap cast installation. But the Volume is really buggy. IF I increase it above 80% it becomes noisy again. So I only have around 20% of the useable volume settings available.... 

Do you have an Idea here?

CarlosDerSeher, 2024-03-11:
Maybe check adf driver as a reference? Also you can try adjusting analog gain settings

DerPicknicker, 2024-03-17:
Hi, 

I will skip that Board and just use it with the current limitations. I will focus more on the TOSLINK-Way. Issue can be closed.

closed 2024-03-17 by DerPicknicker

ReasonableHippo, 2024-06-30:
I have the same mute issue on that board. Could you share what you changed in your snapserver config?
Thanks

DerPicknicker, 2024-06-30:
@ReasonableHippo .. Where does your music come from. Do you use a standard snapcast server?

ReasonableHippo, 2024-07-03:
Yeah its a standard server, but I'm getting the data from a pipe which is probably the issue.

DerPicknicker, 2024-07-04:
Hmm.. I use MusicAssistant.. so maybe it's serversided. But let's see.

ReasonableHippo, 2024-11-03:
i came back to this after months and have found a working solution on a lyrat mini v1.2.
using the config from here:
https://github.com/sonocotta/esparagus-snapclient/blob/master/configs/sdkconfig.lyra-t-mini

The mute bug is caused by 
`es8311_write_reg(ES8311_DAC_REG32, 0x00);` in the es8311.c mute function
commenting that out resulted in wanted behaviour for mute/unmute and starting up unmuted

CarlosDerSeher, 2024-11-03:
Great news, thanks for your investigations