CarlosDerSeher/snapclient #65: Constant audio cutouts esp32 wrover

supermar1010, 2024-02-04:
Hello,
I'm running the idf5 branch, but this was also present on the dev branch. Playback starts and is

CarlosDerSeher, 2024-02-04:
@supermar1010 your board does not initialize right. You got a lot of i2c bus errors in the beginning and audio hal init fails therefore. I wonder why you are even able to hear any sound at all.

Which board are you using? I can't remember adding support for this particular dac yet. Did you do it yourself?

supermar1010, 2024-02-04:
My DAC does not have i2c. That's why I'm getting the errors I suppose.

It is a Sabre ES9023 (https://pdf1.alldatasheet.com/datasheet-pdf/download/485885/ESS/ES9023.html).

CarlosDerSeher, 2024-02-04:
You should chose custom board and then could try using PCM5102A dac driver. From what I see in the datasheet it works very similar like yours.

supermar1010, 2024-02-04:
I was using the PCM5102A in the beginning, but had the same issues, so I tried the Lyrat Board just to see if it made any difference.

After the hint in the other issue I disabled that line and was able to build and run it, but with the same issue :/

CarlosDerSeher, 2024-02-04:
Please post the log from PCM5102A then. Also you could try using the sample_stuffing branch. Some users reported problems with their DACs when using apll tuning for syncing.

supermar1010, 2024-02-05:
This is the PCM5102A log

```
: segment 2: paddr=00044478 vaddr=40080000 ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7428
load:0x40078000,len:16652
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:4284
entry 0x40080668
I (28) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (29) boot: compile time Feb  4 2024 15:46:37
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v1.0
I (37) qio_mode: Enabling QIO for flash chip GD
I (42) boot.esp32: SPI Speed      : 80MHz
I (46) boot.esp32: SPI Mode       : QIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (79) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (87) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  3 factory          factory app      00 00 00010000 00120000
I (102) boot:  4 ota_0            OTA app          00 10 00130000 00120000
I (109) boot:  5 ota_1            OTA app          00 11 00250000 00120000
I (117) boot:  6 storage          Unknown data     01 82 00370000 00090000
I (124) boot: End of partition table
I (129) boot: Defaulting to factory image
I (133) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=30d48h (200008) map
I (197) esp_image: segment 1: paddr=00040d70 vaddr=3ffb0000 size=03700h ( 14080) load
I (201) esp_image: segment 2: paddr=00044478 vaddr=40080000 size=0bba0h ( 48032) load
I (218) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=c8dd0h (822736) map
I (445) esp_image: segment 4: paddr=00118df8 vaddr=4008bba0 size=0804ch ( 32844) load
I (466) boot: Loaded app from partition at offset 0x10000
I (466) boot: Disabling RNG early entropy source...
I (477) cpu_start: Multicore app
I (477) quad_psram: This chip is ESP32-D0WD
I (478) esp_psram: Found 4MB PSRAM device
I (479) esp_psram: Speed: 80MHz
I (483) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
I (491) cpu_start: Pro cpu up.
I (494) cpu_start: Starting app cpu, entry point is 0x40081400
0x40081400: call_start_cpu1 at /opt/esp/idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (960) esp_psram: SPI SRAM memory test OK
I (968) cpu_start: Pro cpu start user code
I (968) cpu_start: cpu freq: 240000000 Hz
I (968) cpu_start: Application information:
I (971) cpu_start: Project name:     snapclient
I (977) cpu_start: App version:      1
I (981) cpu_start: Compile time:     Feb  4 2024 17:09:12
I (987) cpu_start: ELF file SHA256:  f906095cd9f16eee...
I (993) cpu_start: ESP-IDF:          v5.1.1
I (998) cpu_start: Min chip rev:     v0.0
I (1003) cpu_start: Max chip rev:     v3.99 
I (1008) cpu_start: Chip rev:         v1.0
I (1012) heap_init: Initializing. RAM available for dynamic allocation:
I (1020) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1026) heap_init: At 3FFB9EB0 len 00026150 (152 KiB): DRAM
I (1032) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1039) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[... 73 lines cut ...]
I (3285) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (3290) main_task: Returned from app_main()
I (3290) SC: Lookup snapcast service on network
I (6369) SC: Found 192.168.2.101:1704
I (6415) SC: netconn connected
I (6417) SC: netconn sent hello message
I (6426) SC: Buffer length:  1000
I (6426) SC: Latency:        0
I (6426) SC: Mute:           0
I (6426) SC: Setting volume: 29
I (6430) dspProc: Set volume to 0.290000
I (6434) dspProc: Set volume to 0.290000
I (6440) SC: fLaC sampleformat: 44100:16:2
I (6472) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 4
E (6472) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
W (6478) PLAYER: no pcm chunk queue created?
I (6482) PLAYER: created new queue with 37
I (6486) PLAYER: snapserver config changed, buffer 1000ms, chunk 1152 frames, sample rate 44100, ch 2, bits 16 mute 0 latency 0
I (6639) SC: latency buffer full
E (6649) gptimer: gptimer_stop(356): timer is not running
E (6649) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (7602) PLAYER: initial sync age: 7us, chunk duration: 26122us
```

The sample stuffing branch works flawless. So it seems to be the apll tuning stuff.