CarlosDerSeher/snapclient #70: Intermittent loss of higher frequencies during playback

LeoSum8, 2024-04-03:
Hi all,
first of all thank you for continuing Jorgen's work and continuously advancing the snapcast ecosystem!

I have compiled @DerPicknicker's [TAS5805M-branch](https://github.com/DerPicknicker/snapclient/tree/TAS5805M) and flashed it to my [Louder ESP32](https://www.tindie.com/products/sonocotta/louder-esp32/).

Overall, thanks to the good documentation that is provided here, that was a smooth and easy process.

The only issue that I am currently facing is that during playback I intermittently loose high frequencies for a few seconds until they come back for the next few seconds. It sounds like the tweeters of my speakers have a loose connection. But that is not the case, as this "sound effect" does not occur when I use different firmware (e.g. [euphonium](https://github.com/LeoSum8/euphonium_louderESP) or [Phil Schatzmann's snapclient](https://github.com/pschatzmann/arduino-snapclient/))

The serial output doesn't show anything helpful during these intermittent tweeter-losses.

Has anyone experienced anything similar or does anyone have an Idea what might be the problem here? Other than that, playback is nicely in sync and I have no stuttering.

During the tests I used flac codec on the snapserver.

DerPicknicker, 2024-04-03:
Hi @LeoSum8 ..

Cool that you got it working with my code. The problem is already known and caused by the current implementation of the sync. It will be fixed soon @CarlosDerSeher is already working on a different syncing approach. If you want to fix it - see my closed pull request there is a temporarily fix for disabling the APLL stuff.

CarlosDerSeher, 2024-04-03:
by disabling apll you will loose client syncing. Please try the branch https://github.com/CarlosDerSeher/snapclient/tree/sync_with_sample_stuffing

LeoSum8, 2024-04-03:
Hi guys and thank you for the prompt replies!
I am glad to hear that this is a well known problem.
I'll have a closer look at it hopefully tomorrow. I guess I'll need to merge the TAS5805M stuff from DerPicknicker into that branch first. Hopefully the different idf-versions don't make this a big deal. 

I'll let you know how it goes.

DerPicknicker, 2024-04-03:
I think not. You just clone the branch which linked @CarlosDerSeher …

DerPicknicker, 2024-04-03:
@CarlosDerSeher we Are replying almost at the same time.. If this happens more often than people will think we are one person  ;-)

LeoSum8, 2024-04-03:
Ah, I see. So you merged your TAS5805m stuff into that branch already, but not into master yet?

I had first tried to use Carlos' master after I had discovered a TAS5805m driver in the sources, but was very disappointed to find out that I couldn't configure it before building.

CarlosDerSeher, 2024-04-04:
It will be in master soon see #69

For now use the branch from that pull request

LeoSum8, 2024-04-04:
> You'll need to enable the sample_stuffing through menuconfig

By this, do you mean "use sample insertion" within the snapclient configuration?

LeoSum8, 2024-04-04:
So this seems to have worked nicely. 
Thanks! 

Initially I had the problem that it would only play music upon boot for half a second or so and then keep repeating this log output:
`E (564228) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 703808s, 197.544ms`

I think I resolved it by setting the sntp server in menuconfig to my local fritzbox.
But when I pause the stream I am still getting the abovementioned error output. But resuming works without any problem.

I'll listen to music the rest of the day to test further ;)

LeoSum8, 2024-04-04:
Oh, and I noticed that opus doesn't work as it causes a bootloop as soon as the first packet arrives at the client. But I guess that is known? Flac works without a problem so far.

LeoSum8, 2024-04-04:
The short music-burp, followed by silence that I had thought to have fixed via the ntp server setting came back when I changed streams. It seems to appear quite randomly but I could get the client to play when it booted into an idle stream and I only started playback after it had connected. But still then I had problems switching streams.

I think I fixed it now by setting all streams to a sampling rate of 44100:16:2. Before, some had 48000. 48000 should actually have been my global setting but somehow librespot doesn't seem to pick it up and just does 44100.

LeoSum8, 2024-04-04:
Also, what does `diff2Server: 703808s` actually mean? That's about a month :)

LeoSum8, 2024-04-04:
I hadn't set the chunk size manually but did so now. Isn't 20 the default if not set otherwise?

So far it seems to work good. I only have server-side stuttering now, but that is out of scope here, I guess.

Or does anyone have pro-tips for docker-based snapserver that stutters when other containers cause load? :)

LeoSum8, 2024-04-04:
I saw you mentioning that one. I already tried it but am not convinced of it yet :)
I don't understand how it is organized yet and can't really tell what music is from what source. But what was like magic to me is that when I set it up to use my already existing snapserver, it could stream to single clients by creating a new source that can also be seen in the android snapclient app. How does it do that?!

My snapserver setup is a bit more complicated with a couple of other sources I'd like to keep (jellyfin, subsonic, bluetooth, ...). I am not sure if those could be added to the MusicAssistant Snapserver too.

Anyway, this is getting off-topic and is actually solved. I'll close here. Thank you all for the support!!!!

But do you have other means of contact? You could e.g. find me on matrix at @leosum:matrix.org. I'd like to chat about the louder ESP32 and what you did with it so far. Github doesn't have direct messages...

closed 2024-04-04 by LeoSum8

DerPicknicker, 2024-04-04:
@LeoSum8 .. See my discussion here in the repo. Create a new thread and I will respond.

CarlosDerSeher, 2024-04-04:
@LeoSum8 

> Isn't 20 the default if not set otherwise?

Yes it is, glad to read all is resolved now