CarlosDerSeher/snapclient #72: Web installer for snapcast

anabolyc, 2024-04-23:
Hey folks.

I'm a developer of a few audio boards that work with snapclient, including [Louder-esp32](https://www.tindie.com/products/sonocotta/louder-esp32/) and [Esparagus](https://www.crowdsupply.com/sonocotta/esparagus-media-center) set of boards. I'm currently testing them together with the Home Assistant. It really works great, sync is beyond belief.

I went a little further and made a [web-flasher](https://www.crowdsupply.com/sonocotta/esparagus-media-center) that will have pre-built configurations for all my boards that support snapclient (since the build process is not for faint-hearted). 

The fundamental problem that I have is that WiFi credentials are baked into binary and there is no way I can provide a web-flasher to anyone else to use since it will never work in their wifi network. Ideally, I'd like to have [improv-wifi](https://github.com/improv-wifi/) integrated, for web-installer would handle wifi credentials right after firmware flashing. Documentation is very poor though, the only signal that it is possible is that ESPHome made it work somehow.

I wonder if you guys think about this issue and perhaps have a direction you'd like to go to solve it. 

My work-in-progress branch is [here](https://github.com/sonocotta/esparagus-snapclient/tree/features/web-installer)

CarlosDerSeher, 2024-04-23:
@anabolyc without looking to much at your provided links I'd suggest using [WiFi Provisioning](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/provisioning/wifi_provisioning.html) maybe? This is already implemented through WiFi but could be extended to use the serial console as an input too. So after flashing the FW you could set the credentials from your web installer through the serial connection too maybe?

CarlosDerSeher, 2024-04-23:
Also beware of the very soon coming upgrade of master to IDF v5

DerPicknicker, 2024-04-24:
@CarlosDerSeher .. Could you build the WiFi provisioning with the latest branch as I remember the building failed due wrong partition config. Could you please build it and verify that it works? 

I tried that some time ago and if I remember correctly I wasn't able to build it. 

I like those webinstaller because you can setup WiFi also directly in the browser. So would be Cool if this can be integrated into the current code. 

For board from @anabolyc and the boards from espressif it would reduce the amount of time drastically. Just install it with your browser and set it up.

CarlosDerSeher, 2024-04-24:
I didn't try recently but thanks for the pointer

DerPicknicker, 2024-04-25:
@CarlosDerSeher .. 

I can built your latest branch (sync_with_sample_stuffing) but we should change the boot loader size a little bit: 
Generated /project/build/bootloader/bootloader.bin
0x8000 bootloader 0x1000 /project/build/bootloader/bootloader.binBootloader binary size 0x6f30 bytes. 0**xd0 bytes (1%) free.**

What happens if wifi connection is not possible - will the wifi provisioning start again? This would be a good way to changing the wifi without re-flashing the board.

@anabolyc .. Easiest way is to build the firmware with the wifi-provisioning to make it easy to change wifi.

CarlosDerSeher, 2024-04-25:
@DerPicknicker please open an seperate issue with the relevant Information from here.

DerPicknicker, 2024-04-25:
@CarlosDerSeher I will do it. Only for Bootloader or for both?

CarlosDerSeher, 2024-04-25:
bootloader. 

> What happens if wifi connection is not possible - will the wifi provisioning start again? This would be a good way to changing the wifi without re-flashing the board.

This should be a feature request possibly?

> See this https://github.com/jnthas/Improv-WiFi-Library this would allow to add the Wifi-Credentials directly in the browser. I see here some benefit. You're not relying on the APP from Espressif.

Is there a C library for this too?

CarlosDerSeher, 2024-04-25:
@anabolyc What do you think about providing the credentials through the serial interface? Wouldn't that be more convenient to the user?

anabolyc, 2024-04-25:
> @anabolyc What do you think about providing the credentials through the serial interface? Wouldn't that be more convenient to the user? Also have a look here #75

@CarlosDerSeher I need to look into [WiFi Provisioning](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/provisioning/wifi_provisioning.html), not sure how it integrates into esp-web-tools. The beauty of the [improv-wifi](https://github.com/improv-wifi/) is that it is just another step in flashing firmware with esp-web-tools, with wifi-scanning and everything in a single UI. If Wifi provisioning works the same, I'd prefer it, since it comes from the Espressif.

> > See this https://github.com/jnthas/Improv-WiFi-Library this would allow to add the Wifi-Credentials directly in the browser. I see here some benefit. You're not relying on the APP from Espressif.
> 
> Is there a C library for this too?

I think that's all she wrote, however, folks at ESPhome managed to wrap it up into [espidf C component](https://github.com/esphome/esphome/tree/dev/esphome/components/esp32_improv)

CarlosDerSeher, 2024-04-25:
Maybe [esp_prov](https://github.com/espressif/esp-idf/tree/v5.2.1/tools/esp_prov) will be of help too. A python utility for provisioning from a PC

CarlosDerSeher, 2024-04-25:
[esp_prov](https://github.com/espressif/esp-idf/tree/v5.2.1/tools/esp_prov) could be of interest for this. a python util implementing the protocol.

anabolyc, 2024-05-09:
Continuing discussion from [here](https://github.com/CarlosDerSeher/snapclient/issues/75#issuecomment-2101365885)

I see the config for the LyraT board (it si the same for the mini version?). I guess I'll remove wifi credentials in favor of wifi provisioning, also PSRAM should be enabled I guess. 

@CarlosDerSeher  Any other boards/configs you'd like to include?

CarlosDerSeher, 2024-05-09:
I didn't test all of them but all those ADF boards (in sample stuffing branch) compile so maybe those?! I'll check my custom design using adau1961 dac, this too maybe

DerPicknicker, 2024-05-09:
Hi @anabolyc , 

I added in the menuconfig the option to invert the logic-Levels. I think for the Louder it would be good if we have those options: 
- Stereo-Mode 
- Bridge-Mode (playing left channel) 
- Bridge-Mode-Invert (playing right channel)

I added only one SDKCONFIG but this is an easy way to create an completely wireless stereo setup with more power output.

anabolyc, 2024-05-14:
Hey folks. I've added common Espressif boards and AI-Thinker as well. The result is in the branch still, please have a look before I create a pull request: https://sonocotta.github.io/esparagus-snapclient/
If you can, please test with your boards to confirm the build.

> * Stereo-Mode
> * Bridge-Mode (playing left channel)
> * Bridge-Mode-Invert (playing right channel)

I would leave that to more experienced developers to do manually. All the possible configurations would overload the installer too much

DerPicknicker, 2024-05-14:
Great!!! that's awesome. 

How is the general process looking? @CarlosDerSeher will merge the sync with sample stuff and the WiFi improv. 

What is needed to get fresh binary's for those boards?

anabolyc, 2024-05-14:
> Edit: please try the softvolume if not already enabled for the HiFi version. That should allow controlling the volume even without hardware control.

Where I can enable that?

DerPicknicker, 2024-05-14:
> Where I can enable that?

Should be inside the DSP section .. If I remember correctly.

anabolyc, 2024-05-14:
I added short [usage instructions](https://github.com/sonocotta/esparagus-snapclient/blob/features/installer-improved/doc/web-installer.md)

DerPicknicker, 2024-05-14:
@anabolyc did you found the option?

anabolyc, 2024-05-14:
> @anabolyc did you found the option?

Yes, enabled and flashed. It works fine, but did I miss it before? seems to be working exactly the same way.

DerPicknicker, 2024-05-14:
So volume control works fine? 

@CarlosDerSeher could you give us here some details?

CarlosDerSeher, 2024-05-14:
Which DAC we are talking about?

DerPicknicker, 2024-05-14:
PCM5100A from this device: [see here ](https://sonocotta.com/esparagus-hifi-medialink/)

anabolyc, 2024-05-14:
> So volume control works fine?
> 
> @CarlosDerSeher could you give us here some details?

Yes, I can change the volume from Music Assistant. Also, equilizer works
[image]

CarlosDerSeher, 2024-05-14:
So this hasn't hardware volume control and soft volume isn't working as expected? Did I get this right?

DerPicknicker, 2024-05-14:
@anabolyc said that softvolume was working already without the option enabled in DSP settings that's why I asked if you @CarlosDerSeher could give us some more deteails... 

@anabolyc while this works fine now I am even more thinking about getting one for my remaining speakers ;-)

CarlosDerSeher, 2024-05-14:
Shouldn't work without soft volume enabled. At least not when using snapserver volume control. Maybe Music Assistant applies software volme control too? You should see this in the logs though.

anabolyc, 2024-05-14:
When I change tho volume in MA, i see this on logs
<code><pre>
I (9249) PLAYER: initial sync age: 9us, chunk duration: 24000us
I (12803) SC: Buffer length:  1000
I (12803) SC: Latency:        0
I (12803) SC: Mute:           0
I (12803) SC: Setting volume: 44
I (12807) dspProc: Set volume to 0.440000
I (16076) SC: Buffer length:  1000
I (16077) SC: Latency:        0
I (16077) SC: Mute:           0
I (16077) SC: Setting volume: 52
I (16081) dspProc: Set volume to 0.520000
I (16435) SC: Buffer length:  1000
I (16435) SC: Latency:        0
I (16435) SC: Mute:           0
I (16435) SC: Setting volume: 54
I (16439) dspProc: Set volume to 0.540000
</pre></code>

CarlosDerSeher, 2024-05-14:
Try disabling it again and you'll know for sure :)

anabolyc, 2024-05-14:
You're right, when disabled it logs changes but doesn't actually change the volume 
<pre>
I (6522) PLAYER: initial sync age: 8us, chunk duration: 24000us
I (7995) SC: Buffer length:  1000
I (7995) SC: Latency:        0
I (7995) SC: Mute:           0
I (7995) SC: Setting volume: 24
I (9750) SC: Buffer length:  1000
I (9750) SC: Latency:        0
I (9750) SC: Mute:           0
I (9750) SC: Setting volume: 96
I (12070) SC: Buffer length:  1000
I (12070) SC: Latency:        0
I (12070) SC: Mute:           0
I (12071) SC: Setting volume: 12
I (13267) SC: Buffer length:  1000
I (13268) SC: Latency:        0
I (13268) SC: Mute:           0
I (13268) SC: Setting volume: 26
I (14115) SC: Buffer length:  1000
I (14115) SC: Latency:        0
I (14115) SC: Mute:           0
I (14116) SC: Setting volume: 44
I (14909) SC: Buffer length:  1000
I (14909) SC: Latency:        0
I (14909) SC: Mute:           0
I (14910) SC: Setting volume: 16
I (15380) SC: Buffer length:  1000
I (15381) SC: Latency:        0
I (15381) SC: Mute:           0
I (15381) SC: Setting volume: 10
</pre>

Does I hook up to the hardware volume control on the TAS5805M DAC?

DerPicknicker, 2024-05-14:
@anabolyc .. TAS5805M works fine. Tested here. It's hardware .

anabolyc, 2024-05-14:
I've merged the installer to my [master](https://github.com/sonocotta/esparagus-snapclient). Please feel free to pull the code upstream

CarlosDerSeher, 2024-05-15:
Also I saw you based this on current master. After merging "sync with sample stuffing" probably some changes will be required from your side.

anabolyc, 2024-05-15:
That's all right, just let me know when it will be merged

CarlosDerSeher, 2024-06-12:
@anabolyc totaly forgot about it. All merged now. So you can test with master and see if everything still works. Please open a pull request when you are finished

anabolyc, 2024-06-14:
Hey @CarlosDerSeher thanks for letting me know.

I pulled the latest changes from master, my build bootlooping when ethernet is enabled. I will disable it for now, but maybe you know what did I miss in config

[image]

CarlosDerSeher, 2024-06-14:
So all good then? Wrong defaults?

anabolyc, 2024-06-14:
> So all good then? Wrong defaults?

Not entirely. With ethernet enabled, wifi-improv is not triggering, I cannot setup Wifi. Is this by design?

anabolyc, 2024-06-14:
@CarlosDerSeher  I've merged the master to my fork with web-installer, and tested with Esparagus boards. I don't think it makes sense to do PR back to your master, there are hundreds of build errors. It would be simpler to copy over installer files or leave alone installer in my fork.

oliv3r, 2024-07-22:
As I mentioned in #92 using ESPhome as a basis, would offer us a lot of these things out of the box :) EVEN when not connecting to home-assistant ... (you can leave out the `api:` but from the yaml definition to not even talk to home-assistant).