CarlosDerSeher/snapclient #73: Compiled and flashed but no audio out on PCM5201A

leinaht, 2024-04-23:
I followed [these](https://github.com/pschatzmann/ESP32-A2DP/wiki/External-DAC#pcm5102-dac) wiring and GPIO pin instructions and have tweaked my latency and buffer settings to the point where I don't see errors in my logs, but still hear nothing from the audio jack. I may be confused by the I2S Master and I2S Slave settings - here is what I have for master:

> (0) Master i2s mclk
(26) Master i2s bck
(25) Master i2s lrck
(22) Master i2s data out

And for slave:

> (26) Slave i2s bck
(25) Slave i2s lrck
(22) Slave i2s data out

What am I missing?

CarlosDerSeher, 2024-04-23:
You onlyneed the Master interface. Please set save pins to -1

CarlosDerSeher, 2024-04-23:
Also it seems to me sck should be vonnected to mclk so gpio0?
https://www.ti.com/lit/gpn/pcm5102a

leinaht, 2024-04-23:
I initially had sck connected to gnd because that's what was suggested in the ESP32-A2DP repo (and my board has space for a solder jumper there as well). My board only exposes GPIO1 out of 0-2 (and I get an error when assigning it to a higher pin) but when I designate 1 as the MCLK pin boot hangs at this line:

`c_I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: `

It does boot and seem to pick up a value from somewhere even with MCLK set to 0 in the config and pin connected to ground.

leinaht, 2024-04-23:
I was actually misreading the error message when assigning the MCLK pin - it specified pins 0,1, or 3:

`GENERIC_BOARD: Only support GPIO0/GPIO1/GPIO3, gpio_num:2`

Unfortunately I tested with 3 and got the same result.

CarlosDerSeher, 2024-04-24:
What about the mute pin?

CarlosDerSeher, 2024-04-24:
You'll also need to enable soft volume from dsp processor in menuconfig as this particular dac doesn't come with i2c volume control

luar123, 2024-04-24:
I am usingen the same DAC module and can send you my config when I am back home.

leinaht, 2024-04-24:
Solved! It was a hardware issue. There are pins that need to be bridged on this specific board to allow for the soft volume control (marked in the attached image). I had also left the mute pin at the default of 18 and changed it to -1 but not sure if that was part of the issue as well. So my final values are

master I2S:

>(0) Master i2s mclk
(26) Master i2s bck
(25) Master i2s lrck
(22) Master i2s data out

slave I2S:

>(-1) Slave i2s bck
(-1) Slave i2s lrck
(-1) Slave i2s data out

TI PCM5102A interface Configuration:

>(-1) Master mute/unmute for PCM5102A

With jumper 3 bridged to H on the back of the board (marked in red on the attached image)
[image]

Thanks!

luar123, 2024-04-24:
With jumper 3 you unmute the DAC. if you leave it open you need to connect the mute gpio defined on ESP32 to the xsmt pin. You should also close the other jumpers or connect the coresonding pins (see datasheet for functions). Sck is not needed.