CarlosDerSeher/snapclient #74: Increase Bootloader Size (Changing Offset) - Only 1% left for Bootloader when using Wifi provisioning

DerPicknicker, 2024-04-25:
Hi, 

as discussed. Here is the Issue for the small Bootloader Size. I can build the code. But I would recommend adding some safe space for the Bootloader. I think 1% is too small. This only occurs with the wifi provisioning. Tested with sync_with_sample_stuffing.

Generated /project/build/bootloader/bootloader.bin
0x8000 bootloader 0x1000 /project/build/bootloader/bootloader.binBootloader binary size 0x6f30 bytes. 0xd0 bytes (1%) free.

DerPicknicker, 2024-04-25:
I think a small buffer would be good. For the reason that if you want to enable secrurity features like encrypted flash, secure boot or if the firmware gets bigger you have some safe space. In my Project I try to reserve 5% free for the Bootloader. So if I want to add some more features I can reduce the Bootloader size. So this is only a personal preference. 

I don't want to criticise you - just as an input why it might be useful to have some buffer. ;-)

CarlosDerSeher, 2024-04-25:
It's no problem to increase bootloader partition as we got plenty space left since I removed the factory partition entirely :)

CarlosDerSeher, 2024-04-28:
Ok, so what do you think about just removing those (unnecessary) logs. If I set the default output to ERROR i have 12% to spare in the bootloader partition

DerPicknicker, 2024-04-28:
Great, I guess this is ideal. Wasn't aware that logging takes so much space ...

CarlosDerSeher, 2024-04-28:
this will be resolved with upcoming merge of #69

closed 2024-04-28 by CarlosDerSeher