CarlosDerSeher/snapclient #75: Feature Request: Improve WiFi-Provisioning

DerPicknicker, 2024-04-25:
Hi @CarlosDerSeher , 

in the other Issue we discussed the Wifi-provisoning. I think two options would increase the usability a lot. 

The current Situation doesn't allow changing Wifi-SSIDs / Passwords without flashing the board again. 

Option 1: 
Using the current implementation and start the Provisioning again after 3-5 failed connection attempts. 

Option 2: 
Use this Library: https://github.com/jnthas/Improv-WiFi-Library and support the Improv-Protocol. Big benefit is that you don't need the Espressif app and you can flash it with your browser and setup the wifi as well in your browser. I did a small research and I can't find a pure C-Version of the Library. I guess a lot of Code should be working also in pure C. Only the Vectors might be Problem if I am informed correctly.

CarlosDerSeher, 2024-04-25:
Also I saw you can also use serial connection with Improv lib to provision wifi

DerPicknicker, 2024-04-25:
Yes that's the main usecase. The webflasher supports that. So you can flash it and directly set up the WiFi.

DerPicknicker, 2024-04-25:
I think the benefit is that we can build the image with the GitHub action (I will try to build them after syncing is merged). And the webinstaller will automatically have the latest versions online.

CarlosDerSeher, 2024-04-25:
Let's see what @anabolyc thinks about the espressif implementation and if he wants to use that before we put any effort into this.

anabolyc, 2024-04-25:
> I think the benefit is that we can build the image with the GitHub action (I will try to build them after syncing is merged). And the webinstaller will automatically have the latest versions online.

I tried to do that in the other project with artifacts taken from the release, which in turn were generated by GitHub action using the release tag. The problem is that esp-web-tools would only allow firmware to be pulled from the same domain. So I had to place binaries in the repository itself. It is possible to commit them into the repo using GitHub action, it is a bit clumsy but should work.

In any case, until WiFi provisioning is in place, it would not work anyway. I'll try to look into that before EOW.

anabolyc, 2024-04-26:
@CarlosDerSeher @DerPicknicker I just tried example implementation of [WiFi Provisioning](https://github.com/espressif/esp-idf/tree/master/components/wifi_provisioning). Book [example](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr) have only BLE and SoftAP methods implemented, and they both require installing an app on your phone (an extra device) and not particularly elegant.

The Console option seems to be functionally equivalent to [improv-wifi](https://github.com/improv-wifi) but it is not supported by ESP Web Tools. I [asked](https://github.com/esphome/esp-web-tools/issues/508) folks if they are planning to add it, but I would not expect this to happen soon.

So I still think improv-wifi is the best option, as long as it is possible to wrap it up into esp-idf component.

anabolyc, 2024-04-26:
@CarlosDerSeher here is the ESPHome implementation demo. Just to give you an idea how easy this is to flash and configure wifi in the same UI
[image]

CarlosDerSeher, 2024-04-26:
@anabolyc they use BLE as a transport though don't they?

anabolyc, 2024-04-26:
> @anabolyc they use BLE as a transport though don't they?

If you're asking about ESPHome, they use Serial-console provisioning. Espressif's implementation has 3 options, BLE is one of them. Bear in mind, it is not available for all ESP32 versions.

CarlosDerSeher, 2024-04-26:
I've been looking at [this example](https://github.com/esphome/esphome/tree/dev/esphome%2Fcomponents%2Fesp32_improv) and it seems to me it is for BLE provisioning. Do you have a link where serial provisioning using improv is shown?

DerPicknicker, 2024-04-26:
> Ok, found some more [documentation](https://www.improv-wifi.com/serial/). Still, an example would be great.

What is the problem by using the library which I linked? Seems fairly easy to port to pure c.

CarlosDerSeher, 2024-04-26:
Hahaha ok my brain completely forgot about the arduino implementation :)

DerPicknicker, 2024-04-26:
@CarlosDerSeher out of sync => hardsync 😉

DerPicknicker, 2024-04-30:
@CarlosDerSeher .. Great to hear. 

I guess [here](https://web.esphome.io) if you connect you should be able to provision the device.

CarlosDerSeher, 2024-04-30:
Ok that's a start, but I guess there is no possibility to debug with this approach... I have no option to see what's going on on the serial port. If there is no way to generate defined packets so I can step manually through the provisioning process I guess my only option is to attach another (two) usb2serial RX line to my dev board... Any other suggestions how to proceed smarter here?

DerPicknicker, 2024-04-30:
You can use putty to connect to the port and send them manually or you could check the browser developer settings if there is some output what is going on. I guess there should be some logging.

Alternatively print out each packet which is sent to your dev-board like a loopback device.

DerPicknicker, 2024-04-30:
See this [taken from here (282)](https://esphome.io/api/improv__serial__component_8cpp_source):
 
```
 void ImprovSerialComponent::send_response_(std::vector<uint8_t> &response) {
   std::vector<uint8_t> data = {'I', 'M', 'P', 'R', 'O', 'V'};
   data.resize(9);
   data[6] = IMPROV_SERIAL_VERSION;
   data[7] = TYPE_RPC_RESPONSE;
   data[8] = response.size();
   data.insert(data.end(), response.begin(), response.end());
 
   uint8_t checksum = 0x00;
   for (uint8_t d : data)
     checksum += d;
   data.push_back(checksum);
 
   this->write_data_(data);
 }
```

CarlosDerSeher, 2024-04-30:
Yeah, already saw this right after my post. thanks though :)

CarlosDerSeher, 2024-05-02:
@anabolyc @DerPicknicker just a little teaser, I got improv service working and I am able to get credentials through https://web.esphome.io/ into my esp32. What's missing now is integration into wifi_interface component. Hopefully I'll be able to push the branch by the end of the week :)

DerPicknicker, 2024-05-02:
Really Cool! This would be great after that I will do my best to create an GitHub action to build automatically binary's for the web-installer

anabolyc, 2024-05-02:
@CarlosDerSeher that's really cool. Can't wait to try it out. @DerPicknicker I will try to help there, as I want to integrate it into [my web installer](https://sonocotta.github.io/esparagus-snapclient/)

CarlosDerSeher, 2024-05-03:
@DerPicknicker @anabolyc ok implementation is still very error prone as most edge cases aren't tested checked for. For example I don't know what will happen if wrong credentials are supplied. Also there are some dirty delays in the code which should be done better. IP retrieval isn't very good inside wifi_provisioning component. some more work is needed here. But it works mostly and you can try it out and maybe leave some comments on how to improve the implementation.

DerPicknicker, 2024-05-03:
Hey, that sounds great.

I checked your main.c file .. From my perspective we should think about a different approach. 

Some time ago I used the FreeRtos really heavily. I think we should create a task for WiFi provisioning and other tasks. 

Then you could pin the task to a specific core. Inside the task I would create an timerInterrupt which triggers every 5second to check if new serial data is available. Then stop WiFi and get the new SSID / password try to connect. 

That task could also handle the WiFi-Reset feature. That could run independently from the SNAPCAST task so that shouldn't be interrupted by the other task if you pin it to a specific core. 

What do you think about my thoughts @CarlosDerSeher ?

CarlosDerSeher, 2024-05-03:
I am not exactly sure what you mean. Provisioning is a task of it's own. It is only started if the device isn't provisioned already. Uart data is received using events, a queue which blocks until the idf driver unblocks, I guess from an interrupt. No need to check every 5s.

DerPicknicker, 2024-05-03:
Okay let's try to rephrase it : 

I meant that I would regularly check for provisioning data because I changed my WiFi now, I need can just plug the esp32 into my pc and send new WiFi data. Now you can only do it if the device has no WiFi data... That's why I would create a new task that check regularly for new data. 

I would create a task that checks for it every x seconds or checks if new serial data is available doesn't know if there is an api for a serial interrupt. 

You start 2 tasks in the main OTA and HTTP, nothing more?

CarlosDerSeher, 2024-05-03:
Most tasks are started in some init function.

 If we disable logging we could leave provisioning active all the time but I am not sure if this is desirable. Currently logging is only disabled when provisioning is active. So press reset button a couple of times and you will be able to provision your device again.

DerPicknicker, 2024-05-04:
If I pressed the button 3x will also start the Espressif provisioning and the improv WiFi? Or only the last one? 

The only problem I see is if a board doesn't have the button you won't be able to restart the provisioning. That's why I thought about to check from time to time.

CarlosDerSeher, 2024-05-04:
You got a point. Some time ago I found a component which enables logging to a network port. Maybe ghis will be a possible route to go. So disable logging to serial but enable network logs?!

DerPicknicker, 2024-05-04:
@CarlosDerSeher ... 

See here https://github.com/VedantParanjape/esp-wifi-logger

CarlosDerSeher, 2024-05-04:
I'd prefer to not wear out the flash by writing all those logs to it. I think it will be better put to RAM and presented on a udp port or similar. I'll see how much RAM I am willing to sacrifice for that :)

DerPicknicker, 2024-05-05:
Good point. Do we really need logs? I think not really . If the code is stable enough I would just disable them completely.

CarlosDerSeher, 2024-05-05:
@DerPicknicker ok I pushed some changes to the branch because I found this whole discussion about logging obsolete. I tried provisioning and logs always on, and what can I say... it just works together. The improv header "IMPROV" ensures all other data sent is discarded by the web config interface.

DerPicknicker, 2024-05-05:
@CarlosDerSeher .. That sounds great. When do you think it will be merged? After syncing is merged?

CarlosDerSeher, 2024-05-05:
Yes we can merge this together or merge beforehand to sync branch doesn't matter

anabolyc, 2024-05-07:
Hey @CarlosDerSeher @DerPicknicker I can confirm that wifi-provisioning works via web-installer. Great work!
I will try integrating it into the pipeline and let you know how it goes

CarlosDerSeher, 2024-05-07:
@anabolyc I think the implementation might change at some point though as I don't like how I've done it currently. For now I guess it is good enough.

anabolyc, 2024-05-07:
I mostly have it working, just one thing concerns me. Do I need to configure PSRAM or it is assumed on every board?

DerPicknicker, 2024-05-07:
@anabolyc .. Could you share how you did it? Because it would be Cool if we can do that for all supported boards (lyrat, Lyrat Mini, etc.)

anabolyc, 2024-05-07:
@CarlosDerSeher @DerPicknicker 

Here is the [pipeline](https://github.com/sonocotta/esparagus-snapclient/blob/master/.github/workflows/build-binaries.yml)

The idea is that whenever code is committed to the feature branch, the pipeline fires up and builds firmware for defined boards sdkconfigs (I added three Esparagus boards for now). Built images are committed to the same branch and manifest files are updated.

Then I need to pull the latest changes into my feature branch for testing locally and it can be merged to master, where the pipeline is not enabled.

Earlier I added a [web installer](https://github.com/sonocotta/esparagus-snapclient/tree/master/installer) that defines UI for the web installer app. Built artifacts and manifests are published to [docs](https://github.com/sonocotta/esparagus-snapclient/tree/master/docs) folder, which is automatically published to GitHub pages.

Comitting binaries approach is not very neat, but esp-web-tools require firmware files to be served from the same domain (allowed cross-domain sources to be precise), so I cannot pull binaries from build artifacts or release files, they have to be published together with GitHub pages. An alternative approach is to host binaries outside GitHub, but I don't want to maintain and pay for external infra.

anabolyc, 2024-05-07:
@CarlosDerSeher I think I mentioned before, that I can share the Esparagus or Louder-ESP board with you if you plan to support and use it. I appreciate your work here and that's the least I can do to support you. Obviously, @DerPicknicker already has one :)

DerPicknicker, 2024-05-07:
Thank you for the detailed explanation - maybe @CarlosDerSeher can also share his thoughts. I think the Snapcast-Repo should include the Webinstaller for all boards as it's the place where all changes are happening. 

So maybe we can add a GitHub page to this repo where we can get the builded binary's. That includes all Louder-Board and also the Lyrat boards. 

That's only an strategic discussion;-) 

@anabolyc I use your board quite often for testing it's great.

CarlosDerSeher, 2024-05-07:
@anabolyc thanks I appreciate your offer. As I already have my own line level solution the Louder-ESP would be of interest to me. Maybe it is able to replace  my justboom amp hat rpi player. I doubt it though as it probably hasn't enough juice for outside playback in my garden. But for sure there will be a spot for it :)

You are selling these players commercially in Europe? How do you handle CE?

CarlosDerSeher, 2024-05-07:
> Comitting binaries approach is not very neat, but esp-web-tools require firmware files to be served from the same domain (allowed cross-domain sources to be precise), so I cannot pull binaries from build artifacts or release files, they have to be published together with GitHub pages.

I don't really get this part. Normally I would think the best way is to use github releases to host pre-built binaries

anabolyc, 2024-05-07:
> I don't really get this part. Normally I would think the best way is to use github releases to host pre-built binaries

I started with that approach. Problem is releases are served from **github.com** domain. Github pages (and thus installer) from **github.io** domain. JS code cannot fetch files cross-domain unless CORS headers are present, and they are not in fact.

anabolyc, 2024-05-07:
> @anabolyc thanks I appreciate your offer. As I already have my own line level solution the Louder-ESP would be of interest to me. Maybe it is able to replace my justboom amp hat rpi player. I doubt [...]

Pls drop me a line to andriy@sonocotta.com, I'll need your address for shipping. Just to be sure, Louder-ESP is a bare board, while Louder Esparagus is a cased solution, but both have external wifi antenna.

> You are selling these players commercially in Europe? How do you handle CE?

I worked with Crowd Supply on CE/UKCA compliance, they have good lawyers :)

DerPicknicker, 2024-05-08:
> Thank you for the detailed explanation - maybe @CarlosDerSeher can also share his thoughts. I think the Snapcast-Repo should include the Webinstaller for all boards as it's the place where all changes are happening. 

@CarlosDerSeher .. What do you think about this topic?

CarlosDerSeher, 2024-05-08:
You mean the source code only or also a hosted version on github.io where actions push latest binaries?

DerPicknicker, 2024-05-08:
The second part - the changes happen here. So why not build news binary's when changes are made.

CarlosDerSeher, 2024-05-08:
We could do that if someone is willing to put in the effort to make it work. What are the requirements for this exactly?

CarlosDerSeher, 2024-05-08:
Also shouldn't this discussion continue in #72?

closed 2024-05-08 by CarlosDerSeher

anabolyc, 2024-05-08:
> We could do that if someone is willing to put in the effort to make it work. What are the requirements for this exactly?

I can do that. Just give me a list of boards that you're willing to support, I'll add them to my wen installer and push changes upstream

DerPicknicker, 2024-05-08:
@anabolyc 

Lyrat 4.3 @CarlosDerSeher used this

Lyrat Mini 1.3 I used this board. 

I would also recommend creating a binary for generic boards like those boards which doesn't require any special configuration (i2c, psram) ..

commit a619587c (2024-06-12): Improv wifi (#85)

* - add wifi credential reset
  o press reset button (nRESET pin) 3 times
    but wait about 1s between button presses
    the button press counter is reset 5s after boot
* add wifi provisioning service through improv wifi (fix #75)