CarlosDerSeher/snapclient #76: Lots of Resyncing Hard 1 messages

unknown0816, 2024-04-27:
Hi,
I had a board working well, but lately there is no sound coming from the board. I connected the board to the monitor and I saw that there are lots of resyncing hard 1 messages.  What does this mean? Why did this happen so sudden?
fyi: I'm using the master.

W (20038) PLAYER: RESYNCING HARD 1: age 92490us, latency 3234843641211us, free 197272, largest block 110592, rssi: -53
W (20048) PLAYER: RESYNCING HARD 1: age 78124us, latency 3234843641211us, free 197508, largest block 110592, rssi: -53
W (20059) PLAYER: RESYNCING HARD 1: age 65430us, latency 3234843641211us, free 202268, largest block 110592, rssi: -53
W (20078) PLAYER: RESYNCING HARD 1: age 35762us, latency 3234843641211us, free 202268, largest block 110592, rssi: -53
W (20095) PLAYER: RESYNCING HARD 1: age 29417us, latency 3234843641211us, free 202268, largest block 110592, rssi: -53
W (20116) PLAYER: RESYNCING HARD 1: age 25705us, latency 3234843641211us, free 202268, largest block 110592, rssi: -53
W (20137) PLAYER: RESYNCING HARD 1: age 23405us, latency 3234843641211us, free 202268, largest block 110592, rssi: -53
W (20156) PLAYER: RESYNCING HARD 1: age 18235us, latency 3234843641211us, free 202268, largest block 110592, rssi: -53
W (20195) PLAYER: RESYNCING HARD 1: age 33560us, latency 3234843641211us, free 202268, largest block 110592, rssi: -53
W (20217) PLAYER: RESYNCING HARD 1: age 30858us, latency 3234843641211us, free 202268, largest block 110592, rssi: -53
W (20237) PLAYER: RESYNCING HARD 1: age 27153us, latency 3234843641211us, free 202268, largest block 110592, rssi: -53
W (20256) PLAYER: RESYNCING HARD 1: age 22219us, latency 3234843641211us, free 202268, largest block 110592, rssi: -52
W (20276) PLAYER: RESYNCING HARD 1: age 18446us, latency 3234843641211us, free 202268, largest block 110592, rssi: -52
W (20315) PLAYER: RESYNCING HARD 1: age 33343us, latency 3234843641211us, free 202268, largest block 110592, rssi: -52
W (20336) PLAYER: RESYNCING HARD 1: age 30086us, latency 3234843641211us, free 202268, largest block 110592, rssi: -52

CarlosDerSeher, 2024-04-27:
From the little Information you supply I'd say your neighbour is using his microoven a lot currently :)

unknown0816, 2024-04-28:
There is no nearby neighbour. I tested multiple Boards and just the one is not working. And i tried multiple locations. 
What Information do you need to provide help?
Or is the Board broken?

CarlosDerSeher, 2024-04-28:
Which board? Are they all using the same firmware and are they all in the same location?

CarlosDerSeher, 2024-05-08:
@unknown0816 is this resolved or still a problem?

CarlosDerSeher, 2024-05-13:
Ok, I'll close here then as there is no response anymore. Let me know if the problem persits

closed 2024-05-13 by CarlosDerSeher

unknown0816, 2024-06-17:
@CarlosDerSeher Sorry, I was really busy :/

I tested it again today and the problem seems to be still there. But I fear it is a delay-/wifi/or similar problem.

However, I guess you can just keep this closed. Thanks for the help.