CarlosDerSeher/snapclient #77: Board config for PT8211

whc2001, 2024-04-30:
I wanted to use PT8211 as it's ultra cheap (clones like TM8211 is even cheaper) and easy to use. However, it is using something called "Least Significant Bit Justified (LSBJ)" instead of standard I2S. I have checked the [Arduino library](https://github.com/schreibfaul1/ESP32-audioI2S/blob/1d9b299d0197f99fc70335295adcd226dc220f88/src/Audio.cpp#L4929C9-L4947) and it has something like this:

```cpp
m_i2s_config.communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_MSB);
```

However it's directly accessing an `i2s_config_t`. I checked the board definition and seems like it's using ESP-ADF. The most close thing I found is `audio_hal_codec_i2s_iface_t` which does not expose communication format settings. I am wondering whether this can be supported at all with ESP-ADF?

Also a side question that's a little meta assuming the primary question is solvable: this DAC does not have any kind of controlling interface, even a single mute pin. However I think a mute output might still needed for the amplifier part if it exists. Should the board config include a mute pin config even if the DAC does not support it?

whc2001, 2024-04-30:
Nevermind, the I2S part is not related to board config at all...

https://github.com/CarlosDerSeher/snapclient/blob/master/components%2Flightsnapcast%2Fplayer.c#L161

Hmmm not sure what to do, maybe use a macro here to replace the default I2S mode if another is set?

whc2001, 2024-04-30:
Nevermind, the I2S part is not related to board config at all...

https://github.com/CarlosDerSeher/snapclient/blob/fd701a1ead2fdfdc10ec0fc1bd7a92bc93f3c10d/components/lightsnapcast/player.c#L161-173

Hmmm not sure what to do, maybe use a macro here to replace the default I2S mode if another is set?

CarlosDerSeher, 2024-04-30:
> Should the board config include a mute pin config even if the DAC does not support it?

Doesn't hurt to have it I guess. Set it's default value to -1 (unused) and check before gpio initialization. 

Regarding i2s communication format, this is a problem I don't have an elegant solution for yet. Most DACs support I2S anyways, so I didn't think much about it. Using a config value, as you suggest, could be a way to slove this.

CarlosDerSeher, 2024-04-30:
Please also check out the branch sync_with_sample_stuffing. Currently development is mostly happening there and it uses idf v5 and also more up to date adf v2.6 . It will be merged soon and if you plan to open a pull request it would  be great to use this branch as your code base.

whc2001, 2024-05-01:
> Doesn't hurt to have it I guess. Set its default value to -1 (unused) and check before gpio initialization.
>
> Regarding i2s communication format, this is a problem I don't have an elegant solution [...]

Got it. I am not thinking about making it a discrete option in the menuconfig, instead just define it in the board config as it's DAC chip specific. Sadly I am not very familiar with ESP-IDF so I need to explore a bit.

CarlosDerSeher, 2024-05-01:
Did you check out custom_board component? Maybe add it there for this specific DAC?

whc2001, 2024-05-01:
> Did you check out custom_board component? Maybe add it there for this specific DAC?

Yep, by board config I mean add a new custom board and define there's then listen the define using `#ifdef` in player.c. I don't have a computer right now and I'll try asap.