CarlosDerSeher/snapclient #79: Ethernet

LeoSum8, 2024-05-04:
Hi,
is any way of adding Ethernet connection currently supported? I noticed a setting in the menuconfig.
Does anyone have any experience with that?

There are some examples around for connecting w5500 boards to esp32, but I guess that would need to be supported by the firmware.

CarlosDerSeher, 2024-05-04:
Currently only LAN8720 is tested. I made a custom board for that which works well. I think I use gpio16 for the MAC clock out. Also some additional gpio was needed to reset lan8720 after boot to initialize it right with the correct strap options.

Not sure how well the other PHYs will work. I think they should work but you'll have to try.

LeoSum8, 2024-05-04:
So e.g. this one would work out of the box?
https://www.ebay.de/itm/285834564622

Do you have some more documentation on how you hooked it up?

CarlosDerSeher, 2024-05-04:
I am not at home currently. But I think you could check the default values in menuconfig for LAN8720, these should reflect my setup

CarlosDerSeher, 2024-05-04:
@LeoSum8 Here goes the pinout for LAN8720:

ESP32 <--> LAN8720
IO25  <--> RXD0
IO26  <--> RXD1
IO27  <--> CRS_DV
IO23  <--> MDC
IO22  <--> TXD1
IO21  <--> TX_EN
IO19  <--> TXD0
IO18  <--> MDIO
IO17  <--> ETH_nRESET (LAN8720 reset pin)
IO16  <--> nREFCLOCKO_EN (LAN8720 LED2/NINTSEL pin)
IO0    <--> REFCLKO

Also my PHYs LED1 pin is pulled low through a LED + resistor and MDIO is pulled high through a resistor. Not sure how technically skilled you are but here is the schematic:

[image]

CarlosDerSeher, 2024-05-04:
As you can only output RMII clock using GPIO0 through APLL you will be restricted to DACs which do not require MCLK or you'll have to choose a LAN8720 board which comes with its own clock source and use IO0 as an input (as I did). There will probably be needed some adjustments to make it work. Probably just disable this part here as it is specific to how I designed my board (https://github.com/CarlosDerSeher/snapclient/blob/fd701a1ead2fdfdc10ec0fc1bd7a92bc93f3c10d/main/main.c#L2751)

CarlosDerSeher, 2024-05-04:
I just saw the board you linked got a oscillator on the bottom. So yes, probably it will work  but don't take this for granted

LeoSum8, 2024-05-05:
Thank you so much for all the details!
I ordered the board. When it arrives, I will try to set it up.
While doing so, I could add to this project's documentation if that is of interest to you

CarlosDerSeher, 2024-05-05:
Documentation is a bit scarce on some parts so that would ne great 👍

whc2001, 2024-05-06:
> As you can only output RMII clock using GPIO0 through APLL you will be restricted to DACs which do not require MCLK or you'll have to choose a LAN8720 board which comes with its own clock source and [...]

I also found this problem when designing my board. Basically the ESP32 can only accept the external sync clock from GPIO0, however during boot you are basically playing roulette with half the chance to stuck. I'll try later to tie GPIO5 to the EN pin of the active crystal oscillator.

whc2001, 2024-05-06:
> Should work if it goes to HIGH Z then

After checking the datasheet of ESP32 and LAN8720, I just realized I did it wrong and you have used an easier solution. 

What I'm planning to do is to have an active oscillator (more expensive, requires power and provides `EN` input) and have ESP32 control the `EN` pin. However ESP32 have GPIO5 pulled high when booting, while the oscillator will be enabled when EN is high, which is reversed from what we want.

What you did, which is to control the `nINTSEL` (H=ClockIn L=ClockOut) pin using GPIO5 is absolutely genius. When the pin is internally pulled HIGH during boot, LAN8720 expects active clock input and stops working. After the code is executed and the pin pulled LOW in user application, LAN8720 just jumps to life.

I think this also eliminates the need of an active oscillator with `EN` pin so I can just use a normal passive crystal which is cheaper. I'll change my board design and test everything (including the pending PT8211) after back home.

CarlosDerSeher, 2024-05-06:
Works like a charme for me with the crystal :)

DerPicknicker, 2024-05-10:
Would a W5500 work or is a lot of coding required to get it running?

CarlosDerSeher, 2024-05-10:
I can't really tell but it should work. Check out https://github.com/CarlosDerSeher/snapclient/tree/sync_with_sample_stuffing/components%2Feth_interface

Espressif supports it, I just took their code.

DerPicknicker, 2024-05-11:
Hi, Cool. I will get one and give feedback.

LeoSum8, 2024-05-13:
Out of curiosity: Why do you want to go with the W5500 instead of lan8720 if you don't have one laying around already? :)

My lan8720 has not yet arrived :/

DerPicknicker, 2024-05-13:
@LeoSum8 ... My EspLouder has PinHoles matching the W5500-Boards so no need for much diy.

LeoSum8, 2024-05-14:
> @LeoSum8 Here goes the pinout for LAN8720:
> ...
> 
> Also my PHYs LED1 pin is pulled low through a LED + resistor and MDIO is pulled high through a resistor. Not sure how technically skilled you are but here is the schematic:
> ...

My ETH8720 Board just arrived and I am trying to follow your description and schematic to match the correct pins before I hook it up to my ESP32 via jumper wires.

So this is what I matched so far:

| ESP32 | your LAN8720 | my LAN8720 | 
| ------------- | ------------- | ------------- |
| IO25 | RXD0 | RX0
| IO26 | RXD1 | RX1
| IO27 | CRS_DV | CRS
| IO23 | MDC | MDC
| IO22 | TXD1 | TX1
| IO21 | TX_EN | TX_EN
| IO19 | TXD0 | TXD
| IO18 | MDIO | MDIO
| IO17 | ETH_nRESET (LAN8720 reset pin) | ???
| IO5 | nREFCLOCKO_EN (LAN8720 LED2/NINTSEL pin) | ???
| IO0 | REFCLKO | nINT/RETCLK

So it seems that I am missing easy access to ETH_nRESET and nREFCLOCKO_EN pins.
Will this be a problem?

I think I am skilled enough to generally read the schematic, but I cannot see these two aspects you describe in the schematic.

> Also my PHYs LED1 pin is pulled low through a LED + resistor and MDIO is pulled high through a resistor.

Here's a pic of my board:
[image]

And the crystal on the bottom:
[image]

LeoSum8, 2024-05-14:
> @LeoSum8 ... My EspLouder has PinHoles matching the W5500-Boards so no need for much diy.

So you must have Rev. E then. I just checked, mine are D without these PinHoles.

CarlosDerSeher, 2024-05-14:
> Will this be a problem?

@LeoSum8 if you want to control it like I do you need those two pins or it won't work

Have a look at the [datasheet](https://ww1.microchip.com/downloads/aemDocuments/documents/UNG/ProductDocuments/DataSheets/LAN8720A-LAN8720Ai-Data-Sheet-DS00002165.pdf) page 6. There you see the pinout. It is pin2 on the LAN8720a which is nREFCLOCKO_EN (LAN8720 LED2/NINTSEL pin) in your table. youll have to find out how it is connected. I guess there is no schematic available for your board? Aslo check TABLE 2-2: 

> nINTSEL
This configuration strap selects the mode of the
nINT/REFCLKO pin.
• When nINTSEL is floated or pulled to
VDD2A, nINT is selected for operation on the
nINT/REFCLKO pin (default).
• When nINTSEL is pulled low to VSS, REF-
CLKO is selected for operation on the nINT/
REFCLKO pin

A few commets above a good [description](https://github.com/CarlosDerSeher/snapclient/issues/79#issuecomment-2096227204) of what's happening with this pin is given. The reset pin (pin 15) is also needed for this to work.

I'd suggest checking those pins and follow their tracks. Maybe you can remove some resistor and solder a wire instead. Hopwfully you are equipped with a soldering iron :)

If you not you are not totally stuck but your only option will be to provide MAC clock from ESP32 --> to LAN8720 and change the code of I2S to use fixed mclk and use sample stuffing for synchronization. Not sure if your board supports clock in for the PHY as it has the oscillator already on it.

LeoSum8, 2024-05-14:
Thank you so much for holding my hand here! Of course I have a soldering iron. I even have soldering tin! :P

So I followed PIN2 (nREFCLOCKO_EN) and PIN15 (nRST) and I think I found where they are headed. 
I could solder jumper wires to the marked pads and keep your code and Pinout. Or do I need to remove the resistor? Wouldn't that break the connection to wherever these pins are routed currently?

[image]

CarlosDerSeher, 2024-05-14:
nIntsel seems to be connected to the LED in the RJ45 connector. So this will definitly interfere and I should remove the resistor. If you compare my schematic, I've add a mosfet to enable the LED. This way I can ensure it will be floating at power on and still keep the ethernet LED activation. You will loose this functionallity but I guess this won't be much of a deal breaker.

I cant really tell where nRst goes.

CarlosDerSeher, 2024-05-14:
Also I can see a 50MHZ clock there not sure how this is handled by LAN8720a. I have a 25MHz clock on my board

CarlosDerSeher, 2024-05-14:
chapter 3.7.4 in the datasheet discusses clock configuration. You'll need to check how your boards strap options are configured. I fear your's is configured for "REF_CLK In Mode" which could be a bigger problem

LeoSum8, 2024-05-14:
I think I have this board, or at least a clone of it: https://www.waveshare.com/lan8720-eth-board.htm
So the schematics would be these: https://www.waveshare.com/w/upload/0/08/LAN8720-ETH-Board-Schematic.pdf

[image]

And indeed it looks like nRST is going nowhere. On the PCB it also looks like the line ends shortly behind the resistor I marked.

Not sure about what you say about the clock though.

LeoSum8, 2024-05-14:
So this hardware mod would need to be combined with the code changes you described above

> your only option will be to provide MAC clock from ESP32 --> to LAN8720 and change the code of I2S to use fixed mclk and use sample stuffing for synchronization. 

right?

I'll have a look at the code to see if I can figure out what you mean :)

CarlosDerSeher, 2024-05-14:
No, in [main.c](https://github.com/CarlosDerSeher/snapclient/blob/315806c03ff89158e5b100b341ac27c1d3c2164e/main/main.c#L2759) there is this section:
```
  // clang-format off
  // nINT/REFCLKO Function Select Configuration Strap
  //  • When nINTSEL is floated or pulled to
  //    VDD2A, nINT is selected for operation on the
  //    nINT/REFCLKO pin (default).
  //  • When nINTSEL is pulled low to VSS, REF-
  //    CLKO is selected for operation on the nINT/
  //    REFCLKO pin.
  //
  // LAN8720 doesn't stop REFCLK while in reset, so we leave the
  // strap floated. It is connected to IO0 on ESP32 so we get nINT
  // function with a HIGH pin value, which is also perfect during boot.
  // Before initializing LAN8720 (which resets the PHY) we pull the
  // strap low and this results in REFCLK enabled which is needed
  // for MAC unit.
  //
  // clang-format on
  gpio_config_t cfg = {.pin_bit_mask = BIT64(GPIO_NUM_5),
                       .mode = GPIO_MODE_DEF_INPUT,
                       .pull_up_en = GPIO_PULLUP_DISABLE,
                       .pull_down_en = GPIO_PULLDOWN_ENABLE,
                       .intr_type = GPIO_INTR_DISABLE};
  gpio_config(&cfg);
```

Only thing you'll have to change here is to make gpio5 an output andf pull it high. I think won't even need the reset pin if you do it this way. but don't blame me if you brick your board by doing those changes :)

LeoSum8, 2024-05-14:
Oh wow. I'll need to wrap my head around this solution before I execute it.

Of course I'll blame you for any damages. I already blame you for buying the wrong board ;)

I'll report how it goes! Thank you!!

CarlosDerSeher, 2024-05-14:
> Of course I'll blame you for any damages. I already blame you for buying the wrong board ;)

I am not sure if there even is any board out there which does it like I did :) Didn't find one when checking...

There is a possibility that you'll still need the reset pin, maybe LAN8720a needs a clock during POR, I don't know.

CarlosDerSeher, 2024-05-14:
I can't see any possibilty to grab the clock on your board. Is it routed to the pin header? Also sometimes those oscillators have an enable pin. If you could find out which type it is and it has one you could directly control this pin without much board modification. You could also change the oscillator to something like [this ](https://www.mouser.at/datasheet/2/3/ASETDV-3224134.pdf) (if it fits). First I would try to connect the NC pin of your oscillator to ground and see if it stops operation.

CarlosDerSeher, 2024-05-14:
If this works you could connect it like this
[image]

LeoSum8, 2024-05-14:
Ok I googled a bit about those oscillators. So this could be a tri state pin, you think?
Is that a resistor in your drawing? 100k?

CarlosDerSeher, 2024-05-14:
Yes resistor. doesn't really matter which size. everything from 10k to 470k will probably be ok.

whc2001, 2024-05-16:
That's what I have been talking before. The `EN` pin of the active oscillator will make it work when on `HIGH` or floating, only stops outputing clock when `EN` is `LOW`. If using it we might need to use another GPIO to control `EN` since `GPIO5` is internally pulled HIGH when botting then got switched to LOW when the program is running.

CarlosDerSeher, 2024-05-16:
That won't be much of a problem. Should probably be a config option then:)

CarlosDerSeher, 2024-06-18:
@LeoSum8 How is it going? Is this still relevant?

LeoSum8, 2024-06-18:
I am still struggling to find the time for this. But I noticed @whc2001 progress with, at least so it seems, a similar board. 
If it takes stuff off your chest, you can close this. I'll reopen or start a new one if I do find time and run into problems.
Thank you sonfar though!!

CarlosDerSeher, 2024-06-19:
Alright. Take your time. I've just been curious

CarlosDerSeher, 2026-11-15:
i think this can be closed. feel free to reopen if needed

closed 2026-11-15 by CarlosDerSeher