CarlosDerSeher/snapclient #82: Trouble building Ethernet component under IDF v5

whc2001, 2024-05-21:
Testing with `sync_with_sample_stuffing` branch under `ESP-IDF v5.2.1`:
```
[1178/1255] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
ccache E:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.2.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -ID:/Git/snapclient/build/config -ID:/Git/snapclient/main -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/newlib/platform_include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/config/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/config/include/freertos -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/config/xtensa/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/FreeRTOS-Kernel/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/freertos/esp_additions/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_hw_support/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_hw_support/include/soc -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_hw_support/include/soc/esp32 -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_hw_support/port/esp32/. -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/heap/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/log/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/soc/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/soc/esp32 -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/soc/esp32/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/hal/platform_port/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/hal/esp32/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/hal/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_rom/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_rom/include/esp32 -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_rom/esp32 -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_common/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_system/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_system/port/soc -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_system/port/include/private -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/xtensa/esp32/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/xtensa/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/xtensa/deprecated_include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/lwip/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/lwip/include/apps -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/lwip/include/apps/sntp -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/lwip/lwip/src/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/lwip/port/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/lwip/port/freertos/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/lwip/port/esp32xx/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/lwip/port/esp32xx/include/arch -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/lwip/port/esp32xx/include/sys -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_timer/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_wifi/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_wifi/wifi_apps/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_event/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_phy/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_phy/esp32/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_netif/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/nvs_flash/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/spi_flash/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_partition/include -ID:/Git/snapclient/components/wifi_interface/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/wifi_provisioning/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/protocomm/include/common -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/protocomm/include/security -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/protocomm/include/transports -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/protocomm/include/crypto/srp6a -ID:/Git/snapclient/components/audio_board/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/deprecated -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/analog_comparator/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/dac/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/gpio/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/gptimer/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/i2c/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/i2s/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/ledc/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/mcpwm/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/parlio/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/pcnt/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/rmt/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/sdio_slave/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/sdmmc/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/sigma_delta/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/spi/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/temperature_sensor/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/touch_sensor/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/twai/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/uart/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/usb_serial_jtag/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/driver/touch_sensor/esp32/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_pm/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_ringbuf/include -ID:/Git/snapclient/components/dsp_processor/include -ID:/Git/snapclient/components/audio_hal/include -ID:/Git/snapclient/components/audio_hal/driver/es8388 -ID:/Git/snapclient/components/audio_hal/driver/es8374 -ID:/Git/snapclient/components/audio_hal/driver/es8311 -ID:/Git/snapclient/components/audio_hal/driver/es8156 -ID:/Git/snapclient/components/audio_hal/driver/es7243 -ID:/Git/snapclient/components/audio_hal/driver/es7148 -ID:/Git/snapclient/components/audio_hal/driver/es7210 -ID:/Git/snapclient/components/audio_hal/driver/es7243e -ID:/Git/snapclient/components/audio_hal/driver/tas5805m -ID:/Git/snapclient/components/audio_hal/driver/include -ID:/Git/snapclient/components/audio_hal/driver/zl38063 -ID:/Git/snapclient/components/audio_hal/driver/zl38063/api_lib -ID:/Git/snapclient/components/audio_hal/driver/zl38063/example_apps -ID:/Git/snapclient/components/audio_hal/driver/zl38063/firmware -ID:/Git/snapclient/components/audio_sal/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/efuse/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/efuse/esp32/include -ID:/Git/snapclient/components/net_functions/include -ID:/Git/snapclient/managed_components/espressif__mdns/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/console -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/vfs/include -ID:/Git/snapclient/components/opus -ID:/Git/snapclient/components/opus/opus/include -ID:/Git/snapclient/components/opus/opus/silk -ID:/Git/snapclient/components/opus/opus/silk/fixed -ID:/Git/snapclient/components/opus/opus/silk/float -ID:/Git/snapclient/components/opus/opus/celt -ID:/Git/snapclient/components/flac -ID:/Git/snapclient/components/flac/flac/include -ID:/Git/snapclient/components/ota_server/include -IE:/Espressif/frameworks/esp-idf-v5.2.1/components/app_update/inclu [... cut ...]
Compilation failed because eth_interface.c (in "eth_interface" component) includes driver/gpio.h, provided by driver component(s).
However, driver component(s) is not in the requirements list of "eth_interface".
To fix this, add driver to PRIV_REQUIRES list of idf_component_register call in D:\Git\snapclient\components\eth_interface\CMakeLists.txt.
```

Tried to add `REQUIRES driver` or `PRIV_REQUIRES driver` to the `idf_component_register` statement in `components/eth_interface/CMakeLists.txt`, still the same error.

CarlosDerSeher, 2024-05-21:
Supported IDF version is ESP-IDF v5.1.1

whc2001, 2024-05-21:
> Supported IDF version is ESP-IDF v5.1.1

Unfortunately I tried with v5.1 and the result is the same. I have checked the migrating guide and it does not mention anything about peripheral from v5.1 to v5.2, only [v4 to v5](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.0/build-system.html#specify-component-requirements-explicitly).

CarlosDerSeher, 2024-05-21:
Ok so I must have broken something doing the upgrade to IDF v5 . I'll check as soon as I can. What you can try in the meantime is to find the original "Ethernet Basic Example" from IDF v5.1.1 and see if you can replace it in eth_interface. I took this example and adapted it to my needs but it is based on IDF v4.3 and that's probably the issue

CarlosDerSeher, 2024-05-21:
I did a quick check and it seems they changed a few things in this example. So there's gonna be some more work to do on that end I guess

whc2001, 2024-05-21:
I [attempted to port](https://github.com/whc2001/snapclient/commit/f77ef026133e84697933e8329a95f2554c3e3dea) the newer example to this project, but interestingly it failed with the exact same problem. I am starting to think I got a broken environment, as I followed the suggestion exactly. I'll try again later with a fresh install of v5.1 in a Linux environment.

[image]

CarlosDerSeher, 2024-05-21:
@whc2001 ok I am a bit lost here ...
```
#include "driver/gpio.h"
#include "esp_eth.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_netif.h"
```

The only inclusion which is found seems to be `#include "esp_log.h"` but I don't know why it isn't.

I've tried adding `esp_eth` to `wifi_interface` component and also add `#include "esp_eth.h"` and it is found without any issue.
The only difference in the Cmakelist file is the if else part which I also temporary removed without any luck.

Not sure if this is related but I found `#define CONFIG_ETH_USE_SPI_ETHERNET 1` in sdkconfig.h but I selected the internal MAC. Maybe there is some problematic kconfig code?

CarlosDerSeher, 2024-05-21:
Ok so it seems you'll need to replace Cmakefilelist.txt content of eth_interface component by this 

```
idf_component_register(SRCS "eth_interface.c"
                       INCLUDE_DIRS "include"
                       REQUIRES driver esp_eth esp_netif)
```

To make does inclusion errors go away. You'll still need to port the code afterwards.

whc2001, 2024-05-21:
> Ok so it seems you'll need to replace Cmakefilelist.txt content of eth_interface component by this
> 
> ```
> idf_component_register(SRCS "eth_interface.c"
>                        INCLUDE_DIRS "include"
>                        REQUIRES driver esp_eth esp_netif)
> ```
> 
> To make those inclusion errors go away. You'll still need to port the code afterwards. Not sure why the if else isn't working as expected anymore...

I got a new Linux environment and tried this, still complaining about unable to find `driver/gpio.h`. Not sure about what to do at this point...

whc2001, 2024-05-21:
I think I found the problem, the `if` guard in the `CMakeList.txt` of `eth_interface` is not working somehow. I deleted everything only keeping the `idf_component_register` line, and after fixing some careless error in `.c` and `.h` when porting it compiled without problem (I didn't plug in the module so it keeps crashing):
[image]

whc2001, 2024-05-22:
Can you please check this, seems like it's compiling for me. My ESP32 board does not have GPIO21 broken out so I need to get another one to test on hardware...

https://github.com/whc2001/snapclient/tree/idf_v5_eth

CarlosDerSeher, 2024-05-22:
> After some fiddling I found that the if guard in the CMakeList.txt of eth_interface is not working somehow. I deleted everything only keeping the idf_component_register line

That's what I meant with my comment above.

I'll try as soon as I can

whc2001, 2024-05-23:
Seems no good. Added some debug print:

```
[0;31mE (904) snapclient_eth_init: >>>>> init Ethernet - calloc[0m
[0;31mE (914) snapclient_eth_init: >>>>> init internal Ethernet - init_internal[0m
[0;31mE (922) snapclient_eth_init: >>>>> init internal Ethernet - new_esp32[0m
[0;31mE (929) snapclient_eth_init: >>>>> init internal Ethernet - new_esp32 done, x = 1073487700[0m
[0;31mE (938) snapclient_eth_init: >>>>> init internal Ethernet - new_lan87xx[0m
[0;31mE (945) snapclient_eth_init: >>>>> init internal Ethernet - new_lan87xx done, x = 1073495584[0m
[0;31mE (954) snapclient_eth_init: >>>>> init internal Ethernet - driver_install[0m
[0;31mE (1072) eth_phy_802_3: esp_eth_phy_802_3_pwrctl(250): power up timeout[0m
[0;31mE (1072) eth_phy_802_3: esp_eth_phy_802_3_basic_phy_init(433): power control failed[0m
[0;31mE (1076) lan87xx: lan87xx_init(341): failed to init PHY[0m
[0;31mE (1082) esp_eth: esp_eth_driver_install(229): init phy failed[0m
[0;31mE (1089) snapclient_eth_init: eth_init_internal(109): Ethernet driver install failed[0m
[0;31mE (1097) snapclient_eth_init: >>>>> init internal Ethernet done, x = 0[0m
[0;31mE (1104) snapclient_eth_init: original_eth_init(281): internal Ethernet init failed[0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400daaa6
file: "./components/eth_interface/eth_interface.c" line 385
func: eth_init
expression: original_eth_init(&eth_handles, &eth_port_cnt)

abort() was called at PC 0x40089883 on core 0

Backtrace: 0x40081c89:0x3ffb84c0 0x4008988d:0x3ffb84e0 0x4008d0a1:0x3ffb8500 0x40089883:0x3ffb8570 0x400daaa6:0x3ffb85a0 0x400da6db:0x3ffb8620 0x4014b012:0x3ffb8680
```

Not sure if it's the driver or the knockoff Waveshare LAN8720A module. Gonna inspect further later...

DerPicknicker, 2024-05-23:
I can't tell you @whc2001 how much I love the docker image. This IDF stuff is really annoying to install. So maybe use the docker image instead for getting your dev machine ready. I know that it's code related see this as a friendly suggestion;-)

CarlosDerSeher, 2024-05-23:
@whc2001 OK i found the problem. First you forgot to enable this in [main.c](https://github.com/whc2001/snapclient/blob/119bfc7ed25faecdf8bae57bd5c8c81f17c72e83/main/main.c#L2760)

```
#if CONFIG_SNAPCLIENT_USE_INTERNAL_ETHERNET || \
	    CONFIG_SNAPCLIENT_USE_SPI_ETHERNET
```

Altough I am not sure if you'll need this as I don't know how you provide the clock to your LAN8720. This is specific to the my board design.

Another thing you need to add right after [this line ](https://github.com/whc2001/snapclient/blob/119bfc7ed25faecdf8bae57bd5c8c81f17c72e83/components/eth_interface/eth_interface.c#L83) is

  ```
  esp32_emac_config.clock_config.rmii.clock_mode = EMAC_CLK_EXT_IN;
  esp32_emac_config.clock_config.rmii.clock_gpio = 0;
```

With these changes my Ethernet chip initializes and starts up correctly. I am looking forward to your pull request :)

whc2001, 2024-05-24:
> I can't tell you @whc2001 how much I love the docker image. This IDF stuff is really annoying to install. So maybe use the docker image instead for getting your dev machine ready. I know that it's code related see this as a friendly suggestion;-)

Thanks, I already got the environment ready for now.

> @whc2001 OK i found the problem. First you forgot to enable this in [main.c](https://github.com/whc2001/snapclient/blob/119bfc7ed25faecdf8bae57bd5c8c81f17c72e83/main/main.c#L2760)
>
> ```
> #if [...]

I am using the [WaveShare module](https://www.waveshare.com/lan8720-eth-board.htm) where it contains the onboard active oscillator. I decided to ignore the GPIO5 clock inhibit signal for now (as we talked about making it a separate config in #79 before) and just reset multiple times until it boots.

I tried to set the `clock_mode` and `clock_gpio` but the error is the same. I tried to solder a wire to the resistor of nRST and connect it to GPIO17, still no go. There is a small detail that the green LED of the ethernet port lights up even if I don't plug in the cable.

whc2001, 2024-05-24:
Nvm, silly error ;)

> [The LAN8720 on the Waveshare board is bootstraped to use I²C address 1.](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/)

[image]

I'll do some test then generate a pull request for just migrating ethernet to IDF v5 for now. After that I'll look into splitting the clock inhibit signal into separate menuconfig.

whc2001, 2024-05-24:
Update: seems like the connection can be made, but once the snapserver starts send anything to the client, it would jam the connection (?) that `Couldn't get PCM chunk` is output indefinitely and it can no longer be pinged (always getting timeout).

[image]

CarlosDerSeher, 2024-05-24:
I've add a pull request at your repo, see whc2001/snapclient#1 

Also the code seems to work perfectly for me even without the modifications above. I'd still prefer to wait for the connection to be established before going on. What do you think?

whc2001, 2024-05-24:
> @whc2001 I've add a pull request at your repo, see [whc2001#1](https://github.com/whc2001/snapclient/pull/1)
> 
> Also the code seems to work perfectly for me even without the modifications above. I'd still prefer to wait for the connection to be established before going on. What do you think?

I totally agree about this logic. After doing this the `connection error -4` log spam disappeared.

However it does not seem to fix the issue. It still hangs just after connection for some reason. The green LED on the RJ45 does not blink often, and I cannot ping it. I am not sure what is happening and I might need to test this module with some other program to ensure that the module itself is working. I'll report back ASAP. Sorry for that.

whc2001, 2024-05-26:
Good news, the LAN8720A module works perfectly; Bad news, now I'm not even sure where is the problem ;)
I don't know where it stucks, I'll try some debug print but I doubt it would help locating.

CarlosDerSeher, 2024-05-26:
Which board are you using?

CarlosDerSeher, 2024-05-26:
Did you enable sample insertion in menuconfig? If you did try disabling it.

whc2001, 2024-05-26:
> Which board are you using?

I am using a generic ESP32-WROVER dev module (I didn't enable PSRAM for now because seems like it creates more problems) and a WaveShare LAN8720A clone. I have tested with Arduino and ESPHome that this configuration could work (make connection, obtain info from DHCP, ping from another host, connect to icanhazip.com and print out public IP).

[image]

> Did you enable sample insertion (still experimental) in menuconfig? If you did, try disabling it.

I did, I just tried to disable it and there is no change. After ethernet just connected it would sometimes ping once or twice, but as soon as SnapClient starts running it's lost.

[image]

CarlosDerSeher, 2024-05-26:
Did you try using wifi to test if it works that way?

whc2001, 2024-05-29:
Firstly really sorry for taking your time, it turns out I was a total idiot. The following two images speak themselves.

[image]
[image]

I was gonna to wire up the MAX98357 just arrived, only when I realized I never change the default I2S pins which are occupied already by the ethernet MAC. After set them to spare pins (there really aren't any since the LAN8720 already took a lot of pins) and recompiling again...

https://github.com/CarlosDerSeher/snapclient/assets/16266909/7c1c9759-becd-49f1-9361-896c63bb9b8d

Still two things to do, the first is the configurable clock inhibit GPIO which should be easy. Another is that if you pull the ethernet cable when it's working, the last audio frame would repeat indefinitely (if you ever experienced a bluescreen when there is audio playing you know how it sounds like) until it reconnected or reset is performed. It should somehow listen to the `ETHERNET_EVENT_DISCONNECTED` event and shut up the DAC or even better kill the SnapClient until the connection recovers. 

[image]

Unfortunately I will go on another trip which includes my graduation ceremony in June, so it might still take me some time...

CarlosDerSeher, 2024-05-29:
Ok good to know it mostly works now.

Maybe we could do a sanity check regarding double use of gpios during cmake.

> It should somehow listen to the ETHERNET_EVENT_DISCONNECTED event and shut up the DAC or even better kill the SnapClient until the connection recovers

Good thinking. I've never unplugged the cables since I installed my clients :)

CarlosDerSeher, 2024-05-29:
> which includes my graduation ceremony

Oh an congratulations 🎊

whc2001, 2024-06-09:
[Clock enable GPIO is added to the menuconfig](https://github.com/whc2001/snapclient/commit/08b96b3ccefafee9ae77600af137b10499eb3551). Also I've added the [functionality to opt for static IP instead of DHCP](https://github.com/whc2001/snapclient/commit/c39429c3d16878ec16815eb3b9e49cf431f1c58e) (although it would make more sense to configure a static lease in the DHCP server, but I needed a quick and dirty static IP when testing with direct ethernet connection to my computer w/o DHCP server, so why not). 

Tested with hardware and seems like it's working. However `pre-commit` glitched on the first commit and didn't perform formatting. I only realized that after making the second commit and hope that's fine.

CarlosDerSeher, 2024-06-09:
> As for the network connection/disconnection detection, I am thinking about creating a new component (something like net_events) to manage events from both WiFi and wired.

I'd love to see this implemented. The improv component could benefit from this too.
I always wanted to auto switch between ethernet and wifi depending which is available but never had time to do it.

whc2001, 2024-08-04:
Sorry for the long delay, but I'm really sorry that this issue might have to go stale. 

Firstly, I was increasingly busier recently and in the near future, and I may not have a lot of time for this. I later tested and was no longer able to reproduce the issue (after disconnecting the ethernet cable, the inner loop breaks eventually and it would go back to reconnecting). 

Secondly, frankly speaking I was looking at this issue because I assumed myself to have the need for wired ethernet. However later I found that WiFi connection is my only option due to interior layout, so I no longer have the motivation on it. 

And finally, as the last straw that broke the camel's back, I was no longer able to find my LAN8720A module anywhere, which I think I lost it by mistake in the trip. Considering I wouldn't need it in near future I decided I'm too lazy to buy a new one.

Again I am really sorry that I cannot finish this issue. Although I think it still works somehow in the current state.

CarlosDerSeher, 2024-08-04:
No worries. I use wired ethernet and the only problem I found is when the server is restartet I have to power cycle all clients for some reason. Not sure if this happens with wifi too...

whc2001, 2024-08-04:
> No worries. I use wired ethernet and the only problem I found is when the server is restartet I have to power cycle all clients for some reason. Not sure if this happens with wifi too...

I can confirm that on WiFi this seems not a problem. After the connection is dropped it will immediately attempt to reconnect to the server indefinitely. In my new issue I tested different server configuration for many times, after `systemctl restart snapserver` the client would stop for several seconds and start playing again (or die due to the semaphore issue).

closed 2025-01-16 by CarlosDerSeher