CarlosDerSeher/snapclient #83: Add possibility to switch to Bluetooth?

natumbri, 2024-05-24:
Thanks for this port of snapclient - it's working great for me. I use it in conjunction with mopidy and the https://github.com/natumbri/mopidy-youtube project I look after.

One thing I've been thinking about - sometimes I like to switch from listening via my snapcast setup to listening to something directly from my phone, via Bluetooth. 

Would it be difficult to add the capacity to switch between snapcast and BT?

Something like this https://github.com/vedranMv/Ka-Radio32 but with snapclient instead of 'internet radio'.

I'd be happy to help, but am pretty clueless when it comes to writing real code!

Anyway, thanks again for the snapclient!

Cheers
Nik

CarlosDerSeher, 2024-05-25:
Totally doable. We would need to add [this](https://github.com/espressif/esp-idf/tree/v5.2.1/examples%2Fbluetooth%2Fbluedroid%2Fclassic_bt%2Fa2dp_sink%2Fmain) as a new component and reuse pcm chunk memory allocation functions from snapclient. Only start advertising at first and if a connection is made all snapclient related task must be stopped l, all memory freed and i2s handed over to bluetooth functions.

I've always wanted to add bluetooth and stream from esp snapclient to snapserver and back again. Not sure if it is possible though.

I don't have much time to work on this project these days and my first priority is to resolve #54.

DerPicknicker, 2024-06-18:
@CarlosDerSeher .. I think trying to run 2 instances of snapcast on one esp32 host would be also a Cool experiment ;-)

CarlosDerSeher, 2024-06-18:
Why not just mount a second DAC of the same type?

CarlosDerSeher, 2024-06-18:
2 instances would double the RAM and increase TCP bandwidth/RAM requirements

DerPicknicker, 2024-06-18:
@CarlosDerSeher .. 2x the price and for example lyrat has PSRAM and also the boards from anabolyc.. this would be just a Cool experiment.

CarlosDerSeher, 2024-06-18:
But where will you pipe the 2nd instance's output?

This is a little off topic here though, maybe you should open a discussion for this.