CarlosDerSeher/snapclient #86: New version less stable - ESP32 WROOM-32 - PCM5102A

natumbri, 2024-06-13:
Hiya,

Thanks for your work on this. 

I just built the latest version, using the IDF 5.1.1 docker image. I'm using some kind of generic ESP32 WROOM-32 and a generic PCM5102A dac. 

It was working great previously, but the new version seems much less stable, with lots of dropouts. 

It also complained about a missing .h for ethernet (which I then turned off in menuconfig). 

And one of the default pins for the master I2S has changed - perhaps that's because of the change in IDF version?

If there's anything I can do to help you debug, please let me know. 

Cheers
Nik

CarlosDerSeher, 2024-06-13:
Sorry to read that. 
what do you mean by previously (which commit were you using)? using IDF v4? I did a big merge recently upgrading everything to IDF v5 so I kind of was expecting people complaining.

CarlosDerSeher, 2024-06-13:
are you using apll or sample stuffing approach for syncing?

Also yo ucan try to increase this https://github.com/CarlosDerSeher/snapclient/blob/a619587c03ced4284383fef2e6ba2343bb119bb2/components/lightsnapcast/player.c#L1505

CarlosDerSeher, 2024-06-13:
OK i fixed that build error.

Which pin are you talking about?

But I really can't reproduce the hiccups

natumbri, 2024-06-13:
> Sorry to read that.
> what do you mean by previously (which commit were you using)? using IDF v4? I did a big merge recently upgrading everything to IDF v5 so I kind of was expecting people complaining.

Haha - no worries. The same thing happens to me when I make changes to the modipy-youtube thing I look after!

The commit I went back to is fd701a1ead2fdfdc10ec0fc1bd7a92bc93f3c10d which is back to IDF 4.

natumbri, 2024-06-13:
> Which pin are you talking about?

I use 'Custom Audio Board' for my generic PCM5102A DAC. 

Master I2S LRCK is D13 in the earlier version, but is D23 in the current version. 

Cheers
Nik

CarlosDerSeher, 2024-06-13:
Please try to uncheck sample stuffing in menuconfig. Also try increasing hardresync threshold. Also a log would be great. And maybe your server config

CarlosDerSeher, 2024-06-13:
> Master I2S LRCK is D13 in the earlier version, but is D23 in the current version.

This could be related to ethernet component but I can't really tell. As long as it isn't mclk you can use almost any pin you want

natumbri, 2024-06-13:
> > Master I2S LRCK is D13 in the earlier version, but is D23 in the current version.
> 
> This could be related to ethernet component but I can't really tell. As long as it isn't mclk you can use almost any pin you want 

Yeah, I just changed the pin in the config from 23 to 13, and it was fine - but something to watch out for if people upgrade and then complain that there is no sound.

natumbri, 2024-06-13:
> Please try to uncheck sample stuffing in menuconfig. Also try increasing hardresync threshold. Also a log would be great. And maybe your server config

I'll try to do some more systematic testing during the day (in Australia!) and post up the results, including with logs.

The only config option I set for the server is the source:
````
source = tcp://127.0.0.1?name=mopidy_tcp&port=4953&controlscript=meta_mopidy.py
`
Everything else is the default (so, FLAC encoder).

The data fed into the server is 48k/16 bit raw audio. The GStreamer config from mopidy.conf is:
`
## snapcast new tcp sink
output =  audioconvert ! audioresample quality=10 ! audio/x-raw,rate=48000,channels=2,format=S16LE ! tcpclientsink host=127.0.0.1 port=4953
`

CarlosDerSeher, 2024-06-14:
> Yeah, I just changed the pin in the config from 23 to 13, and it was fine 

Maybe it was changed because of ethernet. I'll check and maybe choose only the new pin if ethernet is enabled

CarlosDerSeher, 2024-07-23:
@natumbri this can be closed can it?

natumbri, 2024-07-23:
Hi - I actually didn't go back to this. I've been using the older version a bit and it's been ok. I will try to build the new one, but, yes, you can close this. 

Thanks for your work on it.

Cheers 
Nik

closed 2024-07-23 by CarlosDerSeher