CarlosDerSeher/snapclient #90: Build failed - Failed to create SPIFFS image

mathfreakmichael, 2024-07-10:
Hello good sir, I'm reaching out because I'm very new/green with building ESP32 firmware and am trying to transition away from LMS and squeezelite for my ESP32 muse luxe for Music Assistant. I have tried to build using the README on both windows and docker and keep getting the same error. If you would be so kind as to help me identify if this is operator error or not that would be greatly appreciated. I leave the http server port at the default 8000 and have tried to change it as well.

`cd /project/snapclient/build/esp-idf/ui_http_server && /opt/esp/tools/cmake/3.24.0/bin/cmake -E echo "Failed to create SPIFFS image for partition 'storage'. " "Check project configuration if using the correct partition table file." && /opt/esp/tools/cmake/3.24.0/bin/cmake -E remove /project/snapclient/build/esp-idf/ui_http_server/gkBzm.cmake && /opt/esp/tools/cmake/3.24.0/bin/cmake -E env "FAIL_MESSAGE=Failing the build (see errors on lines above)" /opt/esp/tools/cmake/3.24.0/bin/cmake -P /opt/esp/idf/tools/cmake/scripts/fail.cmake
FAILED: esp-idf/ui_http_server/CMakeFiles/spiffs_storage_bin /project/snapclient/build/esp-idf/ui_http_server/CMakeFiles/spiffs_storage_bin 
cd /project/snapclient/build/esp-idf/ui_http_server && /opt/esp/tools/cmake/3.24.0/bin/cmake -E echo "Failed to create SPIFFS image for partition 'storage'. " "Check project configuration if using the correct partition table file." && /opt/esp/tools/cmake/3.24.0/bin/cmake -E remove /project/snapclient/build/esp-idf/ui_http_server/gkBzm.cmake && /opt/esp/tools/cmake/3.24.0/bin/cmake -E env "FAIL_MESSAGE=Failing the build (see errors on lines above)" /opt/esp/tools/cmake/3.24.0/bin/cmake -P /opt/esp/idf/tools/cmake/scripts/fail.cmake
Failed to create SPIFFS image for partition 'storage'.  Check project configuration if using the correct partition table file.
CMake Error at /opt/esp/idf/tools/cmake/scripts/fail.cmake:3 (message):
  Failing the build (see errors on lines above)

[3/1231] Generating memory.ld linker script...
[4/1231] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************`

CarlosDerSeher, 2024-07-10:
I've updated the Readme. You'll need to copy one of the template sdkconfig first and rename it to sdkconfig

mathfreakmichael, 2024-07-20:
I missed the emails that you responded! Thank you for that and I will try it again today!

closed 2024-07-20 by mathfreakmichael