CarlosDerSeher/snapclient #91: Compile error: unknown type name 'xSemaphoreHandle'

Ebbe, 2024-07-19:
Hi,

I git cloned (including submodules), ran in docker(I use Linux) then copied sdkconfig_lyrat_v4.3.
Running `idf.py build` results in an error:

    In file included from /project/components/audio_board/lyrat_v4_3/board.h:30,
                 from /project/components/audio_board/lyrat_v4_3/board_pins_config.c:31:
    /project/components/audio_hal/include/audio_hal.h:159:3: error: unknown type name 'xSemaphoreHandle'
      159 |   xSemaphoreHandle audio_hal_lock;                  /*!< semaphore of codec */
          |   ^~~~~~~~~~~~~~~~

I tried renaming it to SemaphoreHandle_t, but that resulted in errors elsewhere in the code.

CarlosDerSeher, 2024-07-19:
@Ebbe please fetch latest push from master. Seems I didn't update the templates to IDF v5. Should be alright now.

Ebbe, 2024-07-20:
That did the trick. Thanks!

closed 2024-07-20 by Ebbe