CarlosDerSeher/snapclient #92: [Feature] Build ontop of ESPHome (make this an ESPHome component)

oliv3r, 2024-07-22:
Snapclient on ESP32 is great. What's even cooler, is if we can do small automation tasks. E.g. what if we connect an external button the the ESP32 to mute? Or volume keys? Or be able to turn on/off an external Amplifier?

These features could be added to this firmware for sure, but having it being controlled by Home-Assistant would be even cooler :)

I am developping a PCB for the Vappeby Gen3 Speaker (https://www.ikea.com/us/en/p/vappeby-bluetooth-speaker-black-gen-3-10517371/#content) to use it as an snapclient speaker.

However I don't want to break the existing functionality (e.g. bluetooth speaker, buttons etc) but do want it to be able to remotely powered on.

This is trivial with esphome and home-assistant, but then I can't get snapclient of course. Same vice-versa.

This post is just something to get the discussion going ...

P.S. are you aware of snapclientc? Maybe also worth to look into to combine some efforts.

DerPicknicker, 2024-07-22:
We already had this discussion. Not possible the sync algorithm is highly optimised and it would require a lot of rewriting from @CarlosDerSeher .

Why not just use your PCB use a supported dac and you're ready to go. What benefit would you see? The webinstaller already makes it easy to use.

CarlosDerSeher, 2024-07-22:
@oliv3r I saw snapclientc a while ago but can't find it anymore. Is this still active?

Regarding esphome, this could be done I guess but I really don't know that much about the codebase and how integration could be done. This esphome component would need exclusive control of I2S.

I don't see myself doing this in the near future because I really don't have time for this but could support someone who is willing to try to take that advanture.

DerPicknicker, 2024-07-22:
@CarlosDerSeher ... I asked on discord, and in some other discussions nobody was able to tell me how the ESPHOME structure is looking. For non timing relevant tasks it's fine but for your sync algorithm it's needed to know such basic things. 

I gave up, the documentation lacks and my time is more worth than reading Bad documentation..... sorry for the harsh words the project is great but why they don't offer a drawing of their system structure.

oliv3r, 2024-07-22:
https://github.com/christf/snapcastc last commit in june, so certainly active.

From an architectual side, both suffer from one major issie, and that is that the streaming library is so intertwined with everything. It would be nice to have isolated libraries...

CarlosDerSeher, 2024-07-22:
What exactly do you think needs to be isolated? Which are the problematic sections in this code in your opinion?

oliv3r, 2024-07-24:
With snapcast and snapcastC? for one the library, which @bridadan did; but then this library got stale, and you/@jorgenkraghjakobsen "forked" it if you will. That means we have technically 5 libraries in the wild, each having their own life. From a library point of view, this is undesireable, all should be using one and the same library.

Personally, I blame C++ for this, as it's much harder to integrate into <other lang> systems. Where it a simple C (at least the interface, but you still have to deal with C++ dependencies, libstdc++ at the least, which is for certain platforms unacceptable), like @bidadan did, things would be much cleaner and easier for the library part.

But more things in snapcast and snapcastC are entangled, where they probably shouldn't be.

CarlosDerSeher, 2024-07-24:
So yo mean the snapcast protocol implementation itself should be put in a library? This should be doable but the thing with my repo is, that I highly optimized overall RAM usage and also peak RAM usage so also modules without PSRAM can be used. That's why I don't really use the original library from jorgen / bridadan bbut only parts and also changed some things.

bridadan, 2024-07-24:
If there's a library that's considered the most up-to-date, I'd be happy to place a link to it in my repo's readme (or better yet, feel free to send a PR!). I have no plans to pick it up again in the future and I don't want to add to the confusion 🙂

oliv3r, 2024-07-24:
> So yo mean the snapcast protocol implementation itself should be put in a library? This should be doable but the thing with my repo is, that I highly optimized overall RAM usage and also peak RAM [...]

But making it a library still makes sense, any other system can very effectivily use the library with all RAM optimilizations. Where needed, these could be put behind flags at a later time.

DerPicknicker, 2024-07-25:
The Big Problem is the structure of ESPHOME. It's not clear how they designed their system.. Besides that it misses some options for Components, like prioritizing. I guess you can do that inside your Component but the ESPHOME Runtime has to prioritizing  your component. That's a crucial feature for an Audio-Sync application... 

I found this: https://github.com/mads03dk/esphome currently not finished but maybe a good starting point.

CarlosDerSeher, 2024-07-25:
There are 2 questions which need to be answered regarding esphome:

1. Do we need to use their audio component or are we allowed to lock i2s control for our own component and make them mutually exclusive from config view.

2. Is there a parent task calling all component tasks (or similar), or can we install any tasks with the priorities and on the core we want?

DerPicknicker, 2024-07-30:
@CarlosDerSeher I can't give you an answer... Maybe @mads03dk can say something because he started working on an ESPHOME integration...

alexyao2015, 2024-08-11:
I can help answer some questions since I've worked in the codebase a bit. 

Esphome runs essentially on the Arduino model where it has its "main" setup and loop function calls. Components register their own setup and loop calls which are processed along with other components. The backend implementation of loops are ran depends on if Arduino or idf is selected. On Arduino it uses the standard setup and loop functions. On idf, it creates a loop task that run the main esphome loop. For idf, this means esphome running on a single core always because there is only one loop task.

You are free to essentially do whatever you please inside the constraints of your components, including creating new tasks and using the normal idf calls to gpio. Of course, if you want to use esphome abstraction on top of the gpip, it's also possible but not required. I would think since esp32 is dual core, you could run the Snapcast client on the other inactive core.

CarlosDerSeher, 2024-08-11:
Thanks for your input appreciate it. This sounds like we could indeed implement sync in our own high priority task, so it should be possible to port it I guess.

DerPicknicker, 2024-08-15:
@CarlosDerSeher .. which files are for the snapcast and sample stuffing relevant?

CarlosDerSeher, 2024-08-15:
`player.c` implements i2s and syncing

DerPicknicker, 2024-08-15:
@alexyao2015 ... do you know if we can use C code inside a ESPHOME component?

CarlosDerSeher, 2024-08-15:
Normally c code called from c++ isn't a problem

alexyao2015, 2024-08-15:
C should work fine. It's just platformio under the hood and it copied it all into a platformio project. You'll likely want to choose the idf framework instead of Arduino so you can use freertos tasks.

DerPicknicker, 2024-08-15:
@alexyao2015 thank you for the further information.. if I had more time I would start working on this. Maybe after the summer..

DerPicknicker, 2024-08-16:
@CarlosDerSeher in which file you're connecting to the snapcast server and getting the data? Http task?

CarlosDerSeher, 2024-08-16:
Yes http task.

Most tasks are not pinned to any specific core. Only player task is pinned to core 1 with high priority

DerPicknicker, 2024-08-16:
@CarlosDerSeher ... 

So I think it would be much easier if we can or you can write down all needed functions. Because I tried now understanding your whole concept and until I get every detail it may be faster if you can write down which functions are essentially needed for snapcast connections. 

I think a library based on your syncing algorithm would be amazing because it would be much easier to integrate it into ESPHOME or other systems.

CarlosDerSeher, 2024-08-16:
You are talking about a library specifically for esp32 or C / freeRTOS in general?

When I have time I'll try to do some documentation for the code. Maybe some clean up before that. But there is still #54 which has to be resolved in my opinion or rhe code can't be called stable. Some parts of the code rely on knowing the chunk size/duration and that's a real problem currently and I think  this part needs a bit of more work before it can be put in a library. 

A library of snapclient should be somewhat independent of i2s functions and I think those should be passed as pointers to the lib during initialization? And also DMA buffer size and such...

DerPicknicker, 2024-08-16:
@CarlosDerSeher ..

I am talking about only a esp32 library because this is the most easiest way... 

I think we should have at a top level this functionset 
- connect to snapcast server
- Get audio data 
- Sync audio 

The i2s component is still not clear. Do we need to use the ESPHOME component or can we use the one from espressif. If we use the Espressif implementation we should also include that. (I know that ESPHOME is just a abstraction of Espressif...)

CarlosDerSeher, 2024-08-17:
Audio syncing, audio buffer and i2s are intertwined. Syncing highly relies on i2s DMA buffer size and chunk duration. It works like this (pseudo code):
```
Get audio chunk
Fill DMA buffer
Wait until its timestamp is now
Start i2s playback
While True:
  Get audio chunk
  Check timestamp and do sample stuffing
  Put it to i2s (which blocks until enough space is available)
```

DerPicknicker, 2024-08-20:
@CarlosDerSeher thank you for your clarification. Maybe we will wait until you would call it stable and then we can continue on this discussion?

luar123, 2024-11-22:
Is one of you working on this?

I have worked on esphome components a bit and I am pretty sure that you can use your own i2s component. In fact it should be fairly straight forward to implement a very simple snapcast component based on your code. This would work only with esp-idf framework, same as all other audio components.

Of course there will be limitations: You can't use esphome audio components, and if an i2c dac is used maybe also esphome i2c components can't be used. I would also expect issues with components that tend to block, like displays or some sensors. Esphome components are allowed to block for 10ms and some are even violating this.

Converting all the config options from sdkconfig to yaml will be some work but shouldn't be an issue. Adding control functionality, using esphome i2c implementation etc would be a second step.

DerPicknicker, 2024-11-22:
@luar123 .. I don't have the time to work on this currently... I use it only via webflasher.

Do you have a good page where I can dive into the ESPHOME structure a bit more?

luar123, 2024-11-23:
Unfortunately, I don't have much time either and also other priorities. But maybe I can find the time to create a simple proof-of-concept.

This is the only official documentation: https://esphome.io/guides/contributing.html#contributing-to-esphome
But in order to really understand how everything works together you need to read a lot of code :-(
There is something planned, but no activity so far: https://github.com/esphome/developers.esphome.io

DerPicknicker, 2024-11-23:
@luar123 .. 

That would be great maybe just a small draft - how to add configuration, where to put the needed sync parts etc.

Then I can continue at some point.

CarlosDerSeher, 2024-11-23:
Would be great to see this moved forward. I can't find much time these days to do coding on the project but if I do, my priority is resolving #54 and after that work through all those other minor issues and pull requests.

luar123, 2025-01-01:
Here is a first working implementation: https://github.com/luar123/snapclient/tree/esphome/esphome
I added everything under the esphome folder (but this could also be a separate repository):
* snapcast_esp32.yaml: example config
  * dac sdkconfigs are here because I didn't want to add them to the component (should use esphome dacs anyway)
  * enable_ipv6 is needed, otherwise there are some compile issues, should be fixed in C code
* partitions.csv: copy from base folder, not sure if this is needed. normal esphome table has no spiffs entry
* component/snapclient: esphome component
  * init.py: config validation and c code generation
    * configs for: name, host, port, sample insertion
    * if host is not set, mdns will be used
    * dsp configs and webserver are hardcoded
    * all components can be loaded from the snapclient github repo. No need to duplicate code.
    * For now the local repo is used, because small modifications were needed (see first commit).
  * snapclient.cpp: based on main.c
    * removed network/logger/ota setup (handled by esphome)
    * some changes needed because of c++
It seems to run fine including webserver and ota (player is not stopped) , but I didn't check how good the sync is. Also I did not look into task priorities and on which cores they should run. I also did not test other configs (dacs/ethernet/psram).
As is, you can't use any esphome i2s or i2c components in parallel.

Todos:
* Add configs for dsp and webserver (maybe make webserver optional)
* Test other configs
* allow ipv4 only mode
* Implementation would be much cleaner and without doubling code if all the tasks in main.c could be moved to a snapclient component
* Use esphome audio-dac component
  * this would allow easier config and other i2c components
  * Add a dac state object that can be modified from all the tasks
  * write changes to the dac from the main app task (loop function in esphome)
  * Currently esphome has only support for ES8311 and AIC3204 (and non i2c dacs)
* Use esphome i2s-audio/speaker/mediaplayer component
  * didn't look into that, but I guess it would require some work on the snapclient codebase and on esphome i2s components

CarlosDerSeher, 2025-01-02:
Nice to see this proceed. I'll check it out when I find time. I'll need to install esp home framework?

luar123, 2025-01-04:
Yes, you can install as a python package or use docker: https://esphome.io/guides/getting_started_command_line
Or install the dev environment: https://esphome.io/guides/contributing#setting-up-a-development-environment

Then just run esphome run snapcast_esp32.yaml

luar123, 2025-01-08:
Pushed an update using PR #110. This drops all the unneeded audio_board dependencies and uses esphomes dac (or mute pin) and i2s components. i2s_audio is a patched version to use the new i2s_std driver because the default component uses still the old driver. The patched version won't work with any other esphome components for now.
I want to clean up everything and add a few small features and then create a PR in esphome to get their feedback as well.

CarlosDerSeher, 2025-01-12:
I am  still on to the 20ms bug and there will be a lot of changes to http task and also player task. So please keep this in mind when changing code. We'll probably need to merge some stuff manually if you change it a lot now.

luar123, 2025-01-12:
I don't plan further changes to those parts for now.

For this whole integration it would be great to move http task to its own component and to allow for more configuration/control like stop/resume player task(I had this more or less working when experimenting with power reduction a year ago, but it isn't compatible anymore) or setting the output channel. But this should wait until you merged your changes.

luar123, 2025-03-01:
See https://github.com/esphome/esphome/pull/8350

CarlosDerSeher, 2025-03-01:
Nice :) 
How does this impact further development in this repo? I don't want to break stuff by changing anything important.

alexyao2015, 2025-03-01:
Imo, this should stay as a custom component instead of being merged into the main repo

luar123, 2025-03-02:
> Nice :) How does this impact further development in this repo? I don't want to break stuff by changing anything important.

You won't break anything as the esphome component is linked to a specific commit reference. When there is an update here this needs to be changed manually. Could also link to a branch or tag.

> Also, do you enforce ipv6 or is this optional? In develop branch I am in the middle of addressing this but currently you'll have to choose ipv6 or ipv4 by menuconfig

Esphome uses IP4 by default but that won't compile with current master. That is why we need to add
```
network:
  enable_ipv6: true
```
to the yaml.
Let me know once dev branch is usable, then I will try.

@alexyao2015 It doesn't make a big difference. With the PR it gets more visibility and probably more feedback from devs. It won't get merged as is but you can use it easily with
```
external_components:
  - source: github://pr#8350
    components: [ snapclient ]
```

DerPicknicker, 2025-03-02:
@luar123 .. Can we use Snapclient and other ESPHOME stuff in parallel for example push buttons?

luar123, 2025-03-02:
Yes that should work. Some sensors or displays might mess up audio/synchronization, though, because they block for several ms.

CarlosDerSeher, 2025-03-02:
Shouldn't the higher priority player task still be able to interrupt and take over?

luar123, 2025-03-02:
I guess that is correct. Even if the main loop takes a lot of computing time, player task should still be able to run.