CarlosDerSeher/snapclient #95: Stack overflow during init

aliask, 2024-07-28:
I'm running into a intermittent issue where the STM32 hits a stack overflow during initialisation.

For example:
```
I (98789) main_task: Returned from app_main()
I (98790) SC: try connecting to static configuration 10.20.0.10:1704
I (99163) SC: netconn connected
I (99165) SC: netconn sent hello message
I (99387) SC: Buffer length:  1000
I (99387) SC: Latency:        0
I (99387) SC: Mute:           0
I (99388) SC: Setting volume: 100
I (99392) dspProc: Set volume to 1.000000
I (99397) dspProc: Set volume to 1.000000
I (99438) SC: fLaC sampleformat: 44100:16:2

***ERROR*** A stack overflow in task flac_decoder_ta has been detected.

Backtrace: 0x40375c0a:0x3fcd74d0 0x4037ca51:0x3fcd74f0 0x4037f6fa:0x3fcd7510 0x4037e2e9:0x3fcd7590 0x4037f808:0x3fcd75b0 0x4037f7fe:0x3fcd76a0 0x3fcd76bd:0x3fcd76e0 |<-CORRUPTED
0x40375c0a: panic_abort at /home/aliask/esp/v5.1.3/esp-idf/components/esp_system/panic.c:472
0x4037ca51: esp_system_abort at /home/aliask/esp/v5.1.3/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x4037f6fa: vApplicationStackOverflowHook at /home/aliask/esp/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:581
0x4037e2e9: vTaskSwitchContext at /home/aliask/esp/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3841
0x4037f808: _frxt_dispatch at /home/aliask/esp/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:450
0x4037f7fe: _frxt_int_exit at /home/aliask/esp/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:245
```

In this case, it was in the `flac_decoder_task`, however that's not always the case:

```
[... same boot sequence messages to above ...]
I (337862) dspProc: Set volume to 1.000000
I (337866) dspProc: Set volume to 1.000000

***ERROR*** A stack overflow in task IDLE0 has been detected.
[... same stack trace to above ...]
```

After the crash the device restarts and _usually_ successfully allocates the memory.

My configured stack sizes seem to be the same or larger than the sample configurations, so I'm not sure what's causing this.

My device has 8MB of PSRAM configured, so I have lots of room to play with.

<details>
<summary>Stack-related configuration values</summary>

```
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2304
CONFIG_ESP_MAIN_TASK_STACK_SIZE=3584
CONFIG_ESP_MINIMAL_SHARED_STACK_SIZE=2048
CONFIG_ESP_IPC_TASK_STACK_SIZE=1280
CONFIG_ESP_TIMER_TASK_STACK_SIZE=3584
CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH=2048
CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=3072
CONFIG_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_PTHREAD_STACK_MIN=768
CONFIG_WEBSOCKET_SERVER_TASK_STACK_DEPTH=6000
CONFIG_MDNS_TASK_STACK_SIZE=4096
CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE=2304
CONFIG_MAIN_TASK_STACK_SIZE=3584
CONFIG_IPC_TASK_STACK_SIZE=1280
CONFIG_TIMER_TASK_STACK_SIZE=3584
CONFIG_TIMER_TASK_STACK_DEPTH=2048
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
```
</details>

[... 150 lines cut ...]
I (164673) HTTP: Starting server on 10.20.0.110:8000
I (164674) HTTP: Starting HTTP Server on port: '8000'
I (164679) NETF: Setup mdns
I (164694) NETF: Initializing SNTP
I (164694) NETF: Waiting for system time to be set... (1/10)
I (166695) NETF: The current date/time in UTC is: Sun Jul 28 00:38:47 2024
I (166696) dspProc: dsp_processor_init: init done
I (166698) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (166707) main_task: Returned from app_main()
I (166707) SC: try connecting to static configuration 10.20.0.10:1704
I (166778) SC: netconn connected
I (166780) SC: netconn sent hello message
I (168777) SC: Buffer length:  1000
I (168778) SC: Latency:        0
I (168778) SC: Mute:           0
I (168779) SC: Setting volume: 100
I (168783) dspProc: Set volume to 1.000000
I (168788) dspProc: Set volume to 1.000000
I (168909) SC: fLaC sampleformat: 44100:16:2
I (169595) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 4
W (169597) PLAYER: no pcm chunk queue created?
I (169597) PLAYER: created new queue with 37
I (169602) PLAYER: snapserver config changed, buffer 1000ms, chunk 1152 frames, sample rate 44100, ch 2, bits 16 mute 0 latency 0
```
</details>

CarlosDerSeher, 2024-07-28:
I have to say, I am in the middle of dropping the decoder task completely and moving the decoding part to http task but didn't have time to finish this yet. I did this for flac but not the other decoders.

I am doing this because of #54 but haven't found a working solution yet.

I could push these changes to a new branch for you to play with as I don't think it would be usefull to do much work on this in master.

aliask, 2024-07-28:
OK, seems like there is some relationship here. I changed my stream to PCM@20ms and I don't see the crash any more.

However, still no audio - buffer semaphore never given, repeats `W (34558) PLAYER: latency_buffer_full: can't take semaphore` until the WDT kills process.

Unscientifically, I just increased the size of the FLAC task's stack from `9 * 256` to `12 * 256` and this also fixes the crash. However again, no audio:

```
I (7098) SC: netconn sent hello message
I (7108) SC: Buffer length:  1000
I (7108) SC: Latency:        0
I (7108) SC: Mute:           0
I (7108) SC: Setting volume: 100
I (7118) SC: fLaC sampleformat: 44100:16:2
I (7268) PLAYER: player_setup_i2s: dma_buf_len is 576, dma_buf_count is 4
W (7268) PLAYER: no pcm chunk queue created?
I (7268) PLAYER: created new queue with 37
I (7278) PLAYER: snapserver config changed, buffer 1000ms, chunk 1152 frames, sample rate 44100, ch 2, bits 16 mute 0 latency 0
I (10568) SC: latency buffer full
W (10578) PLAYER: RESYNCING HARD 1: age 693193us, latency 1203663951566us, free 8496531, largest block 8257536, rssi: -73
E (12578) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 1203663s, 951.566ms
E (14578) PLAYER: Couldn't get PCM chunk, recv: messages waiting 0, diff2Server: 1203663s, 951.566ms
W (14798) PLAYER: RESYNCING HARD 1: age 4890606us, latency 1203663951566us, free 8499871, largest block 8257536, rssi: -70
W (14988) PLAYER: RESYNCING HARD 1: age 5042574us, latency 1203663941965us, free 8501207, largest block 8257536, rssi: -70
```

What settings do you recommend trying to get this software to work?

CarlosDerSeher, 2024-07-28:
Did you double check your gpio config for i2s?

aliask, 2024-08-03:
### Short version

I have FLAC "working" @ 26ms, with a strange bug which is probably unrelated to the stack overflow. So it seems that increasing the hard-coded stack size is enough to fix this problem.

### Long version

I have two STM32 S3 devboards, and out of frustration with WiFi stability I switched to the other board to test if it has the same behaviour. This new board connects to WiFi much more reliably, and while I was switching things over I also converted the DAC connection from floating jumper wires to using a breadboard.

After switching board I started getting choppy audio output! Progress! There is no debug output to suggest the source of this problem, apart from some uncorrelated RESYNCING HARD 1/2 messages. However, when I picked up the board the audio suddenly became clear.

I'm unsure if this impedance change fixes something to do with the I2S signals or something else entirely - but it seems that my hardware configuration is the source of some of my problems. Here's a recording of the audio while I place my finger on some unrelated GPIO pins: [stm32s3_audio.mp3.gz](https://github.com/user-attachments/files/16478119/stm32s3_audio.mp3.gz). RSSI doesn't change when I do this, so I guess it's not WiFi.

CarlosDerSeher, 2024-08-03:
Maybe a broken wire somewhere?

Which DAC?

aliask, 2024-08-03:
I think I've figured it out. I'm using a PCM5102A DAC, and when you're not supplying SCK (which I'm not) then you have to ground it. I had left it floating, so I've fixed that up and the stream is a lot more reliable.

I'm still getting significant artifacts (e.g. briefly right after the client connects) and unreliable operation, but I might continue to debug & open a separate issue if I'm not able to figure things out.

Would you like me top open a PR to increase the FLAC stack size? How was the original value calculated?

CarlosDerSeher, 2024-08-03:
> Would you like me top open a PR to increase the FLAC stack size? How was the original value calculated?

Trial and error. Yes feel free to open a PR

closed 2024-08-04 by aliask