CarlosDerSeher/snapclient #99: Build of master fails

LeoSum8, 2024-08-07:
Hi there,
I just wanted to rebuild the current master to update my snapclient.
It's a basic ESP32 mini with a PCM5102A DAC attached.

Last time I successfully built this was 8th of May and I just verified that building the state I had checked out then is still working.

The current master buidl fails with:
```
root@fe81980090c7:/project# idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory /project/build
Executing "ninja all"...
[0/1] Re-running CMake...-- ccache will be used for faster recompilation
-- git rev-parse returned 'fatal: detected dubious ownership in repository at '/project'
To add an exception for this directory, call:

	git config --global --add safe.directory /project'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Processing 3 dependencies:
[3/3] idf (5.1.1)-- Project sdkconfig file /project/sdkconfig
Compiler supported targets: xtensa-esp32-elf

-- App "snapclient" version: 1
-- Adding linker script /project/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /opt/esp/idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /opt/esp/idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Current board name is CONFIG_AUDIO_BOARD_CUSTOM
-- Selected DAC is CONFIG_DAC_PCM5102A
-- Components: app_trace app_update audio_board audio_hal audio_sal bootloader bootloader_support bt cmock console custom_board cxx driver dsp_processor efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_peripherals esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp-dsp espressif__mdns esptool_py eth_interface fatfs flac freertos hal heap http_parser idf_test ieee802154 improv_wifi json libbuffer libmedian lightsnapcast log lwip main mbedtls mqtt net_functions newlib nvs_flash openthread opus ota_server partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ui_http_server ulp unity usb vfs wear_levelling websocket wifi_interface wifi_provisioning wpa_supplicant xtensa
-- Component paths: /opt/esp/idf/components/app_trace /opt/esp/idf/components/app_update /project/components/audio_board /project/components/audio_hal /project/components/audio_sal /opt/esp/idf/components/bootloader /opt/esp/idf/components/bootloader_support /opt/esp/idf/components/bt /opt/esp/idf/components/cmock /opt/esp/idf/components/console /project/components/custom_board /opt/esp/idf/components/cxx /opt/esp/idf/components/driver /project/components/dsp_processor /opt/esp/idf/components/efuse /opt/esp/idf/components/esp-tls /opt/esp/idf/components/esp_adc /opt/esp/idf/components/esp_app_format /opt/esp/idf/components/esp_coex /opt/esp/idf/components/esp_common /opt/esp/idf/components/esp_eth /opt/esp/idf/components/esp_event /opt/esp/idf/components/esp_gdbstub /opt/esp/idf/components/esp_hid /opt/esp/idf/components/esp_http_client /opt/esp/idf/components/esp_http_server /opt/esp/idf/components/esp_https_ota /opt/esp/idf/components/esp_https_server /opt/esp/idf/components/esp_hw_support /opt/esp/idf/components/esp_lcd /opt/esp/idf/components/esp_local_ctrl /opt/esp/idf/components/esp_mm /opt/esp/idf/components/esp_netif /opt/esp/idf/components/esp_netif_stack /opt/esp/idf/components/esp_partition /project/components/esp_peripherals /opt/esp/idf/components/esp_phy /opt/esp/idf/components/esp_pm /opt/esp/idf/components/esp_psram /opt/esp/idf/components/esp_ringbuf /opt/esp/idf/components/esp_rom /opt/esp/idf/components/esp_system /opt/esp/idf/components/esp_timer /opt/esp/idf/components/esp_wifi /opt/esp/idf/components/espcoredump /project/managed_components/espressif__esp-dsp /project/managed_components/espressif__mdns /opt/esp/idf/components/esptool_py /project/components/eth_interface /opt/esp/idf/components/fatfs /project/components/flac /opt/esp/idf/components/freertos /opt/esp/idf/components/hal /opt/esp/idf/components/heap /opt/esp/idf/components/http_parser /opt/esp/idf/components/idf_test /opt/esp/idf/components/ieee802154 /project/components/improv_wifi /opt/esp/idf/components/json /project/components/libbuffer /project/components/libmedian /project/components/lightsnapcast /opt/esp/idf/components/log /opt/esp/idf/components/lwip /project/main /opt/esp/idf/components/mbedtls /opt/esp/idf/components/mqtt /project/components/net_functions /opt/esp/idf/components/newlib /opt/esp/idf/components/nvs_flash /opt/esp/idf/components/openthread /project/components/opus /project/components/ota_server /opt/esp/idf/components/partition_table /opt/esp/idf/components/perfmon /opt/esp/idf/components/protobuf-c /opt/esp/idf/components/protocomm /opt/esp/idf/components/pthread /opt/esp/idf/components/sdmmc /opt/esp/idf/components/soc /opt/esp/idf/components/spi_flash /opt/esp/idf/components/spiffs /opt/esp/idf/components/tcp_transport /project/components/ui_http_server /opt/esp/idf/components/ulp /opt/esp/idf/components/unity /opt/esp/idf/components/usb /opt/esp/idf/components/vfs /opt/esp/idf/components/wear_levelling /project/components/websocket /project/components/wifi_interface /opt/esp/idf/components/wifi_provisioning /opt/esp/idf/components/wpa_supplicant /opt/esp/idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /project/build
[2/1232] cd /project/build/esp-idf/ui_http_server && /opt/esp/tools/cmake/3.24.0/bin/cmake -E echo "Faile...ors on lines above)" /opt/esp/tools/cmake/3.24.0/bin/cmake -P /opt/esp/idf/tools/cmake/scripts/fail.cmakeFAILED: esp-idf/ui_http_server/CMakeFiles/spiffs_storage_bin /project/build/esp-idf/ui_http_server/CMakeFiles/spiffs_storage_bin 
cd /project/build/esp-idf/ui_http_server && /opt/esp/tools/cmake/3.24.0/bin/cmake -E echo "Failed to create SPIFFS image for partition 'storage'. " "Check project configuration if using the correct partition table file." && /opt/esp/tools/cmake/3.24.0/bin/cmake -E remove /project/build/esp-idf/ui_http_server/vmEc5.cmake && /opt/esp/tools/cmake/3.24.0/bin/cmake -E env "FAIL_MESSAGE=Failing the build (see errors on lines above)" /opt/esp/tools/cmake/3.24.0/bin/cmake -P /opt/esp/idf/tools/cmake/scripts/fail.cmake
Failed to create SPIFFS image for partition 'storage'.  Check project configuration if using the correct partition table file.
CMake Error at /opt/esp/idf/tools/cmake/scripts/fail.cmake:3 (message):
  Failing the build (see errors on lines above)

[4/1232] Generating ../../partition_table/partition-table.binPartition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /project/build/log/idf_py_stderr_output_461 and /project/build/log/idf_py_stdout_output_461

```

I did a clean, recursive checkout and try to build via docker.

I only changed a few parameters via menuconfig and kept the rest as default:
- board: custom board
- dac
- I2C/I2S/Mute Pins
- Snapclient mDNS
- wifi-settings (not in the sdkconfig below)

This is my resulting sdkconfig that fails to build:
<details>
[... 1759 lines cut ...]
# CONFIG_MDNS_SKIP_SUPPRESSING_OWN_QUERIES is not set
# CONFIG_MDNS_ENABLE_DEBUG_PRINTS is not set
CONFIG_MDNS_ENABLE_CONSOLE_CLI=y
# CONFIG_MDNS_RESPOND_REVERSE_QUERIES is not set
CONFIG_MDNS_MULTIPLE_INSTANCE=y

#
# MDNS Predefined interfaces
#
CONFIG_MDNS_PREDEF_NETIF_STA=y
CONFIG_MDNS_PREDEF_NETIF_AP=y
CONFIG_MDNS_PREDEF_NETIF_ETH=y
# end of MDNS Predefined interfaces
# end of mDNS
# end of Component config

# CONFIG_IDF_EXPERIMENTAL_FEATURES is not set

# Deprecated options for backward compatibility
...
omitted to fit into githubs max comment lenght
...
# End of deprecated options

</details>

LeoSum8, 2024-08-07:
Ok, I also had to change the config to use the partitions.csv and increase the flash size. Now it builds.

Did the defaults change compared to may? I don't recall having fiddled around with these settings back then.

LeoSum8, 2024-08-07:
To boot successfully I also need to disable ethernet, which seems to be enabled by default.
But it is still not running stable. Will need to investigate further.

CarlosDerSeher, 2024-08-07:
Please use one of the template sdkconfigs and rename it as suggested in the build instructions. I've removed the sdkconfig from git tree as I always had to remove wifi credentials by hand before checking it in. This was very annoying to me

LeoSum8, 2024-08-08:
Thank you for the hint. I overread that point. Taking the lyrat as a starting point, it works fine now.
I think the main issue was that before I hadn't activated sample_insertion (btw. is this the same as sample stuffing???) but I didn't compare all values.

Shall I add my sdkconfig in a pull request or do you not want examples for custom boards? I think the PCM5102A is a cheap DAC and a good starting point for people to try out your project.