CarlosDerSeher/snapclient #101: Volume quiet on LyraT 4.3

jonzobot, 2024-08-13:
Hi Carlos, 

As requested under issue #68, that commit appears to have created a small problem relating to volume on LyraT 4.3. Specifically the volume levels have been reduced substantially. It may be that the fix and config changes for that issue may need to be limited to the board in question, i.e. AI Thinker. It is probably not an issue if one is using an external amplifier, but I am using the on-board amp.

hjri, 2024-08-22:
I had to uncomment https://github.com/CarlosDerSeher/snapclient/blob/master/components/audio_hal/driver/es8388/es8388.c#L462-L468 this section to restore audio volume on my Muse Luxe

CarlosDerSeher, 2025-01-19:
@hjri what do you think about just reverting back to the original ADF driver and modify it only for AI thinker board? And what does this `reg -= 0x0c` do?

hjri, 2025-01-20:
It's been a long time, i don't remember exactly, according to documentation it should set amplifier gain to 0db instead of -3db or something, and it's still quiet

haven't experimented since, especially since snapclient just sorta dies after a while

CarlosDerSeher, 2025-01-20:
Ok then I'll just revert to the original driver and add the diff to the current one only for AI thinker

CarlosDerSeher, 2025-01-20:
@hjri if you find time you you could try this and see if it makes things better for you

hjri, 2025-01-20:
i'll give it a try and also compared to one of the "stock" firmwares as well just to confirm i'm not gaslighting myself

CarlosDerSeher, 2025-06-17:
So this is still an issue?

commit d1bdd3cf (2025-01-20): fix #101