CarlosDerSeher/snapclient #102: boot loop at Lyra-T-Mini V1.2

auhgit, 2024-09-23:
i try to get it running on a LyraT-Mini V1.2 and get a boot loop : 

i'm not a developer, but like to have snapclient running on a ESP32 ;-)

ELF file SHA256: 2696e7e39b03d96a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7428
load:0x40078000,len:16652
load:0x40080400,len:4
load:0x40080404,len:4284
entry 0x40080668
I (29) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (29) boot: compile time Jul 25 2024 08:19:47
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v3.1
I (37) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 ota_0            OTA app          00 10 00010000 001b0000
I (102) boot:  4 ota_1            OTA app          00 11 001c0000 001b0000
I (110) boot:  5 storage          Unknown data     01 82 00370000 00090000
I (118) boot: End of partition table
I (122) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=266cch (157388) map
I (174) esp_image: segment 1: paddr=000366f4 vaddr=3ffb0000 size=026bch (  9916) load
I (177) esp_image: segment 2: paddr=00038db8 vaddr=40080000 size=07260h ( 29280) load
I (189) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=7d7c4h (513988) map
I (331) esp_image: segment 4: paddr=000bd7ec vaddr=40087260 size=0859ch ( 34204) load
I (350) boot: Loaded app from partition at offset 0x10000
I (350) boot: Disabling RNG early entropy source...
I (361) cpu_start: Multicore app
I (362) quad_psram: This chip is ESP32-D0WD
I (362) esp_psram: Found 8MB PSRAM device
I (364) esp_psram: Speed: 80MHz
I (367) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (375) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (384) cpu_start: Pro cpu up.
I (388) cpu_start: Starting app cpu, entry point is 0x40081494
I (380) cpu_start: App cpu up.
I (862) esp_psram: SPI SRAM memory test OK
I (870) cpu_start: Pro cpu start user code
I (870) cpu_start: cpu freq: 240000000 Hz
I (870) cpu_start: Application information:
I (873) cpu_start: Project name:     snapclient
I (878) cpu_start: App version:      1
I (882) cpu_start: Compile time:     Jul 25 2024 08:21:58
I (888) cpu_start: ELF file SHA256:  2696e7e39b03d96a...
I (894) cpu_start: ESP-IDF:          v5.1.1
I (899) cpu_start: Min chip rev:     v0.0
I (904) cpu_start: Max chip rev:     v3.99 
I (909) cpu_start: Chip rev:         v3.1
I (914) heap_init: Initializing. RAM available for dynamic allocation:
I (921) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (927) heap_init: At 3FFB6AE0 len 00029520 (165 KiB): DRAM
I (933) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (939) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (946) heap_init: At 4008F7FC len 00010804 (66 KiB): IRAM
I (952) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (960) spi_flash: detected chip: gd
I (964) spi_flash: flash io: qio
W (968) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (981) app_start: Starting scheduler on CPU0
I (986) app_start: Starting scheduler on CPU1
I (986) main_task: Started on CPU0
I (996) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1004) main_task: Calling app_main()
I (1019) SC: Start codec chip
I (1024) DRV8311: ES8311 in Slave mode
W (1079) I2C_BUS: I2C bus has been already created, [port:0]
I (1081) SC: Audio board_init done
I (1081) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (1089) DRV8311: Enter into es8311_mute(), mute = 1

I (1091) SC: init player
I (1092) PLAYER: enable initial sync timer
I (1097) PLAYER: Start player_task
I (1101) PLAYER: init player done
I (1101) PLAYER: started sync task
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40100839  PS      : 0x00060830  A0      : 0x800da402  A1      : 0x3ffc1cd0  
A2      : 0xffffffff  A3      : 0x00000001  A4      : 0x3ffc4054  A5      : 0x00000000  
A6      : 0x00000003  A7      : 0x00060323  A8      : 0x8014b572  A9      : 0x3ffc1cb0  
A10     : 0x3ffb2be8  A11     : 0x00060820  A12     : 0x00000004  A13     : 0x00060823  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001f  LBEG    : 0x40087c49  LEND    : 0x40087c59  LCOUNT  : 0xfffffffe  

Backtrace: 0x40100836:0x3ffc1cd0 0x400da3ff:0x3ffc1cf0 0x400f888a:0x3ffc1d10

auhgit, 2024-09-25:
I just copy sdkconfig.lyra-t-mini to sdkconfig. I use idf 5.4  .

But at the moment i screwed up my toolchain and nothings works :-(

`/home/auh/ESP32-Dev/CarlosDerSeher-snapclient/components/flac/flac/src/libFLAC/stream_decoder.c:2774:103: error: passing argument 2 of 'FLAC__bitreader_read_rice_signed_block' from incompatible pointer type [-Wincompatible-pointer-types]
 2774 |                         if(!FLAC__bitreader_read_rice_signed_block(decoder->private_->input, residual + sample, u, rice_parameter))
      |                                                                                              ~~~~~~~~~^~~~~~~~
      |                                                                                                       |
      |                                                                                                       FLAC__int32 * {aka long int *}
In file included from /home/auh/ESP32-Dev/CarlosDerSeher-snapclient/components/flac/flac/src/libFLAC/stream_decoder.c:46:
/home/auh/ESP32-Dev/CarlosDerSeher-snapclient/components/flac/flac/src/libFLAC/include/private/bitreader.h:84:76: note: expected 'int *' but argument is of type 'FLAC__int32 *' {aka 'long int *'}
   84 | FLAC__bool FLAC__bitreader_read_rice_signed_block(FLAC__BitReader *br, int vals[], uint32_t nvals, uint32_t parameter);
      |                                                                        ~~~~^~~~~~
/home/auh/ESP32-Dev/CarlosDerSeher-snapclient/components/flac/flac/src/libFLAC/stream_decoder.c:2783:94: error: passing argument 2 of 'FLAC__bitreader_read_raw_int32' from incompatible pointer type [-Wincompatible-pointer-types]
 2783 |                                 if(!FLAC__bitreader_read_raw_int32(decoder->private_->input, &i, rice_parameter))
      |                                                                                              ^~
      |                                                                                              |
      |                                                                                              int *
/home/auh/ESP32-Dev/CarlosDerSeher-snapclient/components/flac/flac/src/libFLAC/include/private/bitreader.h:76:77: note: expected 'FLAC__int32 *' {aka 'long int *'} but argument is of type 'int *'
   76 | FLAC__bool FLAC__bitreader_read_raw_int32(FLAC__BitReader *br, FLAC__int32 *val, uint32_t bits);
      |                                                                ~~~~~~~~~~~~~^~~
[1179/1361] Building C object esp-idf/flac/CMakeFiles/__idf_flac.dir/flac/src/libFLAC/stream_encoder.c.obj
ninja: build stopped: subcommand failed.
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).`

And sorry again, i'm not a developer ...

CarlosDerSeher, 2024-09-25:
Please use ESP-IDF v5.1.1 as per the instructions

auhgit, 2024-09-25:
ahh ok, sorry, did not read that i have to use IDF 5.1.1 , just try with IDF 5.3.1 an i got a build, but can't test now. 

THX

closed 2024-09-25 by auhgit

CarlosDerSeher, 2024-09-26:
You could try https://sonocotta.github.io/esparagus-snapclient/