CarlosDerSeher/snapclient #106: Feature: DSP option: mix chanels

jagjordi, 2024-11-03:
I want to suggest an option for the DSP processor. I am using this for a multi-room audio setup and in some rooms I have a single speaker. It would be very good to be able to merge both left and right channels so that the speaker works as a MONO. 
Right now I am using a simple DAC plus a separate audio amplifier. So what I am doing is that I am joining both audio channels after the DAC with a simple two resistor circuit. I am thinking to creating a new board that uses one of the amplifiers with a built-in DAC. For that setup, it would not be possible to merge the two channels after they have been amplified. Without having to use big and expensive power resistors, This is why I think it would be a very nice option to be able to do this merging of the channels in the digital domain on the DSP

CarlosDerSeher, 2024-11-04:
Shouldn't be too hard. Mixing PCM samples is as simple as adding two samples, L+R. The question is, how we should handle int16_t over and underflows. Just simple clipping maybe?

normanalie, 2024-11-04:
Some DACs offer an option to sum the L+R audio channels. 
However, if you'd prefer to handle it DSP side, the approach taken by professional audio consoles is typically either a true L+R summing - where if the same signal is present in both L and R, the resulting sum will be 6dB louder - or an L+R-6dB summing, which maintains the same signal level as the original sources. 
For this use case, I think L+R-6dB would be the best approach, as we're not aiming for a real summation but rather to create a mono signal that retains maximum information in a single-speaker setup.

jagjordi, 2024-11-04:
Why not divide the signal by2 before summing? (out=L>>2 + R>>2)
That's what a two resistor circuit does.

CarlosDerSeher, 2024-11-04:
> Why not divide the signal by2 before summing?

You'll lose a lot of information that way, especially if there are low volume parts in one channel. Also it won't be necessary most of the time.

> true L+R summing - where if the same signal is present in both L and R, the resulting sum will be 6dB louder - or an L+R-6dB summing, which maintains the same signal level as the original sources.

As I am no audio expert, how do you do this implementation wise? Using some FIR / IIR filter structure. How do you analyze the audo sample before summing them? Is this done in the frequncy or time domain?

CarlosDerSeher, 2024-11-04:
After giving it another thought wouldn't it be best to just sum L+R using an int32, then divide it by 2 and cast back to int16?

jagjordi, 2024-11-04:
2 things:

For the addition I am more in favor of doing the /2 mix because the "perceived loudness" would be correct. If you add without dividing then you are amplifiying by a factor of 2.

For the implementation, I don;t know much about how the low level implementation on the ESP platform but if you divide by 2 before adding you dont need to cast to int32. A int16 divided by two has a maximum range of 15 bits, so when you add 2 of them the maximum range is 16 bit again. There is no possibility of overflow. you can also implement the /2 by right shifting 2. the >> operator should perform arithmetic shift (preserving the sign bit) when it shifts a signed number

CarlosDerSeher, 2024-11-04:
Watch out! `x/2 == x >> 1`

DerPicknicker, 2024-11-22:
Maybe for some of you are interested in real DSP hardware. The TAS5805M DAC has a DSP builtin. 

The board designer of those boards enabled the DSP features for snapcast see here: https://sonocotta.github.io/esparagus-snapclient/

miLORD1337, 2025-03-02:
Hi all,
I tried this using a PCM5102A. When it works, it works (no blatant noise issues). Unfortunately, I seldomly get sound at all, and sometimes it's way out of sync and skipping sections.

Other ESP snapclients with older firmware still work as expected with the same server.

I monitored the following:

```
I (39323) PLAYER: initial sync age: 1us, chunk duration: 24000us
W (39324) PLAYER: RESYNCING HARD 2: age 91089us, latency 1635351305206us, free 186204, largest block 110592, 0, rssi: -62
W (39397) PLAYER: RESYNCING HARD 1: age 27713us, latency 1635351305206us, free 180968, largest block 110592, rssi: -63
W (39402) PLAYER: RESYNCING HARD 1: age 8541us, latency 1635351305206us, free 184208, largest block 110592, rssi: -63
I (39425) PLAYER: DMA completely loaded
I (39427) PLAYER: initial sync age: 1us, chunk duration: 24000us
W (39429) PLAYER: RESYNCING HARD 2: age 99557us, latency 1635351305206us, free 177248, largest block 110592, 0, rssi: -63
I (39479) PLAYER: DMA completely loaded
I (39481) PLAYER: initial sync age: 1us, chunk duration: 24000us
W (39483) PLAYER: RESYNCING HARD 2: age 105471us, latency 1635351305206us, free 177248, largest block 110592, 0, rssi: -63
I (39496) PLAYER: DMA completely loaded
I (39515) PLAYER: initial sync age: 1us, chunk duration: 24000us
W (39539) PLAYER: RESYNCING HARD 2: age -586us, latency 1635351305206us, free 185088, largest block 110592, 0, rssi: -63
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400fe6cc  PS      : 0x00060730  A0      : 0x00000000  A1      : 0x3ffbfee0
0x400fe6cc: player_task at ../snapclient/components/lightsnapcast/player.c:1362

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb4c68  A5      : 0x40091548
A6      : 0xc4d772fa  A7      : 0x3ffbfee0  A8      : 0x800fe6c9  A9      : 0x3ffbfea0
A10     : 0x00000000  A11     : 0x3ffbd4b4  A12     : 0x00000001  A13     : 0xffffffff
A14     : 0x00000001  A15     : 0x3ffbd48c  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
0x4000c2e0: memcpy in ROM
0x4000c2f6: memcpy in ROM

Backtrace: 0x400fe6c9:0x3ffbfee0
0x400fe6c9: player_task at ../snapclient/components/lightsnapcast/player.c:1362

ELF file SHA256: 9a92752009366fe9

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:5392
load:0x40078000,len:16204
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3500
entry 0x400805cc
I (387) SC: Start codec chip
I (387) HAL: INIT
I (388) HAL: codec_hal done
I (388) AUDIO_BOARD: board-handle done
I (390) SC: Audio board_init done
I (394) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (399) PLAYER: deinit player done
I (403) PLAYER: player_setup_i2s: dma_buf_len is 100, dma_buf_count is 22, sample rate: 44100, bits: 16
I (414) PLAYER: enable initial sync timer
I (418) PLAYER: Start player_task
I (422) PLAYER: init player done
I (422) PLAYER: started sync task
I (444) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (517) WIFI_IF: wifi_init_sta finished. Trying to connect to ***
I (517) UI_HTTP: Initializing SPIFFS
I (527) WIFI_IF: retry to connect to the AP
[... 33 lines cut ...]
I (6735) SC: sta mac: ***:***:***:***:***:***
I (6739) SC: netconn sent hello message
I (6750) SC: Buffer length:  500
I (6750) SC: Latency:        0
I (6750) SC: Mute:           0
I (6754) SC: Setting volume: 100
I (6758) dspProc: Set volume to 1.000000
I (6762) dspProc: Set volume to 1.000000
I (6768) SC: ############### CODEC
I (6771) SC: fLaC sampleformat: 48000:16:2
I (6830) PLAYER: player_setup_i2s: dma_buf_len is 100, dma_buf_count is 22, sample rate: 48000, bits: 16
I (6831) PLAYER: created new queue with 20
I (6834) PLAYER: snapserver config changed, buffer 500ms, chunk 1152 frames, sample rate 48000, ch 2, bits 16 mute 0 latency 0
I (7628) SC: latency buffer full
I (7636) PLAYER: DMA completely loaded
I (7768) PLAYER: initial sync age: 1us, chunk duration: 24000us
W (7770) PLAYER: RESYNCING HARD 2: age 91125us, latency 1635390778881us, free 182440, largest block 110592, 0, rssi: -50
W (7995) PLAYER: RESYNCING HARD 1: age 179838us, latency 1635390778881us, free 176196, largest block 102400, rssi: -50
W (7999) PLAYER: RESYNCING HARD 1: age 160471us, latency 1635390778881us, free 176736, largest block 102400, rssi: -50
W (8011) PLAYER: RESYNCING HARD 1: age 148609us, latency 1635390778881us, free 178664, largest block 102400, rssi: -50
```

...and lots more of the last four lines.

It's possibly unrelated to this feature. Should I open a new issue instead?

CarlosDerSeher, 2025-03-02:
> and sometimes it's way out of sync and skipping sections.

Could #116 be your solution ?

I've listened to music using the mix feature today and it seemed to work OK but there was 1 song where literally something was missing in the audio. It was still there but very low volume. Possibly destructive interference by mixing L+R ?

DerPicknicker, 2025-03-03:
Your buffer is too small.. 500ms is way too less for the bad WiFi of an esp32. Try 1000ms.

miLORD1337, 2025-03-03:
Isn't 700 ms the recommendation? But regardless, as I wrote above, other ESPs with older firmware work a treat in this setup.

DerPicknicker, 2025-03-03:
@miLORD1337 .. If you live in a urban area with a lot of 2.4ghz networks 700ms are too less for an esp32. Those WiFi-ICs are bad for such traffic intense workloads. 

I would try to use 1000ms as an buffer.

miLORD1337, 2025-03-03:
I tried it just for good measure. Same result, unfortunately:

```
I (3922) SC: ############### CODEC
I (3925) SC: fLaC sampleformat: 48000:16:2
I (3947) PLAYER: player_setup_i2s: dma_buf_len is 100, dma_buf_count is 22, sample rate: 48000, bits: 16
I (3948) PLAYER: created new queue with 41
I (3951) PLAYER: snapserver config changed, buffer 1000ms, chunk 1152 frames, sample rate 48000, ch 2, bits 16 mute 0 latency 0
I (4521) SC: latency buffer full
I (4669) PLAYER: DMA completely loaded
I (5275) PLAYER: initial sync age: 1us, chunk duration: 24000us
W (6572) PLAYER: RESYNCING HARD 2: age -37335us, latency 1729567970229us, free 179540, largest block 65536, 0, rssi: -62
W (8120) PLAYER: RESYNCING HARD 1: age 1465138us, latency 1729567970229us, free 177044, largest block 110592, rssi: -67
W (8124) PLAYER: RESYNCING HARD 1: age 1445923us, latency 1729567970229us, free 176212, largest block 110592, rssi: -67
W (8136) PLAYER: RESYNCING HARD 1: age 1433992us, latency 1729567970229us, free 178760, largest block 110592, rssi: -67
```
and also
```
I (3853) SC: ############### CODEC
I (3856) SC: fLaC sampleformat: 48000:16:2
I (3921) PLAYER: player_setup_i2s: dma_buf_len is 100, dma_buf_count is 22, sample rate: 48000, bits: 16
I (3923) PLAYER: created new queue with 41
I (3926) PLAYER: snapserver config changed, buffer 1000ms, chunk 1152 frames, sample rate 48000, ch 2, bits 16 mute 0 latency 0
I (4232) SC: latency buffer full
I (4241) PLAYER: DMA completely loaded
W (5113) PLAYER: couldn't get memory to insert chunk, inserting an chunk containing just 0
W (5113) PLAYER: 10632, 3328, 10584, 3328
I (5132) PLAYER: initial sync age: 1us, chunk duration: 24000us
E (5162) SC: write_callback, failed to allocate PCM chunk payload
E (5162) SC: http_get_task: FLAC__stream_decoder_process_single failed
W (5225) PLAYER: couldn't get memory to insert chunk, inserting an chunk containing just 0
W (5225) PLAYER: 5320, 1920, 19272, 13824
W (5227) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5235) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5243) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5250) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5257) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5264) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5271) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5279) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5286) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5293) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5300) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5308) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5315) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5322) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5329) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5336) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5344) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5351) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5358) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5365) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5372) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5380) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5387) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5394) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5401) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5408) SC: malloc decoderChunk.inData failed, wait 1ms and try again
W (5416) SC: malloc decoderChunk.inData failed, wait 1ms and try again
E (10421) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (10421) task_wdt:  - IDLE1 (CPU 1)
E (10421) task_wdt: Tasks currently running:
E (10421) task_wdt: CPU 0: IDLE0
E (10421) task_wdt: CPU 1: http
E (10421) task_wdt: Print CPU 1 backtrace

Backtrace: 0x40084EF6:0x3FFB2290 0x40082C35:0x3FFB22B0 0x400D85A7:0x3FFD6AB0
--- 0x40084ef6: esp_crosscore_isr at .../components/esp_system/crosscore_int.c:96
0x40082c35: _xt_lowint1 at .../components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240
0x400d85a7: http_get_task at .../snapclient/main/main.c:1484
```

CarlosDerSeher, 2025-03-03:
I see your problem in the log. You are using 48kHZ sample rate and flac with 20ms chunk size?
Your log has this line:
`I (5275) PLAYER: initial sync age: 1us, chunk duration: 24000us`
But it should look like this
`I (277400) PLAYER: initial sync age: 6us, chunk duration: 26122us`

Question is why this is incorrect...

miLORD1337, 2025-03-03:
> You are using 48kHZ sample rate and flac with 20ms chunk size?

That's correct. Maybe we could check this at @duardda's setup as well and see if it's the same wrong duration?

CarlosDerSeher, 2025-03-03:
Check around here: https://github.com/CarlosDerSeher/snapclient/blob/c883af58ad900a150ad2f4b06e42a718297cca54/components/lightsnapcast/player.c#L1303

The code should set it to the correct value, `chunkDuration_us` is initialized at task creation with 24000 and it seems it never enters the if ?? Strange, as it works for me.

CarlosDerSeher, 2025-03-04:
@miLORD1337 sorry for the confusion. It is right after all as you have your sample rate set to 48kHz which results in 24ms chunk duration. I've got 44.1kHz and therefore 26.122ms

CarlosDerSeher, 2025-03-04:
> Which one was that, btw? I'm interested in what that sounds like.

https://youtu.be/0TfqbuTBqX8

At 1:00 you  an hear it.

CarlosDerSeher, 2025-03-04:
@miLORD1337 maybe we should really open another issue for the discussion of the resync issue of yours or this thread will become confusing.

miLORD1337, 2025-03-05:
> @miLORD1337 maybe we should really open another issue for the discussion of the resync issue of yours or this thread will become confusing.

Agreed, please see #117.

miLORD1337, 2025-04-07:
I backported this to the last working commit I found in #117 (c2285846f4dff518f2db6799858a55cbf9e003ac). No blatant noise issues, works fine 👍 

As OP didn't report back, I thought I'd chime in.

CarlosDerSeher, 2025-04-28:
Please use develop branch

sobchek, 2025-04-29:
> Please use develop branch

@CarlosDerSeher Thank you. I'd like to backport just the DSP improvement (Mono Mixdown) without pulling in new network and snapcast changes. I attempted earlier today but, well, failed.

Is there a minimal set of files you'd recommend pulling?

CarlosDerSeher, 2025-04-29:
Try just pulling in dsp_processor component. If this doesn't work you could ask @miLORD1337 how he did it.

commit 4eced44a (2025-02-23): fix low volume due to wrong init of audioDAC_data
add support for mixing stereo to mono through dsp_processor (fixes #106)
commit 4eb01805 (2025-03-01): fix low volume due to wrong init of audioDAC_data
add support for mixing stereo to mono through dsp_processor (fixes #106)
add sdkconfigs to github action workflow
suppress warnings from flac git submodule
remove unused variable and address some compiler warnings
remove some dead code