CarlosDerSeher/snapclient #112: Feature: ESP32-S3

Dotsch2005, 2025-02-11:
Is it possible to run the snapclient on an ESP32-S3?

I have changed the sdkconfig target with `idf.py set-target esp32s3` and changed the confiuration via `idf.py menuconfig`.

I am stuck at `error: ‘I2S_CLK_SRC_APLL’ undeclared`.

I have read so that APLL is not available on the ESP32-S3. 

Is there a way to get the ESP32-S3 up and running?

CarlosDerSeher, 2025-02-11:
Please try the branch [fix_20ms_chunk_size_bug](https://github.com/CarlosDerSeher/snapclient/tree/fix_20ms_chunk_size_bug). Latest bug fixes and updates ar in there. They wilI be merged to master soon but there still has to be some testing.

I don't know about S3 but there is a template sdkconfig for S2 which also doesn't have APLL. Maybe you can work your way from there

CarlosDerSeher, 2025-02-11:
Not sure if this could be a problem here, but in this branch there has been an upgrade to IDF v5.1.5

Also it dies at HAL init, which codec are you using?

Dotsch2005, 2025-02-11:
Thank you for your support.

I have updated to version v5.1.5 and started fresh again. Unfortunately, nothing has changed. I have also increased the debug log, but after the HAL: INIT I see nothing.

I assume that the codec means the audio board. I am using the TAS5805M here. (ESP32-S3 Louder)

Dotsch2005, 2025-02-11:
I'm sorry, I feel stupid. The GPIO ports have changed between the ESP32 version and the ESP32-S3.

However, it now aborts after the following:
```
V (1556) partition: Partition table MD5 verified
I (1565) SC: Start codec chip
I (1565) HAL: INIT
W (1568) TAS5805M: Power down pin: 33
W (1592) TAS5805M: Setting to HI Z
W (1602) TAS5805M: Setting to PLAY
I (1602) HAL: codec_hal done
I (1602) AUDIO_BOARD: board-handle done
I (1603) SC: Audio board_init done
I (1607) AUDIO_HAL: Codec mode is 2, Ctrl:1
W (1612) PLAYER: no sync task created?
W (1616) PLAYER: no pcm chunk queue created?
W (1621) PLAYER: no latency buffer semaphore created?
I (1627) PLAYER: deinit player done
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759a1
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x18dc
load:0x403c9700,len:0x4
load:0x403c9704,len:0xd50
load:0x403cc700,len:0x3068
entry 0x403c9918
I (26) boot: ESP-IDF v5.1.5 2nd stage bootloader
```

Dotsch2005, 2025-02-11:
The Espressif documentation specifies the following processor `Xtensa® dual-core 32-bit LX7 microprocessor (with single precision FPU)`.

The TAS5805M power down pin 33 (should be 17) is also wrong, but can I find the configuration for it?

CarlosDerSeher, 2025-02-11:
Search for this line, `no sync task created` I am pretty sure your problem is related to failed task creation. Player task is hard-coded to core 1. If you change all tasks assigned to core 1 to "no affinity" you are hopefully good to go?!

CarlosDerSeher, 2025-02-11:
You can change the pinout from menuconfig audio hal

CarlosDerSeher, 2025-02-11:
normal boot log should look like this, so you are good until it fails...

```
I (375) SC: Start codec chip
I (376) HAL: INIT
I (376) ADAU1961: Power ON CODEC
I (376) ADAU1961: Looking for a adau1961 chip at address 0x70
I (381) ADAU1961: Found a adau1961 chip at address 0x70
I (391) HAL: codec_hal done
I (391) AUDIO_BOARD: board-handle done
I (395) SC: Audio board_init done
I (399) AUDIO_HAL: Codec mode is 2, Ctrl:1
W (405) PLAYER: no sync task created?
W (408) PLAYER: no pcm chunk queue created?
W (413) PLAYER: no latency buffer semaphore created?
I (418) PLAYER: deinit player done
I (423) PLAYER: player_setup_i2s: dma_buf_len is 100, dma_buf_count is 22, sample rate: 44100, bits: 16
I (433) PLAYER: enable initial sync timer
I (437) PLAYER: Start player_task
I (441) PLAYER: init player done
I (441) PLAYER: started sync task
```

luar123, 2025-02-11:
Just tried it here:

* I think `no sync task created` is normal at startup because `init_player()` calls `deinit_player()` first.
* Check your port config. I got similar errors initially when some i2s pins where not valid.
* I a coming further, but then it crashes with a stack overflow:
```
 (9049) NETF: Setup mdns
I (9052) dspProc: dsp_processor_init: init done
I (9056) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (9066) SC: try connecting to static configuration 192.168.x.x:1704
I (9073) main_task: Returned from app_main()
I (9079) SC: netconn connected
I (9083) SC: netconn sent hello message

***ERROR*** A stack overflow in task IDLE1 has been detected.

Backtrace: 0x4037590a:0x3fca1f40 0x4037c10d:0x3fca1f60 0x4037c8a2:0x3fca1f80 0x4037d101:0x3fca2000 0x4037c1c4:0x3fca2020 0x4037c1ba:0x00060023 |<-CORRUPTED
--- 0x4037590a: panic_abort at esp-idf-v5.1.5/components/esp_system/panic.c:466
0x4037c10d: esp_system_abort at esp-idf-v5.1.5/components/esp_system/port/esp_system_chip.c:84
0x4037c8a2: vApplicationStackOverflowHook at esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:582
0x4037d101: vTaskSwitchContext at esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/tasks.c:3841
0x4037c1c4: _frxt_dispatch at esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:450
0x4037c1ba: _frxt_int_exit at esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:245
```

CarlosDerSeher, 2025-02-11:
@luar123 you are right. I'll change these `warnings` to `verbose` logs in a future commit, as they are somewhat missleading. Changed it locally on my dev machine for now.

CarlosDerSeher, 2025-02-11:
@luar123 oh didn't know about this. What exactly does this `idf.py save-defconfig` do?

luar123, 2025-02-11:
It creates a `sdkconfig.defaults` file that includes all non-default values from your `sdkconfig` file. It is never changed by `idf` but is applied when no `sdkconfig` file is present, like when running `set-target`.
See https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#using-sdkconfig-defaults

CarlosDerSeher, 2025-02-11:
This sounds like something for the future commit list too. Thanks

luar123, 2025-02-11:
Seems we could even split it in multiple files, one for base configs (maybe target dependent) and one for the different audio board settings: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#custom-sdkconfig-defaults

CarlosDerSeher, 2025-02-11:
This would have been my next question :) as the IDLE task stack size was never a problem on ESP32

CarlosDerSeher, 2025-02-11:
So a file sdkconfig.defaults.esp32s3 should look like this

```
CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=768
```

luar123, 2025-02-11:
I have tried only the default size 1536.

CarlosDerSeher, 2025-02-11:
So I decreased this myself at some point while optimizing RAM usage? Totally forgot about this

Dotsch2005, 2025-02-17:
Thank you very much [CarlosDerSeher](https://github.com/CarlosDerSeher) and [luar123](https://github.com/luar123)

I have now tried a few things and was able to solve the first problem. After help from [anabolyc](https://github.com/anabolyc) I was able to get the board to run.

The problem here was with the [fix_20ms_chunk_size_bug ](https://github.com/CarlosDerSeher/snapclient/tree/fix_20ms_chunk_size_bug) was with the CustomBoard TAS5805M. I used the TAS5805M component from [features/s3-support](https://github.com/sonocotta/esparagus-snapclient/tree/features/s3-support).

Now I can get the board to work with WiFi. However, when I start streaming, the following error occurs:
```
I (2661) OTA: idf.py build ; curl snapclient.local:8032 --data-binary @- < build/snapclient.bin
I (2671) SC: try connecting to static configuration 192.168.1.100:1704
I (2681) main_task: Returned from app_main()
I (2691) SC: netconn connected
I (2691) SC: netconn sent hello message
I (2861) SC: Buffer length:  1000
I (2861) SC: Latency:        0
I (2861) SC: Mute:           0
I (2861) SC: Setting volume: 72
I (2861) dspProc: Set volume to 0.720000
I (2871) dspProc: Set volume to 0.720000
I (2871) SC: pcm sampleformat: 48000:16:2
I (3071) SC: latency buffer full
I (986831) dspProc: got new setting for dspfEQBassTreble
W (986841) PLAYER: pcm chunk queue not created
W (986861) PLAYER: pcm chunk queue not created
I (986871) PLAYER: player_setup_i2s: dma_buf_len is 100, dma_buf_count is 22, sample rate: 48000, bits: 16
W (986871) PLAYER: no pcm chunk queue created?
I (986871) PLAYER: created new queue with 48
I (986881) PLAYER: snapserver config changed, buffer 1000ms, chunk 960 frames, sample rate 48000, ch 2, bits 16 mute 0 latency 0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4202a675  PS      : 0x00060830  A0      : 0x803816fc  A1      : 0x3fcb3d80
--- 0x4202a675: player_task at D:/Dev/Github/CarlosDerSeher/snapclient/components/lightsnapcast/player.c:1371

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3c111f94  A5      : 0x3fcd8460
A6      : 0x00000001  A7      : 0x3fcb3d80  A8      : 0x8202a672  A9      : 0x3fcb3d40
A10     : 0x00000000  A11     : 0x3fcb3da4  A12     : 0x3fca91e4  A13     : 0x0000000a  
A14     : 0x00000001  A15     : 0x00000f00  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000c  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000
--- 0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0x4202a672:0x3fcb3d80 0x403816f9:0x3fcb3ed0
--- 0x4202a672: player_task at D:/Dev/Github/CarlosDerSeher/snapclient/components/lightsnapcast/player.c:1371
0x403816f9: vPortTaskWrapper at D:/Apps/esp/v5.1.5/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 5187c5dd295e0e57

Rebooting...
```
The ESP32S3 has a direct JTAG driver. I have tried to debug from here with VSCode and the ESP-IDF/OpenOCD. Unfortunately I can't get a working debug environment to run.

luar123, 2025-02-17:
Did you enable PSRAM (I think most/all S3 modules have PSRAM)? Otherwise 1000ms buffer length is too much. You would need to reduce it in the snapserver config to ~700ms (see readme).

CarlosDerSeher, 2025-02-17:
> The problem here was with the fix_20ms_chunk_size_bug and the CustomBoard TAS5805M. I used the TAS5805M component from features/s3-support. With this it starts and connects to the snapcast server.

You probably have a wrong pin out configured then? I never tried this DAC myself, but others reported it works from customBoard.

> --- 0x4202a675: player_task at D:/Dev/Github/CarlosDerSeher/snapclient/components/lightsnapcast/player.c:1371

What's the code at this line?

Dotsch2005, 2025-02-18:
> Did you enable PSRAM (I think most/all S3 modules have PSRAM)? Otherwise 1000ms buffer length is too much. You would need to reduce it in the snapserver config to ~700ms (see readme).

Thanks for the hint, I have changed it. In your example, the PSRAM was not activated. I have activated the PSRAM in my current configuration (8MB - ESP32-S3-WROOM-N16R8).

> You probably have a wrong pin out configured then? I never tried this DAC myself, but others reported it works from customBoard.
>
>
> What's the code at this line?

In the current branch [fix_20ms_chunk_size_bug](https://github.com/CarlosDerSeher/snapclient/tree/fix_20ms_chunk_size_bug) it is line [1351](https://github.com/CarlosDerSeher/snapclient/blob/38d749e6cc305cddfc62241fc60ffcde8056cf55/components/lightsnapcast/player.c#L1351) (old [1371](https://github.com/CarlosDerSeher/snapclient/blob/0cc92a25977c850f3e4038aa4a3f90976d02ea23/components/lightsnapcast/player.c#L1371))
```
fragment = chnk->fragment;
p_payload = fragment->payload;
size = fragment->size;
```

I think that `chnk->fragment` is empty and therefore no memcopy can be made.

CarlosDerSeher, 2025-02-18:
@Dotsch2005 what's your server config? mainly chunk size. Which codec do you use? PCM I see from your log? You could try enabling [this line](https://github.com/CarlosDerSeher/snapclient/blob/38d749e6cc305cddfc62241fc60ffcde8056cf55/components/lightsnapcast/player.c#L1347) and maybe add chnk->fragment and fragment->payload pointers to this print just to see if they are NULL

Dotsch2005, 2025-02-18:
Yes, the codec is set to PCM in the Snapcast Config. Here is the config:

```
#######################################################################################
#     ______                                                                          #
#    / _____)                                                                         #
#   ( (____   ____   _____  ____    ___  _____   ____  _   _  _____   ____            #
#    \____ \ |  _ \ (____ ||  _ \  /___)| ___ | / ___)| | | || ___ | / ___)           #
#    _____) )| | | |/ ___ || |_| ||___ || ____|| |     \ V / | ____|| |               #
#   (______/ |_| |_|\_____||  __/ (___/ |_____)|_|      \_/  |_____)|_|               #
#                          |_|                                                        #
#                                                                                     #
#  Snapserver config file                                                             #
#                                                                                     #
#  Source: https://github.com/badaix/snapcast/blob/develop/server/etc/snapserver.conf #                                                                          #
#                                                                                     #
#######################################################################################

# default values are commented
# uncomment and edit to change them

# Settings can be overwritten on command line with:
#  "--<section>.<name>=<value>", e.g. --server.threads=4

# General server settings #####################################################
#
[server]
# Number of additional worker threads to use
# - For values < 0 the number of threads will be 2 (on single and dual cores)
#   or 4 (for quad and more cores)
# - 0 will utilize just the processes main thread and might cause audio drops
#   in case there are a couple of longer running tasks, such as encoding
#   multiple audio streams
#threads = -1

# the pid file when running as daemon
#pidfile = /var/run/snapserver/pid

# the user to run as when daemonized
#user = snapserver
# the group to run as when daemonized
#group = snapserver

# directory where persistent data is stored (server.json)
# if empty, data dir will be
#  - "/var/lib/snapserver/" when running as daemon
#  - "$HOME/.config/snapserver/" when not running as daemon
datadir = /data/

#
###############################################################################

# HTTP RPC ####################################################################
#
[http]
# enable HTTP Json RPC (HTTP POST and websockets)
enabled = true

# address to listen on, can be specified multiple times
# use "0.0.0.0" to bind to any IPv4 address or :: to bind to any IPv6 address
# or "127.0.0.1" or "::1" to bind to localhost IPv4 or IPv6, respectively
# use the address of a specific network interface to just listen to and accept
# connections from that interface
bind_to_address = 0.0.0.0

# which port the server should listen to
port = 1780

# serve a website from the doc_root location
# disabled if commented or empty
[... 165 lines cut ...]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4202b1cc  PS      : 0x00060c30  A0      : 0x803816fc  A1      : 0x3fcb3fc0  
--- 0x4202b1cc: player_task at D:/Dev/Github/CarlosDerSeher/snapclient/components/lightsnapcast/player.c:1352 (discriminator 13)

A2      : 0x00000000  A3      : 0x3c111f94  A4      : 0x3fcdd984  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x3fcb3fc0  A8      : 0x8202b1ca  A9      : 0x3fcb3fa0  
A10     : 0x00000d53  A11     : 0x3c0e0348  A12     : 0x3c0e0b20  A13     : 0x00000d53
A14     : 0x3c0e0348  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe
--- 0x400556d5: strlen in ROM
0x400556e5: strlen in ROM

Backtrace: 0x4202b1c9:0x3fcb3fc0 0x403816f9:0x3fcb4110
--- 0x4202b1c9: player_task at D:/Dev/Github/CarlosDerSeher/snapclient/components/lightsnapcast/player.c:1352 (discriminator 13)
0x403816f9: vPortTaskWrapper at D:/Apps/esp/v5.1.5/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: ad0965b5a1141297

Rebooting...
```
The problem is that the chnk object is not filled.

What does this function do after the `while (1) {`?

anabolyc, 2025-05-29:
Not sure if that helps, but some time ago I managed to build and run [snapclient code](https://github.com/sonocotta/esparagus-snapclient/tree/features/s3-support) on the [Louder-ESP32-S3](https://www.tindie.com/products/sonocotta/louder-esp32/) and [Esparagus Echo](https://github.com/sonocotta/esparagus-echo) with minimal code changes. The problem I didn't have time to solve (and even investigate properly) is that it was terribly stuttering, with a rain of warnings in the log.

I mentioned that before, if some of you guys have a commitment to make it work, I'm willing to share the above boards at my own expense.

anabolyc, 2025-05-31:
Just FIY, there seems to be an [ESPHome port](https://github.com/esphome/esphome/pull/8350#issuecomment-2923667355) of snapclient. I'm planning to test it with S3, but with regular ESP32 seems to work quite well. The benefit is that it adds flexible configuration options via ESPHome config, and with the latest TAS5805m driver, all the capabilities of the DAC will be directly configurable on the fly, 15-band EQ to be noted among others

anabolyc, 2025-06-02:
I can confirm that it works equally well on Louder-ESP32 and Louder-ESP32-S3 boards

The TAS5805M EQ works as expected (apart from the known settings restore bug, in progress). If it were combined with TAS5805m power states and digital volume, this would be the best ESP32 snapcast implementation to date.

And the ESPHomes "live" UI is so to the benefit of the component

[image]

luar123, 2025-06-02:
The esphome component uses the code from this repository (master branch from February) except for DAC/I2C/wifi handling, so it should work natively as well. Maybe some tuning of sdkconfigs is needed for S3.

anabolyc, 2025-11-14:
Quick update. 

With a few minor tweaks, I managed to get a stable, nice-working build for ESP32-S3 from the master branch with minimal changes. The thing that is failing me is Improv-WiFi, which is not coded for anything but ESP32. I think hardcoded UART0 is a problem, and I'm looking into that. As soon as it is fixed, I will push a PR and update my [web-installer](https://sonocotta.github.io/esparagus-snapclient/) to support known S3 boards

anabolyc, 2025-11-20:
I think this issue can be closed now