CarlosDerSeher/snapclient #113: Merge with ESP Mesh-Lite

pillepalle127, 2025-02-13:
Awesome project!
I am trying to merge it with ESP Mesh-Lite.
So far it runs on the main branch for 5min until crash. But on the branch [fix_20ms_chunk_size_bug](https://github.com/CarlosDerSeher/snapclient/tree/fix_20ms_chunk_size_bug) it just crashes right away somewhere about trying start http.
However I couldn't find any closely related change. 
Any hint?

I (1059) HTTP: Start http task=0.0.0.0
I (1059) HTTP: Starting server on 0.0.0.0:80
I (1059) HTTP: Starting HTTP Server on port: '80'
E (1065) httpd: httpd_server_init: error in creating msg socket (23)
E (1072) HTTP: Failed to start file server!
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x401056ff
--- 0x401056ff: http_server_task at /home/pillepalle/Dokumente/elektronik/ESP_workspace_snapclient/snapclient_20ms/components/ui_http_server/ui_http_server.c:394 (discriminator 1)

file: "./components/ui_http_server/ui_http_server.c" line 394
func: http_server_task
expression: start_server("/html", CONFIG_WEB_PORT)

abort() was called at PC 0x40089a6f on core 0
--- 0x40089a6f: _esp_error_check_failed at /home/pillepalle/Dokumente/elektronik/Espressif-IDE/IDF/esp-idf-v5.1.5/components/esp_system/esp_err.c:50

CarlosDerSeher, 2025-02-13:
Ahhh I like it. I was thinking about this too. Unfortunately I don't have any idea about the cause. Maybe try to just disable the http server as a test? Also check your Partition layout and if you have all the files for the server on it. The spiffs partition should be created and flashed during build/flash.

CarlosDerSeher, 2025-02-13:
And what exactly causes the error check to fail? You can find out more details by looking in start server maybe?

pillepalle127, 2025-02-14:
Thanks for the quick reaction. I am trying to keep up...
No removed everything related to HTTP. Takes longer to crash but still crashing. Takes about 5sec (before maybe 2sec). 
My changes are pretty small & dumb; I just replaced the content of the wifi_init() with the example https://github.com/espressif/esp-mesh-lite/tree/master/examples/mesh_local_control

I dumped the whole repo here:
https://github.com/pillepalle127/snapclientmesh_bf

`W (941) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
I (952) [vendor_ie]: esp_wifi_connect return ESP_OK 
I (955) SC: Connected to AP
I (1108) wifi:ap channel adjust o:1,1 n:6,1
I (1109) wifi:new:<6,1>, old:<1,1>, ap:<6,1>, sta:<6,1>, prof:1
I (1109) wifi:state: init -> auth (b0)
I (1113) NETF: Setup mdns
I (1117) dspProc: dsp_processor_init: init done
I (1120) wifi:state: auth -> assoc (0)
I (1124) main_task: Returned from app_main()
I (1130) wifi:state: assoc -> run (10)
I (1157) wifi:connected with xxxxxx, aid = 5, channel 6, 40U, bssid = 98:9b:cb:f6:96:60
I (1157) wifi:security: WPA2-PSK, phy: bgn, rssi: -25
I (1159) wifi:pm start, type: 1

I (1162) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (1176) wifi:<ba-add>idx:0 (ifx:0, 98:9b:cb:f6:96:60), tid:6, ssn:2, winSize:64
I (1193) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2171) esp_netif_handlers: sta ip: 192.168.1.149, mask: 255.255.255.0, gw: 192.168.1.1
I (2172) bridge_wifi: Connected with IP Address:192.168.1.149
I (2176) bridge_common: [WIFI_AP_DEF ]Name Server1: 192.168.1.4
I (2183) [vendor_ie]: RTC store: temp_mesh_id:149; ssid:xxxxxx; bssid:00:00:00:00:00:00; crc:2034612399 
I (2196) WIFI: TCP client write task is running
I (10955) WIFI: System information, channel: 6, layer: 1, self mac: a8:42:e3:ae:88:44, parent bssid: 98:9b:cb:f6:96:60, parent rGuru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40089c4e  PS      : 0x00050033  A0      : 0x40089b0b  A1      : 0x3ffbc7c0  
--- 0x40089c4e: vPortYieldFromInt at /home/xxxxxx/Dokumente/elektronik/Espressif-IDE/IDF/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:647
0x40089b0b: _frxt_int_exit at /home/xxxxxx/Dokumente/elektronik/Espressif-IDE/IDF/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:245

A2      : 0xffffffb0  A3      : 0x00000000  A4      : 0x000002db  A5      : 0x3ffb22d0  
A6      : 0x00000001  A7      : 0x3ffb6efc  A8      : 0x40082bd2  A9      : 0x3f428800  
--- 0x40082bd2: _xt_lowint1 at /home/xxxxxx/Dokumente/elektronik/Espressif-IDE/IDF/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1233

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x80089cf9  A13     : 0x3ffb22c0  
A14     : 0x3ffb4138  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xffffffb0  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40089c4b:0x3ffbc7c0 0x40089b08:0x3ffbc7d0 0x0005001e:0xa5a5a5a5 |<-CORRUPTED
--- 0x40089c4b: vPortYieldFromInt at /home/xxxxxx/Dokumente/elektronik/Espressif-IDE/IDF/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:646
0x40089b08: _frxt_int_exit at /home/xxxxxx/Dokumente/elektronik/Espressif-IDE/IDF/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:244

`

pillepalle127, 2025-02-20:
Merging snapclient-only scope (drop everything networking related) and mesh_lite example into an empty project runs just fine. 
I neither nailed down where the conflict is leading to the cyclic resets neither what is getting lost by removing your huge networking scope so far though.

CarlosDerSeher, 2025-02-20:
I am not sure if i understand what you are doing.

I am using lwip netconn api maybe there is your problem?