CarlosDerSeher/snapclient #119: Interesting crash **always** reproducable under specific conditions

2q2code, 2025-04-02:
```
W (8033) PLAYER: RESYNCING HARD 2: age -5494us, latency 1380494318482us, free 181680, largest block 94208, 1, rssi: -46
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f7670  PS      : 0x00060730  A0      : 0x00000000  A1      : 0x3ffc0080  
--- 0x400f7670: player_task at /project/components/lightsnapcast/player.c:1351

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb4ca8  A5      : 0x400b712c  
A6      : 0x6c4dc08e  A7      : 0x3ffc0080  A8      : 0x800f766d  A9      : 0x3ffc0040  
A10     : 0x00000000  A11     : 0x3ffdff0c  A12     : 0x00000001  A13     : 0xffffffff  
A14     : 0x00000001  A15     : 0x3ffdfee4  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
--- 0x4000c2e0: memcpy in ROM
0x4000c2f6: memcpy in ROM

Backtrace: 0x400f766d:0x3ffc0080
--- 0x400f766d: player_task at /project/components/lightsnapcast/player.c:1351

ELF file SHA256: a7da2a17d012ae30

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:5392
load:0x40078000,len:16204
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

load:0x40080404,len:3500
entry 0x400805cc
```
This crash occurs at a peculiar junction but only with a particular combination of apps!

Prerequisite Setup:

- A Snapcast server with a loopback channel set up to allow MPD to play through it
- Rigellan for iOS (available freely)
- Rigellan set up to control your MPD installation

Steps:

1. Using Rigellan, Start any (ALAC tracks, but it doesn't matter, as this is Snapcast working directly with the loopback channel on the soundcard) track. This alone may cause a reboot, but it may not.
2. Without killing the Rigellan app, swipe away to any other app.
3. Return to the Rigellan app. This will trigger the above crash.

Note:

Using mpc (linux terminal client) causes NO crashes. I am guessing that a certain combination of requests are sent by Rigellan to the MPD server.... but as that has nothing to do with the Snapcast server directly, I have NO IDEA what is going on here.

Vague Diagram of Comms:

Rigellan --> MPD -> ALSA Loopback Stream --> Snapcast Server --> Snapcast Client (1+)

🤷‍♀

2q2code, 2025-04-02:
The relevant line of the `snapcast.conf` is:
`source = alsa:///?name=MPD&device=hw:0,1,0`

In `mpd.conf`:

```
audio_output {
	type		"alsa"
	name		"ALSA Loopback Device (hw:0,1,0)"
	device	"hw:0,0,0"	
	format  "48000:16:2"
}
```

2q2code, 2025-04-02:
This never occurs with Spotify usage.

CarlosDerSeher, 2025-04-02:
Try 44100kHz this could be related to #117

Master seems to have become unstable during the latest commits.

2q2code, 2025-04-02:
This issue existed since we began chatting. It has taken so long for me to get my computer all back up and running that I was curious whether it came up and was fixed already. I'll reconfigure everything for 44.1 and let you know. On Apr 2, 2025 5:22 PM, from Carlosderseher ***@***.***>
  Try 44100kHz this could be related to #117
Master seems to have become unstable during the latest commits.

Try 44100kHz this could be related to #117
Master seems to have become unstable during the latest commits.