CarlosDerSeher/snapclient #125: bootloop PCM5102A in ESP32WROVER

ExploWare, 2025-06-04:
I have already reported some on another issue, but I split it off because it is not the same module and appears to be a different cause/error.
[My previous posts](https://github.com/CarlosDerSeher/snapclient/issues/124#issuecomment-2932589637)

I have tried some other GPIO settings, but that was throwing other issues, as you can see in this log it notices an unsuccessful I2S GPIO 

```
load:0x40080404,len:3500
entry 0x400805cc
I (881) SC: Start codec chip
I (881) HAL: INIT
I (881) HAL: codec_hal done
I (881) AUDIO_BOARD: board-handle done
I (884) SC: Audio board_init done
I (888) AUDIO_HAL: Codec mode is 2, Ctrl:1
W (892) PLAYER: no sync task created?
W (897) PLAYER: no pcm chunk queue created?
W (902) PLAYER: no latency buffer semaphore created?
I (907) PLAYER: deinit player done
I (912) PLAYER: player_setup_i2s: dma_buf_len is 100, dma_buf_count is 22, sample rate: 44100, bits: 16
E (921) i2s_std: i2s_std_set_gpio(138): ws invalid
E (927) i2s_std: i2s_channel_init_std_mode(211): initialize channel failed while setting gpio pins
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400f87a9
--- 0x400f87a9: player_setup_i2s at /project/components/lightsnapcast/player.c:266
```

I've been testing the Invert Logic settings, just to check all options. While that did not throw the E error lines like above, it also did not work, throwing the same `Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.` error

ExploWare, 2025-06-05:
my last used config was completely based on the sdkconfig_PCM5102A example, I have changed nothing except for the actual pins used. 
```
root@84c0ecea2ad4:/project# git diff sdkconfig_PCM5102A sdkconfig
diff --git a/sdkconfig_PCM5102A b/sdkconfig
index 49bf395..3175df1 100644
--- a/sdkconfig_PCM5102A
+++ b/sdkconfig
@@ -1,6 +1,6 @@
 #
 # Automatically generated file. DO NOT EDIT.
-# Espressif IoT Development Framework (ESP-IDF)  Project Configuration
+# Espressif IoT Development Framework (ESP-IDF) 5.1.5 Project Configuration
 #
 CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
 CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
@@ -400,20 +400,11 @@ CONFIG_DAC_I2C_ADDR=0x20
 # I2S master interface
 #
 CONFIG_MASTER_I2S_MCLK_PIN=0
-CONFIG_MASTER_I2S_BCK_PIN=26
-CONFIG_MASTER_I2S_LRCK_PIN=25
-CONFIG_MASTER_I2S_DATAOUT_PIN=22
+CONFIG_MASTER_I2S_BCK_PIN=16
+CONFIG_MASTER_I2S_LRCK_PIN=23
+CONFIG_MASTER_I2S_DATAOUT_PIN=24
 # end of I2S master interface
 
-#
-# I2S slave interface
-#
-CONFIG_SLAVE_I2S_BCK_PIN=-1
-CONFIG_SLAVE_I2S_MCLK_PIN=-1
-CONFIG_SLAVE_I2S_LRCK_PIN=-1
-CONFIG_SLAVE_I2S_DATAOUT_PIN=-1
-# end of I2S slave interface
-
 #
 # TI PCM5102A interface Configuration
 #
@@ -1807,6 +1798,15 @@ CONFIG_MDNS_TASK_AFFINITY_NO_AFFINITY=y
 # CONFIG_MDNS_TASK_AFFINITY_CPU0 is not set
 # CONFIG_MDNS_TASK_AFFINITY_CPU1 is not set
 CONFIG_MDNS_TASK_AFFINITY=0x7FFFFFFF
+
+#
+# MDNS Memory Configuration
+#
+CONFIG_MDNS_TASK_CREATE_FROM_INTERNAL=y
+CONFIG_MDNS_MEMORY_ALLOC_INTERNAL=y
+# CONFIG_MDNS_MEMORY_CUSTOM_IMPL is not set
+# end of MDNS Memory Configuration
+
 CONFIG_MDNS_SERVICE_ADD_TIMEOUT_MS=2000
 CONFIG_MDNS_TIMER_PERIOD_MS=100
 # CONFIG_MDNS_NETWORKING_SOCKET is not set
```

Now, I've spotted the mute pin to be set to 33, and that one seems to be in use by something else, according to my schematic, so I've just changed that one now also

```
@@ -400,24 +400,15 @@ CONFIG_DAC_I2C_ADDR=0x20
 # I2S master interface
 #
 CONFIG_MASTER_I2S_MCLK_PIN=0
-CONFIG_MASTER_I2S_BCK_PIN=26
-CONFIG_MASTER_I2S_LRCK_PIN=25
-CONFIG_MASTER_I2S_DATAOUT_PIN=22
+CONFIG_MASTER_I2S_BCK_PIN=16
+CONFIG_MASTER_I2S_LRCK_PIN=23
+CONFIG_MASTER_I2S_DATAOUT_PIN=24
 # end of I2S master interface
 
-#
-# I2S slave interface
-#
-CONFIG_SLAVE_I2S_BCK_PIN=-1
-CONFIG_SLAVE_I2S_MCLK_PIN=-1
-CONFIG_SLAVE_I2S_LRCK_PIN=-1
-CONFIG_SLAVE_I2S_DATAOUT_PIN=-1
-# end of I2S slave interface
-
 #
 # TI PCM5102A interface Configuration
 #
-CONFIG_PCM5102A_MUTE_PIN=33
+CONFIG_PCM5102A_MUTE_PIN=22
 # end of TI PCM5102A interface Configuration
 
 #
```

but it doesn't give me anything else either.
Looking through the config, I can't find any other suspicious `PIN=` references either

[image]

CarlosDerSeher, 2025-06-05:
So this is a custom pcb design? I may be wrong but isn't this

`CONFIG_MASTER_I2S_DATAOUT_PIN=24`

an illegal gpio? I can't find it anywhere in the pinout

ExploWare, 2025-06-05:
Just to be sure:
I am using the physical count PIN numbers or should I be using GPIO numbers? The config seems to be using pins since it is mentioning PINs

ExploWare, 2025-06-05:
When I configure it using GPIO numbers, I get a different result:
```
Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /project
Executing "/opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /project/build/snapclient.elf -m '/opt/esp/python_env/idf5.1_py3.8_env/bin/python' '/opt/esp/idf/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:5392
load:0x40078000,len:16204
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

load:0x40080404,len:3500
entry 0x400805cc
I (284) SC: Start codec chip
I (284) HAL: INIT
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:5392
load:0x40078000,len:16204
load:0x40080400,len:4
--- 0x40080400: _init at ??:?
```

sdkconfig PCM example > Mine
```
root@ee5c1656a365:/project# git diff sdkconfig_PCM5102A sdkconfig 
diff --git a/sdkconfig_PCM5102A b/sdkconfig
index 49bf395..a307d81 100644
--- a/sdkconfig_PCM5102A
+++ b/sdkconfig
@@ -1,6 +1,6 @@
 #
 # Automatically generated file. DO NOT EDIT.
-# Espressif IoT Development Framework (ESP-IDF)  Project Configuration
+# Espressif IoT Development Framework (ESP-IDF) 5.1.5 Project Configuration
 #
 CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
 CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
@@ -399,25 +399,16 @@ CONFIG_DAC_I2C_ADDR=0x20
 #
 # I2S master interface
 #
-CONFIG_MASTER_I2S_MCLK_PIN=0
-CONFIG_MASTER_I2S_BCK_PIN=26
-CONFIG_MASTER_I2S_LRCK_PIN=25
-CONFIG_MASTER_I2S_DATAOUT_PIN=22
+CONFIG_MASTER_I2S_MCLK_PIN=6
+CONFIG_MASTER_I2S_BCK_PIN=16
+CONFIG_MASTER_I2S_LRCK_PIN=23
+CONFIG_MASTER_I2S_DATAOUT_PIN=24
 # end of I2S master interface
 
-#
-# I2S slave interface
-#
-CONFIG_SLAVE_I2S_BCK_PIN=-1
-CONFIG_SLAVE_I2S_MCLK_PIN=-1
-CONFIG_SLAVE_I2S_LRCK_PIN=-1
-CONFIG_SLAVE_I2S_DATAOUT_PIN=-1
-# end of I2S slave interface
-
 #
 # TI PCM5102A interface Configuration
 #
-CONFIG_PCM5102A_MUTE_PIN=33
+CONFIG_PCM5102A_MUTE_PIN=9
 # end of TI PCM5102A interface Configuration
 
 #
@@ -449,7 +440,16 @@ CONFIG_ENV_GPIO_RANGE_MIN=0
 CONFIG_ENV_GPIO_RANGE_MAX=39
 CONFIG_ENV_GPIO_IN_RANGE_MAX=39
 CONFIG_ENV_GPIO_OUT_RANGE_MAX=33
-# CONFIG_SNAPCLIENT_USE_INTERNAL_ETHERNET is not set
+CONFIG_SNAPCLIENT_USE_INTERNAL_ETHERNET=y
+CONFIG_SNAPCLIENT_ETH_PHY_IP101=y
+# CONFIG_SNAPCLIENT_ETH_PHY_RTL8201 is not set
+# CONFIG_SNAPCLIENT_ETH_PHY_LAN87XX is not set
+# CONFIG_SNAPCLIENT_ETH_PHY_DP83848 is not set
+# CONFIG_SNAPCLIENT_ETH_PHY_KSZ80XX is not set
+CONFIG_SNAPCLIENT_ETH_MDC_GPIO=23
+CONFIG_SNAPCLIENT_ETH_MDIO_GPIO=18
+CONFIG_SNAPCLIENT_ETH_PHY_RST_GPIO=17
+CONFIG_SNAPCLIENT_ETH_PHY_ADDR=0
 # CONFIG_SNAPCLIENT_USE_SPI_ETHERNET is not set
 # end of Snapclient Ethernet Configuration
 
@@ -1807,6 +1807,15 @@ CONFIG_MDNS_TASK_AFFINITY_NO_AFFINITY=y
 # CONFIG_MDNS_TASK_AFFINITY_CPU0 is not set
 # CONFIG_MDNS_TASK_AFFINITY_CPU1 is not set
 CONFIG_MDNS_TASK_AFFINITY=0x7FFFFFFF
+
+#
+# MDNS Memory Configuration
+#
+CONFIG_MDNS_TASK_CREATE_FROM_INTERNAL=y
+CONFIG_MDNS_MEMORY_ALLOC_INTERNAL=y
+# CONFIG_MDNS_MEMORY_CUSTOM_IMPL is not set
+# end of MDNS Memory Configuration
+
 CONFIG_MDNS_SERVICE_ADD_TIMEOUT_MS=2000
 CONFIG_MDNS_TIMER_PERIOD_MS=100
 # CONFIG_MDNS_NETWORKING_SOCKET is not set
```

CarlosDerSeher, 2025-06-05:
Yes you should use the gpio number

CarlosDerSeher, 2025-06-07:
This could be related to #117. Things to try first:
Set stream to 44.1kHz
Set buffer to <750ms if no PSRAM is available 

Have fun